luxonis/depthai-core #39: cmake fails, missing data_writer.cpp

jhurliman, 2021-01-13:
```
CMake Error at CMakeLists.txt:60 (add_library):
  Cannot find source file:

    /home/jhurliman/Code/depthai-core/shared/depthai-shared/src/general/data_writer.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .ispc

CMake Error at CMakeLists.txt:60 (add_library):
  No SOURCES given to target: depthai-core
```

themarpe, 2021-01-13:
Hi @jhurliman 
Have you made sure to initialize submodules using? 
```
git submodule update --init --recursive
```
And make sure you are using the right branch (gen2 is on `gen2` and `gen2_develop` for cutting edge)

sfmajors373, 2021-01-17:
I'm trying to build an Arch Linux AUR package for this so I am building from the tar balls, which means the above solution does not work for me.  Do you have a suggestion for how to fix this issue when building from tar balls?

themarpe, 2021-01-18:
@sfmajors373 Unfortunately as far as I'm aware, Github does not serve tarballs which include submodules. 
One option would be to download the tarballs of the references submodules separatelly and extract them into required directory.
We'll most likely add a download to a complete tarball (with submodules) on releases.

closed 2021-02-13 by themarpe

themarpe, 2021-02-13:
@jhurliman Error was presented with a more user friendly message in f8d034785aea920c1ff047dd8b43785fdb6eff52.
Note that the repository still requires to be cloned to get the needed submodules.
Final complete source tar/zips are going to be created for releases

themarpe, 2021-03-02:
@sfmajors373 and @jhurliman 
As mentioned, releases now contain both .zip and .tar.gz which includes the submodules as well.
https://github.com/luxonis/depthai-core/releases/tag/v2.0.0
Hope this enables your usecases better :)