luxonis/depthai-core #43: How to build in Windows 10?

wureka, 2021-01-19:
Environmant: Windows 10 Pro
Cmake version: 3.19.2

```
cd e:/cxx-projects/
git clone https://github.com/luxonis/depthai-core.git
cd depthai-core
git submodule update --init --recursive

```
In cmake ui:
Where is the source code: e:/cxx-projects/depthai-core
Where to build the libraries:  e:/cxx-projects/depthai-core/build

When I clicked "Configure" on cmake ui, the below errors appeared:
```
Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19042.
[hunter] Calculating Toolchain-SHA1
[hunter] Calculating Config-SHA1
[hunter] HUNTER_ROOT: C:/.hunter
[hunter] [ Hunter-ID: 062a19a | Toolchain-ID: 09f7782 | Config-ID: 2a6efb3 ]
[hunter] NLOHMANN_JSON_ROOT: C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install (ver.: 3.8.0)
CMake Warning (dev) at CMakeLists.txt:29 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable nlohmann_json_ROOT is set to:

    C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

[hunter] NLOHMANN_JSON_SCHEMA_VALIDATOR_ROOT: C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install (ver.: 2.1.1)
CMake Warning (dev) at CMakeLists.txt:33 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable nlohmann_json_schema_validator_ROOT is set to:

    C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

[hunter] XLINK_ROOT: C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install (ver.: luxonis-2020.2)
[hunter] Building XLink
loading initial cache file C:/.hunter/_Base/062a19a/09f7782/2a6efb3/cache.cmake

loading initial cache file C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Build/XLink/args.cmake

-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19042.

-- The C compiler identification is MSVC 19.28.29335.0

-- The CXX compiler identification is MSVC 19.28.29335.0

-- Check for working C compiler: D:/App (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Check for working CXX compiler: D:/App (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done
[... 289 lines cut ...]
  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /Zc:wchar_t /Zc:forScope /Zc:inline /GR /Fo"cmTC_07f98.dir\Debug\\" /Fd"cmTC_07f98.dir\Debug\vc142.pdb" /Gd /TP /errorReport:queue "D:\App\CMake\share\cmake-3.19\Modules\CMakeCXXCompilerABI.cpp"

  cmTC_07f98.vcxproj -> E:\cxx-projects\depthai-core\build\CMakeFiles\CMakeTmp\Debug\cmTC_07f98.exe

Detecting C compiler ABI info compiled with the following output:
Change Dir: E:/cxx-projects/depthai-core/build/CMakeFiles/CMakeTmp

Run Build Command(s):D:/App (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe cmTC_9e977.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m && Microsoft (R) Build Engine version 16.8.2+25e4d540b for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.28.29335 for x64

  Copyright (C) Microsoft Corporation.  All rights reserved.

  CMakeCCompilerABI.c

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_9e977.dir\Debug\\" /Fd"cmTC_9e977.dir\Debug\vc142.pdb" /Gd /TC /errorReport:queue "D:\App\CMake\share\cmake-3.19\Modules\CMakeCCompilerABI.c"

  cmTC_9e977.vcxproj -> E:\cxx-projects\depthai-core\build\CMakeFiles\CMakeTmp\Debug\cmTC_9e977.exe

```

So, any suggestion? 
Many thanks.

Luxonis-Brandon, 2021-01-19:
Will likely need @themarpe 's help on this one.

Just to make sure (and sorry for the potentially dumb question).  Are you needing to build from source?  I ask because we do provide pre-compiled binaries for Windows 10.  There are of course myriad reasons (pun intended) why you'd want to build from source, so that's why my question might be a dumb one.  :-). Just wanted to be sure.

wureka, 2021-01-20:
@Luxonis-Brandon um... I just wanted to build to DLL/.so files so that maybe I could access it with JNA in Java. If there is  depthai lib/jar for Java, then I don't need to build depthai-core from source.

Luxonis-Brandon, 2021-01-20:
Ah totally understand now @wureka .  So it's been a while since I tried to build on Windows 10.  But here are the notes that I have from then:

https://discuss.luxonis.com/d/39-depthai-sneak-peak-into-windows-support/2

I do not yet understand the errors above though yet.  So I'm not sure what is going wrong.  I haven't done much Windows debugging before.  

One thing I wonder though is if building via Visual Studio will make this work (sorry if you are already doing this and I'm not understanding):

> And for building, I installed cmake from here and Visual Studio from here, making sure to select Desktop Development with C++ option (below):
[image]

Thoughts?

And sorry about the delays here.

Thanks,
Brandon

themarpe, 2021-01-20:
Hi @wureka - sorry for the delay. 
The issue seems to boil down to files of XLink dependency not being saves as UTF-8 (while including non ascii characters).

For a quick fix from your side, follow this:
https://stackoverflow.com/a/37871934

Otherwise I'm in process of modifying these files, so this won't cause an issue in the future.

wureka, 2021-01-21:
@Luxonis-Brandon 

> One thing I wonder though is if building via Visual Studio will make this work (sorry if you are already doing this and I'm not understanding):

Well, I am not familiar with C++ stuff. But based on my experience for building opencv/openvino from source code, before start building in visual studio, I should use cmake to generate some required files, right? Based on the building instructions in README.md in depthai-core, it askes the same thing.

And I just followed the building instructions in README.md in depthai-core, ane below errors appeared:

```
PS E:\cxx-projects\depthai-core\build> cmake ..
-- Building for: Visual Studio 16 2019
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19042.
-- The CXX compiler identification is MSVC 19.28.29335.0
-- The C compiler identification is MSVC 19.28.29335.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/App (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/App (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- [hunter] Calculating Toolchain-SHA1
-- [hunter] Calculating Config-SHA1
-- [hunter] HUNTER_ROOT: C:/.hunter
-- [hunter] [ Hunter-ID: 062a19a | Toolchain-ID: 09f7782 | Config-ID: 2a6efb3 ]
-- [hunter] NLOHMANN_JSON_ROOT: C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install (ver.: 3.8.0)
CMake Warning (dev) at CMakeLists.txt:29 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable nlohmann_json_ROOT is set to:

    C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found nlohmann_json: C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.8.0")
-- [hunter] NLOHMANN_JSON_SCHEMA_VALIDATOR_ROOT: C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install (ver.: 2.1.1)
CMake Warning (dev) at CMakeLists.txt:33 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable nlohmann_json_schema_validator_ROOT is set to:

    C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- [hunter] XLINK_ROOT: C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Install (ver.: luxonis-2020.2)
-- [hunter] Building XLink
loading initial cache file C:/.hunter/_Base/062a19a/09f7782/2a6efb3/cache.cmake
loading initial cache file C:/.hunter/_Base/062a19a/09f7782/2a6efb3/Build/XLink/args.cmake
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19042.
-- The C compiler identification is MSVC 19.28.29335.0
-- The CXX compiler identification is MSVC 19.28.29335.0
-- Check for working C compiler: D:/App (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: D:/App (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
[... 202 lines cut ...]
  CMakeCXXCompilerABI.cpp

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /EHsc /RTC1 /MDd /GS /Zc:wchar_t /Zc:forScope /Zc:inline /GR /Fo"cmTC_9c1b2.dir\Debug\\" /Fd"cmTC_9c1b2.dir\Debug\vc142.pdb" /Gd /TP /errorReport:queue "D:\App\CMake\share\cmake-3.19\Modules\CMakeCXXCompilerABI.cpp"

  cmTC_9c1b2.vcxproj -> E:\cxx-projects\depthai-core\build\CMakeFiles\CMakeTmp\Debug\cmTC_9c1b2.exe

Detecting C compiler ABI info compiled with the following output:
Change Dir: E:/cxx-projects/depthai-core/build/CMakeFiles/CMakeTmp

Run Build Command(s):D:/App (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe cmTC_f428d.vcxproj /p:Configuration=Debug /p:Platform=x64 /p:VisualStudioVersion=16.0 /v:m && Microsoft (R) Build Engine version 16.8.2+25e4d540b for .NET Framework

Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) C/C++ Optimizing Compiler Version 19.28.29335 for x64

  Copyright (C) Microsoft Corporation.  All rights reserved.

  CMakeCCompilerABI.c

  cl /c /Zi /W3 /WX- /diagnostics:column /Od /Ob0 /D WIN32 /D _WINDOWS /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"cmTC_f428d.dir\Debug\\" /Fd"cmTC_f428d.dir\Debug\vc142.pdb" /Gd /TC /errorReport:queue "D:\App\CMake\share\cmake-3.19\Modules\CMakeCCompilerABI.c"

  cmTC_f428d.vcxproj -> E:\cxx-projects\depthai-core\build\CMakeFiles\CMakeTmp\Debug\cmTC_f428d.exe

PS E:\cxx-projects\depthai-core\build\CMakeFiles>
```

themarpe, 2021-01-21:
@wureka - the way you've decided to perform the build is correct.

If you want the fix immediately without changing the locale settings as mentioned above, you can modify the following file: `cmake/Hunter/config.cmake` (replace the current block with the one below)
```
hunter_config(
    XLink
    VERSION "luxonis-2020.2"
    URL "https://github.com/luxonis/XLink/archive/49528345a0e2f307d01f5eedc1a4ad47615af08b.tar.gz"
    SHA1 "0941bd940f5966ed3a1947ddf6b2bc18ba3e1e0e"
    CMAKE_ARGS
        CMAKE_POSITION_INDEPENDENT_CODE=ON
)
```

This should fix your issue, before an update is release from our side.

themarpe, 2021-02-13:
Fixed in 2bf5ebf8d08e3575835724a7bc8774e534b52d8a
Closing.

closed 2021-02-13 by themarpe