luxonis/depthai-core #55: `dai::Pipeline` provides a thread unsafe interface

kunaltyagi, 2021-02-13:
The interface class `dai::Pipeline` has a variable of type `std::shared_ptr<PipelineImpl>`.

It's not possible to get a pipeline which is guaranteed to be thread safe once returned from a function. As such, any driver which exposes `dai::Pipeline` can't guarantee run-time consistency (and in some cases, even safety).

A failure case can be as simple as:
* device init called with a pipeline
* pipeline modified before the input/output queues are created for the original pipeline

Recommendations:
* Create a `dai::FrozenPipeline` which is **only** move constructible from a `dai::Pipeline`
* Allow copy construction of `dai::Pipeline` from `dai::FrozenPipeline` (with different `PipelineImpl` objects)
* `dai::Device` works only on the frozen pipeline

themarpe, 2021-02-13:
The `dai::Device` does already work on a sort of "frozen pipeline". It essentially serializes it in the constructor, therefore any changes to the given pipeline after the constructor finishes don't affect the devices pipeline.

Functions that operate on `dai::Pipeline` are not thread safe, as it does not provide any benefits as of now. (Although that could also be an option)

Preventing unsynchronized access to the same PipelineImpl is the goal in this case.
I think we have 2 options: 
 - `const dai::Pipeline` (similar to `dai::PipelineView` if existed) where the liability falls onto users receiving this object.
 - `dai::FrozenPipeline` where the language offers (some) protection. 
 
The `dai::FrozenPipeline` must do 1 extra step to IMO really provide the language protection needed:
On move construction check whether std::shared_ptr is `unique` or `use_count` == 1, otherwise do copy of `dai::Pipeline`.

What do you think?

kunaltyagi, 2021-02-15:
I've created a [simple example](https://github.com/kunaltyagi/depthai-core/blob/thread_test/examples/src/thread_issue.cpp) (needed to added a few sleep in the `dai::Device` class).

Here, the pipeline creator gives a pipeline to a thread setting up the device, and then manages to modify it after the serialization. I might be mistaken, so please help me correct my understanding.

Your condition regarding `use_count` makes complete sense.

`dai::PipelineView` has a similar issue where the user has a big liability, and needs to copy/serialize it asap. I vote for a `dai::FrozenPipeline` over a `PipelineView`

themarpe, 2021-02-15:
The example you have provided does indeed perform modification of the pipeline after device was booted with it. Although the pipeline which device was serialized with is the one that is given to its constructor (at that point in time). If modified after, device still operates on the pipeline as it was when given in constructor.

Thinking about it - `dai::FrozenPipeline` will not only have to create when `dai::PipelineImpl` shared_ptr count > 1 but also if any nodes shared_ptr count > 1.

So moving to `dai:FrozenPipeline` only comes into account when no other code holds node pointers (as they might be non-const, therefore not providing the protection).

I think both cases are applicable: returning `const dai::Pipeline` for when *driver* (or something else) wants to provide just the view to the pipline (which can be used by `dai::Device` which also only requires a view) and the `dai::FrozenPipeline` which can be used to conveniently pass and accept a pipeline that is guaranteed to not change in the lifetime of the object.

I will take a stab at this most likely with the grouping and host pipeline functionality

kunaltyagi, 2021-02-16:
Just pointing it out: The queues in the device ctor are created on the modified pipeline, not the "original" pipeline.