luxonis/depthai-core #62: Library namespace

oysstu, 2021-02-19:
As far as I can tell, the library does not use any namespaces for its classes. Since some of the classes have very generic names (e.g. Device"), this can easily lead to collisions when included in other projects. I've seen the "dai" namespace being used in some places, but please consider using this everywhere.

Luxonis-Brandon, 2021-02-19:
Thanks @oysstu .  Makes sense and will discuss with team.  I -think- the conclusion was to standardize on dai for both Python and C++, and we are moving to that completely.  But will check.

themarpe, 2021-02-19:
@oysstu whole (`gen2`) library uses `dai` namespace for all public classes / symbols.

If you've found a case where this isn't true please let me know.

oysstu, 2021-02-20:
I see, I wasn't aware of the pipeline builder gen2. This may be a topic for another issue, or a discussion board, but how stable is the gen2 C++ interface currently? Looks like it was forked last summer, so it may make more sense to build new applications off gen2 instead of the main branch right now.

Luxonis-Brandon, 2021-02-20:
Great question @oysstu .  We are about to do our first formal release of Gen2 hopefully this month.  So I agree and would recommend developing C++ off of Gen2.  Specifically, for now, the gen2_develop branch:
https://docs.luxonis.com/projects/api/en/latest/

We had meant to transition entirely to Gen2 in December but are a bit behind.

Luxonis-Brandon, 2021-02-20:
And here is the repo that shows an example of how to use Gen2 C++ API.  A lot carries over directly from the Python API (as the Python API is just a wrapper on C++): https://github.com/luxonis/depthai-core-example
This whole repo is Gen2 based.

oysstu, 2021-02-21:
That's great, I'll close this issue seeing as all my questions are answered. Thanks!

closed 2021-02-21 by oysstu