luxonis/depthai-core #66: Source dependency submodules

oysstu, 2021-02-25:
Please consider adding the source dependencies as submodules (i.e. nlohmann_json_schema_validator, XLink). You should be able to use your current hunter_config macros with the GIT_SUBMODULE argument. I've had to disable the hunter build system, as it sets a bunch of cmake variables which interferes with the parent build process, and having the dependencies as submodules makes tracking down the dependencies manually a bit easier.

Also note that I tried to install the library in a separate build process, thus enabling me to let hunter retrieve the dependencies. This didn't work as the install process installs the static libraries (libdepthai-core.a, libdepthai-resources.a), but not the static libraries of the source dependencies mentioned above. 

If you have another way of isolating the build process, please let me know. I'm not that familiar with the hunter build system.

themarpe, 2021-02-25:
Hi @oysstu 
The goal of using package manager (`Hunter` in our case) is to remove the submodules from the repository.
I do agree that `Hunter` has its own set of details (quirks) to deal with, as it works best as a top/root project package manager.

The option you've specified, by installing the libraries will have to be fixed for static builds. You've reminded me about a way to describe this dependencies, so I'll try to integrate that in the upcoming days.

For getting it to work now, compile as a shared library: `-D BUILD_SHARED_LIBS=ON` and install. This will give you all the needed files to link and include:
 - include/
 - include/depthai/3rdparty
 - include/depthai-shared/3rdparty

oysstu, 2021-02-25:
Ah great, I was looking for a way to compile shared libraries instead of the static libs, but I looked in the hunter documentation and the CMakeList.txt's and not the cmake docs. I have no problem with the usage of hunter, but trying to integrate it directly into another build system was a bit of a handful.

Thanks for the help!

oysstu, 2021-02-25:
Reopening this, as it did not work out of the box.  The shared libraries were successfully built. However, the cmake configuration installed still causes cmake to try to link with nlohmann_json and XLInk. I can successfully link with depthai-core if I remove the interface link libraries below in depthai-coreConfig.cmake. I'm guessing this has something do do with the way your target_link_libraries is set up.

```
set_target_properties(depthai-core PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;${_IMPORT_PREFIX}/include/depthai/3rdparty;${_IMPORT_PREFIX}/include/depthai-shared/3rdparty"
  INTERFACE_LINK_LIBRARIES "nlohmann_json::nlohmann_json;XLink"
)
```

themarpe, 2021-02-25:
I haven't done any `find_package` usage from install `depthai-core`
It seems like the `INTERFACE_INCLUDE_DIRECTORIES` contains the correct directories, but nevertheless my install doesn't fill out the config correctly.

I'll refactor this to install the actual targets `nlohmann_json` and `XLink` alongside `depthai-core` with their respective `*Config.cmake` files, as well as give a try with static library by installing all (private as well) targets. 

Thanks for giving this a try.

Again a right now solution is: specify include directories manually instead of relaying on `find_package` and `*Config.cmake`

themarpe, 2021-03-02:
@oysstu latest release of Gen2 (now on `main` and `develop`) include a complete set of dependencies needed to build either shared or static.

Example:
 -  Build and install `depthai-core` to default installation (`[build_dir]/install`)
 - Your project: `find_package(depthai)` while providing `-D depthai_DIR=[build_dir]/install/lib/cmake/depthai`
 
 The above will correctly search for and find the needed dependencies. In case shared libs are used, only public dependencies are installed, otherwise private are as well (so final linking finds all needed static libraries)

closed 2021-03-02 by themarpe