luxonis/depthai-core #70: cmake configurate fails -- can not find ...DatatypeEnum.cpp

diablodale, 2021-03-02:
Initial run of cmake configure fails with error `Cannot find source file: C:/njs/depthai-core/shared/depthai-shared/src/datatype/DatatypeEnum.cpp`

### Setup

* Microsoft Windows [Version 10.0.19042.804]
* cmake 3.19.1
* Visual Studio 2019 Community v16.8.6

### Repro

1. clone this repo to directory. Currently at 376d965d55aae4bc3ee851bd694a7076b9aa88c5
2. checkout gen2_develop
3. `C:\Program Files\CMake\bin\cmake.exe" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -Hc:/njs/depthai-core -Bc:/njs/depthai-core/build -G Ninja`

### Result

```
...
[cmake] -- LIST OF RESOURCE COMPILED FILES: C:/njs/depthai-core/build/resources/depthai-device-fwp-5e4974e930f4ecbe1bf718463f6a5da664e68788.tar.xz;C:/njs/depthai-core/build/resources/depthai-bootloader-0.0.11.cmd
[cmake] -- Check for working incbin assembly directive
[cmake] -- Check for working incbin assembly directive - does not work
[cmake] -- CMRC - Compatibility mode, creating intermediate character array files
[cmake] -- OpenCV or required libraries (opencv_core;opencv_imgproc) not found. OpenCV Support disabled
[cmake] -- Configuring done
[cmake] CMake Error at CMakeLists.txt:102 (add_library):
[cmake]   Cannot find source file:
[cmake] 
[cmake]     C:/njs/depthai-core/shared/depthai-shared/src/datatype/DatatypeEnum.cpp
[cmake] 
[cmake]   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
[cmake]   .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .ispc
[cmake] 
[cmake] 
[cmake] -- Generating done
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.
```

### Expected

cmake configure to complete with no errors. I first need a clean config and generate before I can build.

### Workaround

None known. Running configure a second time results in same error.

alex-luxonis, 2021-03-02:
Could you check if the git submodules were initialized?
```
git submodule update --init --recursive
```
https://github.com/luxonis/depthai-core#building

diablodale, 2021-03-02:
Yes, my oversight. After git subm, I can configure and it built something.

Given that you are automating most of the packages with hunter, consider continuing that automation by running the git subm update within the cmake. Doing one without the other might lead to errors like mine.

Luxonis-Brandon, 2021-03-02:
Thanks for the idea.  Just brought it up internally.  I don't know enough to comment one way or another on it.

themarpe, 2021-03-02:
@diablodale Can you check whether CMake detected Git on your system. 
I've retested on `main` and it issues the designed error on both Ubuntu20.04 and Windows 10 (Git installed)

```
CMake Error at shared/depthai-shared.cmake:34 (message):
  Submodule 'depthai-shared' not initialized/updated.  Run 'git submodule
  update --init --recursive' first
Call Stack (most recent call first):
  CMakeLists.txt:88 (include)

-- Configuring incomplete, errors occurred!
```

I do agree that adding the actual call to `git submodule update --init --recursive` in the case of the error message would be the way to go, just have to make sure to only execute when uninitialized. 
But if git wasn't found, the latter won't work either.

Will take a stab at it in the upcoming days. Thanks

diablodale, 2021-03-02:
Hi. 
1. Fresh clone of repo and checkout to gen2_develop
2. Edited depthai-shared.cmake and depthai-bootloader-shared.cmake to both output message inside each their IF(GIT_FOUND)
3. Edited again both to COMMAND_ECHO STDOUT, ECHO_OUTPUT_VARIABLE, ECHO_ERROR_VARIABLE on each their execute_process(git_exe...)
4. configure cmake

### Result

Same error as OP. Notice towards the beginning that git is found.

```
[main] Configuring folder: depthai-core 
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.exe" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -Hc:/njs/lux/depthai-core -Bc:/njs/lux/depthai-core/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The CXX compiler identification is MSVC 19.28.29337.0
[cmake] -- The C compiler identification is MSVC 19.28.29337.0
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.28.29333/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] CMake Warning (dev) at CMakeLists.txt:30 (set):
[cmake]   Cannot set "DEPTHAI_VERSION": current scope has no parent.
[cmake] This warning is for project developers.  Use -Wno-dev to suppress it.
[cmake] 
[cmake] -- Found Git: C:/njs/wslgit/git.EXE (found version "2.30.0") 
[cmake] -- [hunter] Calculating Toolchain-SHA1
[cmake] -- [hunter] Calculating Config-SHA1
[cmake] -- [hunter] HUNTER_ROOT: C:/.hunter
[cmake] -- [hunter] [ Hunter-ID: 062a19a | Toolchain-ID: 8d58f30 | Config-ID: 2ddc225 ]
[cmake] -- [hunter] NLOHMANN_JSON_ROOT: C:/.hunter/_Base/062a19a/8d58f30/2ddc225/Install (ver.: 3.9.1)
[cmake] -- [hunter] XLINK_ROOT: C:/.hunter/_Base/062a19a/8d58f30/2ddc225/Install (ver.: luxonis-2021.2-develop)
[cmake] -- [hunter] BZIP2_ROOT: C:/.hunter/_Base/062a19a/8d58f30/2ddc225/Install (ver.: 1.0.8-p0)
[cmake] -- [hunter] FP16_ROOT: C:/.hunter/_Base/062a19a/8d58f30/2ddc225/Install (ver.: 0.0.0-febbb1c-p0)
[cmake] -- [hunter] LIBARCHIVE_ROOT: C:/.hunter/_Base/062a19a/8d58f30/2ddc225/Install (ver.: 3.4.2-p2)
[cmake] -- [hunter] SPDLOG_ROOT: C:/.hunter/_Base/062a19a/8d58f30/2ddc225/Install (ver.: 1.8.2)
[cmake] -- Looking for pthread.h
[cmake] -- Looking for pthread.h - not found
[cmake] -- Found Threads: TRUE  
[cmake] -- Found nlohmann_json: C:/.hunter/_Base/062a19a/8d58f30/2ddc225/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.9.1") 
[cmake] -- Compiling depthai-core resources in PATCH_ONLY mode
[cmake] -- Downloading Depthai device side binaries from server...
[cmake] -- Downloading depthai and patch
[cmake] -- commit: 
[cmake] -- Downloading and checking depthai-device-fwp.tar.xz
[cmake] -- [download 0% complete]
[cmake] -- [download 1% complete]
[cmake] -- [download 2% complete]
[cmake] -- [download 3% complete]
[cmake] -- [download 4% complete]
[cmake] -- [download 5% complete]
[cmake] -- [download 6% complete]
[cmake] -- [download 7% complete]
[cmake] -- [download 8% complete]
[cmake] -- [download 9% complete]
[cmake] -- [download 10% complete]
[cmake] -- [download 11% complete]
[cmake] -- [download 12% complete]
[cmake] -- [download 13% complete]
[cmake] -- [download 14% complete]
[cmake] -- [download 15% complete]
[cmake] -- [download 16% complete]
[cmake] -- [download 17% complete]
[cmake] -- [download 18% complete]
[cmake] -- [download 19% complete]
[... 136 lines cut ...]
[cmake] -- [download 86% complete]
[cmake] -- [download 89% complete]
[cmake] -- [download 91% complete]
[cmake] -- [download 93% complete]
[cmake] -- [download 96% complete]
[cmake] -- [download 98% complete]
[cmake] -- [download 100% complete]
[cmake] -- LIST OF RESOURCE COMPILED FILES: C:/njs/lux/depthai-core/build/resources/depthai-device-fwp-5e4974e930f4ecbe1bf718463f6a5da664e68788.tar.xz;C:/njs/lux/depthai-core/build/resources/depthai-bootloader-0.0.11.cmd
[cmake] -- Check for working incbin assembly directive
[cmake] -- Check for working incbin assembly directive - does not work
[cmake] -- CMRC - Compatibility mode, creating intermediate character array files
[cmake] -- OpenCV or required libraries (opencv_core;opencv_imgproc) not found. OpenCV Support disabled
[cmake] -- Configuring done
[cmake] CMake Error at CMakeLists.txt:102 (add_library):
[cmake]   Cannot find source file:
[cmake] 
[cmake]     C:/njs/lux/depthai-core/shared/depthai-shared/src/datatype/DatatypeEnum.cpp
[cmake] 
[cmake]   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
[cmake]   .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .ispc
[cmake] 
[cmake] 
[cmake] -- Generating done
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.
```

themarpe, 2021-03-02:
@diablodale I think the issue arises because you are configuring outside the cloned repository.

Can you retest by moving current directory inside the repository?
In that case, I'll issue a fix

diablodale, 2021-03-02:
Personally, I use vscode. cmake reports a cwd outside the project. I tested by adding into the top level cmakefiles.txt...
```
execute_process(
    COMMAND cmd.exe /c cd
    OUTPUT_VARIABLE statusCommit
    OUTPUT_STRIP_TRAILING_WHITESPACE
    COMMAND_ECHO STDOUT
    ECHO_OUTPUT_VARIABLE
    ECHO_ERROR_VARIABLE
)
```

On configure of the clean repo directly above it is...
```
'cmd.exe' '/c' 'cd'
[cmake] C:\Program Files\Microsoft VS Code
```

If some part of your cmake code needs the cwd to be within the top level project or some other subdirectory, I recommend you use the vars made for that like CMAKE_SOURCE_DIR or CMAKE_CURRENT_SOURCE_DIR. Commands like execute_process() accept a WORKING_DIRECTORY param so that you can control from which directory you want to run.

themarpe, 2021-03-02:
Issued a fix.
@diablodale thanks for reporting the problem :)

closed 2021-03-02 by themarpe

diablodale, 2021-03-02:
verified via `develop` branch.

On cmake configure, I correctly received the error
> Submodule 'depthai-shared' not initialized/updated.  Run 'git submodule update --init --recursive' first

I then ran that git command. Then cmake configure again. The configure succeeded.
I then built and resulted in exit 0 and two built libs. 👍

I'll open separate issue for build warnings.

Luxonis-Brandon, 2021-03-02:
Awesome thanks @diablodale !  (Also, on a completely new subject: Thank you for using the same Github handle and Discord handle... makes it so easy to track the conversation back/forth!)

commit 126f5fcf (2021-03-02): Fixed #70 - Git check not using correct working directory