luxonis/depthai-core #75: Fails starting the device

kota-rr, 2021-03-08:
DepthAI sporadically fails starting stereo modules device with 
```
terminate called after throwing an instance of 'std::logic_error'
 what():  Stream: __rpc_main isn't opened.
```
using gen2 pipeline https://github.com/luxonis/depthai-core/tree/gen2_develop

The device can start without problems most of the time just by restarting the pipeline.

themarpe, 2021-03-08:
@kota-rr can you describe your current setup? Are there multiple devices and/or are you connecting and disconnecting in quick succession?

Also one thing to check (or switch to) is the `main` branch (or `develop` for cutting edge) as gen2 was officially released last week, we have moved over to those 2 branches. 

The latest changes (from around a week ago) on those branches also contain some bug fixes for certain scenarios where the above error might happen, so make sure that you are using the latest

kota-rr, 2021-03-12:
> @kota-rr can you describe your current setup? Are there multiple devices and/or are you connecting and disconnecting in quick succession?
It's a single OAK-D camera setup, and it's not quick connection and disconnection.

> Also one thing to check (or switch to) is the `main` branch (or `develop` for cutting edge) as gen2 was officially released last week, we have moved over to those 2 branches.
>
> The latest changes [...]

I have been using `gen2_develop` with a bit old commit [68e6f9ec](https://github.com/luxonis/depthai-core/commit/68e6f9ec1901d71adc3cfa76c9e9c6d3c8aff02a) commited about two weeks ago.
I will update the branch and will try testing.

Luxonis-Brandon, 2021-03-12:
Thank you for doing so and also for the reports here.  Hoping that this is already fixed in the latest improvements, but if it isn't, we'll get to the bottom of it!

closed 2021-09-03 by themarpe