luxonis/depthai-core #77: ctest failing on develop branch

diablodale, 2021-03-08:
Hi. ctest is failing on the unchanged develop branch 9719f55fba6c98f57151025d849ed2fdd139bf08 😬
I don't see instructions in this repo on how to run the unit tests.
I set `DEPTHAI_BUILD_TESTS=ON`, config, debug build, and then `ctest -j15 -C Debug -T test --output-on-failure`

Is this project's ctest working? if so, what part of above setup needs to be changed?
Or should I ignore ctest and just submit PRs? I sure would like to have unit tests before I submit😉

```
[ctest]    Site: DALE-XMG
[ctest]    Build name: Win32-ninja
[ctest] Test project C:/njs/depthai-core/build
[ctest]     Start 1: color_camera_node_test
[ctest]     Start 4: neural_network_test
[ctest]     Start 2: image_manip_node_test
[ctest]     Start 3: pipeline_test
[ctest] 1/4 Test #3: pipeline_test ....................   Passed    0.76 sec
[kit] OK running C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat amd64 10.0.19041.0, env vars: [["CL",""],["_CL_",""],["INCLUDE","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\include;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\include;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\ucrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\shared;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\um;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\winrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\cppwinrt"],["LIBPATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x86\\store\\references;C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319"],["LINK",""],["_LINK_",""],["LIB","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\ucrt\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\um\\x64"],["PATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\\\Extensions\\Microsoft\\IntelliCode\\CLI;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX64\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\VCPackages;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TestWindow;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TeamFoundation\\Team Explorer;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\MSBuild\\Current\\bin\\Roslyn;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\devinit;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\10.0.19041.0\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\\\MSBuild\\Current\\Bin;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Strawberry\\c\\bin;C:\\Strawberry\\perl\\site\\bin;C:\\Strawberry\\perl\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\libnvvp;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\Nsight Compute 2019.4.0\\;;C:\\Program Files\\Docker\\Docker\\resources\\bin;C:\\ProgramData\\DockerDesktop\\version-bin;C:\\Users\\dale\\AppData\\Local\\Microsoft\\WindowsApps;C:\\njs\\wslgit;;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\Llvm\\x64\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\CMake\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\Ninja;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\Linux\\bin\\ConnectionManagerExe"],["TMP","C:\\Users\\dale\\AppData\\Local\\Temp"],["FRAMEWORKDIR","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKDIR64","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKVERSION","v4.0.30319"],["FRAMEWORKVERSION64","v4.0.30319"],["UCRTCONTEXTROOT",""],["UCRTVERSION","10.0.19041.0"],["UNIVERSALCRTSDKDIR","C:\\Program Files (x86)\\Windows Kits\\10\\"],["VCINSTALLDIR","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\"],["VCTARGETSPATH",""],["WINDOWSLIBPATH","C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0"],["WINDOWSSDKDIR","C:\\Program Files (x86)\\Windows Kits\\10\\"],["WINDOWSSDKLIBVERSION","10.0.19041.0\\"],["WINDOWSSDKVERSION","10.0.19041.0\\"],["VISUALSTUDIOVERSION","16.0"]]
[ctest] 2/4 Test #1: color_camera_node_test ...........   Passed    3.53 sec
[ctest] 3/4 Test #2: image_manip_node_test ............***Failed    5.11 sec
[ctest] 
[ctest] 4/4 Test #4: neural_network_test ..............***Failed    8.05 sec
[ctest] [14442C10C1A1C6D200] [2.375] [NeuralNetwork(2)] [warning] Network compiled for 6 shaves, maximum available 16, compiling for 8 shaves likely will yield in better performance
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 
[ctest] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ctest] neural_network_test.exe is a Catch v2.11.0 host application.
[ctest] Run with -? for options
[ctest] 
[ctest] -------------------------------------------------------------------------------
[ctest] Neural network node data checks
[ctest] -------------------------------------------------------------------------------
[ctest] ..\tests\src\neural_network_test.cpp(41)
[ctest] ...............................................................................
[ctest] 
[ctest] ..\tests\src\neural_network_test.cpp(98): FAILED:
[ctest]   REQUIRE( receivedLogMessage == false )
[ctest] with expansion:
[ctest]   1 == false
[ctest] 
[ctest] ===============================================================================
[ctest] test cases:   1 |   0 passed | 1 failed
[ctest] assertions: 121 | 120 passed | 1 failed
[ctest] 
[ctest] 
[ctest] 
[ctest] 50% tests passed, 2 tests failed out of 4
[ctest] 
[ctest] Total Test time (real) =   8.11 sec
[ctest] 
[ctest] The following tests FAILED:
[ctest] 	  2 - image_manip_node_test (Failed)
[ctest] 	  4 - neural_network_test (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
[extension] [7631] cmake.ctest finished (returned 8)
```

szabi-luxonis, 2021-03-08:
@diablodale Please update develop to latest. I issued a hotfix: https://github.com/luxonis/depthai-core/commit/9a0474522ba1c961b2ae087a4613b336dd8f648d

diablodale, 2021-03-08:
No improvement. I am currently on
```
commit 9a0474522ba1c961b2ae087a4613b336dd8f648d (HEAD -> develop, upstream/develop)
Author: SzabolcsGergely <szabi@luxonis.com>
Date:   Fri Mar 5 00:35:29 2021 +0200

    Fix build error from PR #67
```

After checkout to the commit, I rm-rf the entire build/ directory. config'd, build debug, and ran ctest as above. The latest output is

```
[proc] Executing command: ctest -j15 -C Debug -T test --output-on-failure
[ctest]    Site: DALE-XMG
[ctest]    Build name: Win32-ninja
[ctest] Create new tag: 20210308-1649 - Experimental
[ctest] Test project C:/njs/depthai-core/build
[ctest]     Start 1: color_camera_node_test
[ctest]     Start 2: image_manip_node_test
[ctest]     Start 3: pipeline_test
[ctest]     Start 4: neural_network_test
[ctest] 1/4 Test #3: pipeline_test ....................   Passed    0.73 sec
[kit] OK running C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat amd64 10.0.19041.0, env vars: [["CL",""],["_CL_",""],["INCLUDE","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\include;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\include;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\ucrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\shared;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\um;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\winrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\cppwinrt"],["LIBPATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x86\\store\\references;C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319"],["LINK",""],["_LINK_",""],["LIB","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\ucrt\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\um\\x64"],["PATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\\\Extensions\\Microsoft\\IntelliCode\\CLI;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX64\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\VCPackages;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TestWindow;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TeamFoundation\\Team Explorer;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\MSBuild\\Current\\bin\\Roslyn;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\devinit;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\10.0.19041.0\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\\\MSBuild\\Current\\Bin;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Strawberry\\c\\bin;C:\\Strawberry\\perl\\site\\bin;C:\\Strawberry\\perl\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\libnvvp;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\Nsight Compute 2019.4.0\\;;C:\\Program Files\\Docker\\Docker\\resources\\bin;C:\\ProgramData\\DockerDesktop\\version-bin;C:\\Users\\dale\\AppData\\Local\\Microsoft\\WindowsApps;C:\\njs\\wslgit;;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\Llvm\\x64\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\CMake\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\Ninja;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\Linux\\bin\\ConnectionManagerExe"],["TMP","C:\\Users\\dale\\AppData\\Local\\Temp"],["FRAMEWORKDIR","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKDIR64","C:\\Windows\\Microsoft.NET\\Fram [... cut ...]
[ctest] ===============================================================================
[ctest] test cases:   1 |   0 passed | 1 failed
[ctest] assertions: 121 | 120 passed | 1 failed
[ctest] 
[ctest] 
[ctest] 
[ctest] 50% tests passed, 2 tests failed out of 4
[ctest] 
[ctest] Total Test time (real) =   8.14 sec
[ctest] 
[ctest] The following tests FAILED:
[ctest] 	  2 - image_manip_node_test (Failed)
[ctest] 	  4 - neural_network_test (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
[extension] [5305] cmake.ctest finished (returned 8)
```

diablodale, 2021-03-08:
Just in case, I `git submodule update --init --recursive`. It reported...
```
Submodule path 'shared/depthai-shared': checked out '128f377f678bd0ade7957741ebfe7832c58fbe39'
```

However, still errors
```
[proc] Executing command: ctest -j15 -C Debug -T test --output-on-failure
[ctest]    Site: DALE-XMG
[ctest]    Build name: Win32-ninja
[ctest] Test project C:/njs/depthai-core/build
[ctest]     Start 2: image_manip_node_test
[ctest]     Start 4: neural_network_test
[ctest]     Start 1: color_camera_node_test
[ctest]     Start 3: pipeline_test
[ctest] 1/4 Test #3: pipeline_test ....................   Passed    0.81 sec
[kit] OK running C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat amd64 10.0.19041.0, env vars: [["CL",""],["_CL_",""],["INCLUDE","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\include;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\include;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\ucrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\shared;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\um;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\winrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\cppwinrt"],["LIBPATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x86\\store\\references;C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319"],["LINK",""],["_LINK_",""],["LIB","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\ucrt\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\um\\x64"],["PATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\\\Extensions\\Microsoft\\IntelliCode\\CLI;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX64\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\VCPackages;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TestWindow;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TeamFoundation\\Team Explorer;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\MSBuild\\Current\\bin\\Roslyn;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\devinit;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\10.0.19041.0\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\\\MSBuild\\Current\\Bin;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Strawberry\\c\\bin;C:\\Strawberry\\perl\\site\\bin;C:\\Strawberry\\perl\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\libnvvp;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\Nsight Compute 2019.4.0\\;;C:\\Program Files\\Docker\\Docker\\resources\\bin;C:\\ProgramData\\DockerDesktop\\version-bin;C:\\Users\\dale\\AppData\\Local\\Microsoft\\WindowsApps;C:\\njs\\wslgit;;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\Llvm\\x64\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\CMake\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\Ninja;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\Linux\\bin\\ConnectionManagerExe"],["TMP","C:\\Users\\dale\\AppData\\Local\\Temp"],["FRAMEWORKDIR","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKDIR64","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKVERSION","v4.0.30319"],["FRAMEWORKVERSION64","v4.0.30319"],["UCRTCONTEXTROOT",""],["UCRTVERSION","10.0.19041.0"],["UNIVERSALCRTSDKDIR","C:\\Program Files (x86)\\Windows Kits\\10\\"],["VCINSTALLDIR","C:\\Progr [... cut ...]
[ctest] assertions: 121 | 120 passed | 1 failed
[ctest] 
[ctest] 
[ctest] 4/4 Test #1: color_camera_node_test ...........***Failed   12.73 sec
[ctest] 
[ctest] 
[ctest] 50% tests passed, 2 tests failed out of 4
[ctest] 
[ctest] Total Test time (real) =  12.79 sec
[ctest] 
[ctest] The following tests FAILED:
[ctest] 	  1 - color_camera_node_test (Failed)
[ctest] 	  4 - neural_network_test (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
[extension] [1217] cmake.ctest finished (returned 8)

```

szabi-luxonis, 2021-03-08:
@diablodale 
Please configure build as:
`cmake .. -DDEPTHAI_BUILD_EXAMPLES=ON -DDEPTHAI_TEST_EXAMPLES=ON`
then run tests as `ctest`
Note that `Test  #5: webcam_mobilenet` will report a false error that will be addressed.
We are working on the refactorization of tests.

diablodale, 2021-03-08:
Looking at `neural_network_test.cpp` and the output...
line 56 check for log level WARN or worse.
The output gives the WARN of 6 vs 8 shaves.
Therefore that means the test would fail due to that warn.

diablodale, 2021-03-08:
Multiple failures. part of which are caused by the WARN test logic

```
[proc] Executing command: ctest -j15 -C Debug -T test --output-on-failure
[ctest]    Site: DALE-XMG
[ctest]    Build name: Win32-ninja
[ctest] Test project C:/njs/depthai-core/build
[ctest]       Start  2: image_manip_node_test
[ctest]       Start  4: neural_network_test
[ctest]       Start  1: color_camera_node_test
[ctest]       Start  3: pipeline_test
[ctest]       Start  5: camera_preview
[ctest]       Start  6: camera_video
[ctest]       Start  7: mono_camera
[ctest]       Start  8: camera_mobilenet
[ctest]       Start  9: webcam_mobilenet
[ctest]       Start 10: mjpeg_encoding
[ctest]       Start 11: h264_encoding
[ctest]       Start 12: stereo
[ctest]       Start 13: image_manip
[ctest]       Start 14: image_manip_warp
[ctest]       Start 15: color_camera_control
[ctest]  1/22 Test  #3: pipeline_test .......................   Passed    1.24 sec
[ctest]       Start 16: system_information
[kit] OK running C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat amd64 10.0.19041.0, env vars: [["CL",""],["_CL_",""],["INCLUDE","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\include;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\include;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\ucrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\shared;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\um;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\winrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\cppwinrt"],["LIBPATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x86\\store\\references;C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319"],["LINK",""],["_LINK_",""],["LIB","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\ucrt\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\um\\x64"],["PATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\\\Extensions\\Microsoft\\IntelliCode\\CLI;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX64\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\VCPackages;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TestWindow;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TeamFoundation\\Team Explorer;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\MSBuild\\Current\\bin\\Roslyn;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\devinit;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\10.0.19041.0\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\\\MSBuild\\Current\\Bin;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Strawberry\\c\\bin;C:\\Strawberry\\perl\\site\\bin;C:\\Strawberry\\perl\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\libnvvp;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\Nsight Compute 2019.4.0\\;;C:\\Program Files\\Docker\\Docker\\resources\\bin;C:\\ProgramData\\DockerDesktop\\version-bin;C:\\Users\\dale\\AppData\\Local\\Microsoft\\WindowsApps;C:\\njs\\wslgit;;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\Llvm\\x64\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\CMake\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\Ninja;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\Linux\\bin\\ConnectionManagerExe"],["TMP","C:\\Users\\dale\\AppData\\Loca [... cut ...]
[ctest] assertions: 121 | 120 passed | 1 failed
[ctest] 
[ctest]
[ctest] 22/22 Test  #1: color_camera_node_test ..............***Failed  103.84 sec
[ctest] 
[ctest] 
[ctest] 91% tests passed, 2 tests failed out of 22
[ctest] 
[ctest] Total Test time (real) = 103.91 sec
[ctest] 
[ctest] The following tests FAILED:
[ctest] 	  1 - color_camera_node_test (Failed)
[ctest] 	  4 - neural_network_test (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
[extension] [2458] cmake.ctest finished (returned 8)
```

diablodale, 2021-03-08:
Found the ctest issues:

1. The tests are currently *not* designed to be run in parallel. Therefore, the ctest params `-j` or `--parallel` must not be used
2. Warnings like `NeuralNetwork(2)] [warning] Network compiled for 6 shaves, maximum available 16, compiling for 8 shaves likely will yield in better performance` cause the test to fail due to tests\src\neural_network_test.cpp line 55

After I fixed the above two, all tests passed successfully.

I recommend the README.md for this project have a Testing section. And in that section put the instructions we both listed above and ensure no parallel options are used. Finally, your team chose how to handle warnings in the neural net test.

diablodale, 2021-03-09:
Hi. What is your team's process to close an issue? E.g., some teams have bots that autoclose after the related PR is merged.

The PR you merged works for me and from my view this issue has been resolved.

szabi-luxonis, 2021-03-09:
> Hi. What is your team's process to close an issue? E.g., some teams have bots that autoclose after the related PR is merged.
> 
> The PR you merged works for me and from my view this issue has been resolved.

We have bots too that closes the issue, but in this case the one who opened the issue should confirm if the problem is resolved.

PR #78 (2021-03-08): Gen2 tests fix
Fixes #77
closed 2021-03-09 by szabi-luxonis