luxonis/depthai-core #81: ctest tests report Passed yet they are actually failing

diablodale, 2021-03-08:
While churning today to get ctest working, I encountered multiple times the ctest suite of tests reporting that tests Passed, yet they actually failed. I was able to capture the most recent test run in which this occurred.

### Setup

* Develop branch 9a0474522ba1c961b2ae087a4613b336dd8f648d. Please note, this does not include the 6->8 shaves fix
* PR #80 applied to that branch
* Microsoft Windows [Version 10.0.19042.804]
* Visual Studio 2019 Community v16.9.0
* Build for 64-bit, static lib

### Repro

1. Setup as above
2. Config, build
3. Run ctest

### Result

* Test 4 did fail as reported
* Tests 5 through 22 actually failed. There were no popup windows, camera output, or other such UI.
* Instead, all I saw was ctest every ~5 seconds reporting "Passed"

```
[proc] Executing command: ctest -j1 -C Debug -T test --output-on-failure
[ctest]    Site: DALE-XMG
[ctest]    Build name: Win32-ninja
[ctest] Test project C:/njs/depthai-core/build
[ctest]       Start  1: color_camera_node_test
[kit] OK running C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat amd64 10.0.19041.0, env vars: [["CL",""],["_CL_",""],["INCLUDE","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\include;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\include;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\ucrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\shared;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\um;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\winrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\cppwinrt"],["LIBPATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x86\\store\\references;C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319"],["LINK",""],["_LINK_",""],["LIB","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\ucrt\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\um\\x64"],["PATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\\\Extensions\\Microsoft\\IntelliCode\\CLI;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX64\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\VCPackages;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TestWindow;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TeamFoundation\\Team Explorer;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\MSBuild\\Current\\bin\\Roslyn;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\devinit;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\10.0.19041.0\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\\\MSBuild\\Current\\Bin;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Strawberry\\c\\bin;C:\\Strawberry\\perl\\site\\bin;C:\\Strawberry\\perl\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\libnvvp;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\Nsight Compute 2019.4.0\\;;C:\\Program Files\\Docker\\Docker\\resources\\bin;C:\\ProgramData\\DockerDesktop\\version-bin;C:\\Users\\dale\\AppData\\Local\\Microsoft\\WindowsApps;C:\\njs\\wslgit;;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\Llvm\\x64\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\CMake\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\Ninja;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\Linux\\bin\\ConnectionManagerExe"],["TMP","C:\\Users\\dale\\AppData\\Local\\Temp"],["FRAMEWORKDIR","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKDIR64","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKVERSION","v4.0.30319"],["FRAMEWORKVERSION64","v4.0.30319"],["UCRTCONTEXTROOT",""],["UCRTVERSION","10.0.19041.0"],["UNIVERSALCRTSDKDIR","C:\\Program Files (x86)\\Windows Kits\\10\\"],["VCINSTALLDIR","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\"],["VCTARGETSPATH",""],["WINDOWSLIBPATH","C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0"],["WINDOWSSDKDIR","C:\\Program Files (x86)\\Windows Kits\\10\\"],["WINDOWSSDKLIBVERSION","10.0.19041.0\\"],["WINDOWSSDKVERSION","10.0.19041.0\\"],["VISUALSTUDIOVERSION","16.0"]]
[ctest]  1/22 Test  #1: color_camera_node_test ..............   Passed    3.45 sec
[ctest]       Start  2: image_manip_node_test
[ctest]  2/22 Test  #2: image_manip_node_test ...............   Passed    3.38 sec
[ctest]       Start  3: pipeline_test
[ctest]  3/22 Test  #3: pipeline_test .......................   Passed    0.50 sec
[ctest]       Start  4: neural_network_test
[ctest]  4/22 Test  #4: neural_network_test .................***Failed    7.08 sec
[ctest] [14442C10C1A1C6D200] [4.903] [NeuralNetwork(2)] [warning] Network compiled for 6 shaves, maximum available 16, compiling for 8 shaves likely will yield in better performance
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[ctest] 1
[ctest] 2
[ctest] 3
[ctest] 0
[... 55 lines cut ...]
[ctest] 18/22 Test #18: opencv_support ......................   Passed    5.06 sec
[ctest]       Start 19: mobilenet_device_side_decoding
[ctest] 19/22 Test #19: mobilenet_device_side_decoding ......   Passed    5.05 sec
[ctest]       Start 20: tiny_yolo_v3_device_side_decoding
[ctest] 20/22 Test #20: tiny_yolo_v3_device_side_decoding ...   Passed    5.08 sec
[ctest]       Start 21: tiny_yolo_v4_device_side_decoding
[ctest] 21/22 Test #21: tiny_yolo_v4_device_side_decoding ...   Passed    5.07 sec
[ctest]       Start 22: camera_mobilenet_sync
[ctest] 22/22 Test #22: camera_mobilenet_sync ...............   Passed    5.07 sec
[ctest] 
[ctest] 95% tests passed, 1 tests failed out of 22
[ctest] 
[ctest] Total Test time (real) = 105.75 sec
[ctest] 
[ctest] The following tests FAILED:
[ctest] 	  4 - neural_network_test (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
[extension] [7565] cmake.ctest finished (returned 8)

```

### Expected

Failure reports when the tests are unable to communicate with camera, render UI, etc.

themarpe, 2021-03-11:
@diablodale Can you try increasing timeout using `TEST_TIMEOUT` environment variable to eg. `10` seconds.

As tests are executed and then sent a termination signal, sometimes UI doesn't have enough time to show it yet.
A test of example can still be successful even if UI wasn't presented yet.

diablodale, 2021-03-11:
I can change that env var. What are you asking me to test?

The issue I'm reporting is multiple tests failed yet the test report was Passed. Have all the tests been designed to not need a UI. And therefore, me not seeing the UI with my eyes is *not* an indication that some part of the tests failed?
If yes, then ok...this issue is not an issue. I'll close it.
If no, then the test cases/harness have a bug that are reporting "passed" yet they actually failed.

themarpe, 2021-03-11:
Correct, testing the examples doesn't actually test UI visualization but rather error messages, exceptions or device misconfiguration (pipeline, etc...).

If UI doesn't show up even with larger timeout, then something else might be wrong (and test still saying passed), otherwise the tests are deemed successful.

diablodale, 2021-03-11:
I suspect it is the latter. Something in test#4 caused the OAK-D and/or USB to be in a state that caused all the following tests to silently fail. I acknowledge that it is possible that my laptop was busy doing something else that it could not show UI within 5 seconds. But, I doubt it because...

* it repro'd on an 8-core intel, 32gb ram, rtx2070 super, two month old laptop that was running VSCode, chrome, and running the test case. Massive CPU and memory unused.
* happened with 18 tests...and only the 18 tests after the one that failed. It wasn't a random fail here and there which is more typical of timing issues. It is mostly impossible for my laptop to be 100% cpu bound for (18*5 sec) of this test.
* ..and I saw this whole OP scenario more than once. Its just the last time that I recognized the pattern and captured it for this issue.

I have no data other than what I collected and the repro steps. I will definitely be watching for similar behavior again when I get to heavy development in H2 this year.

Did you try my repo steps yourself? Multiple runs of the the extended ctest suite? And all of those runs reported passed and showed UI?

themarpe, 2021-03-11:
Possible although usually device being in an unconnectable state would fail one way or another.

Did you hear autofocus clicks between those 18 test runs?

Will also give this a test on my end now.

diablodale, 2021-03-12:
I setup, build debug, and run ctest as in the OP. However, I forgot 🍀 to connect the OAK-D into the usb3 port.
The result was 18 tests "Passed" and 4 tests "Failed". 😆
All should fail.

I think this is another method to expose something errant in the way the test cases/harness is evaluating pass/fail. The OP exposed it in test#4. Today, I exposed it by not connecting the usb3 cable.

```
[proc] Executing command: ctest -j1 -C Debug -T test --output-on-failure
[ctest]    Site: DALE-XMG
[ctest]    Build name: Win32-ninja
[ctest] Test project C:/njs/depthai-core/build
[ctest]       Start  1: color_camera_node_test
[kit] OK running C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat amd64 10.0.19041.0, env vars: [["CL",""],["_CL_",""],["INCLUDE","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\include;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\include;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\ucrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\shared;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\um;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\winrt;C:\\Program Files (x86)\\Windows Kits\\10\\include\\10.0.19041.0\\cppwinrt"],["LIBPATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x86\\store\\references;C:\\Program Files (x86)\\Windows Kits\\10\\UnionMetadata\\10.0.19041.0;C:\\Program Files (x86)\\Windows Kits\\10\\References\\10.0.19041.0;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319"],["LINK",""],["_LINK_",""],["LIB","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\ATLMFC\\lib\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\lib\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\ucrt\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\lib\\10.0.19041.0\\um\\x64"],["PATH","C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\\\Extensions\\Microsoft\\IntelliCode\\CLI;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.28.29910\\bin\\HostX64\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\VCPackages;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TestWindow;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\TeamFoundation\\Team Explorer;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\MSBuild\\Current\\bin\\Roslyn;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Team Tools\\Performance Tools;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\Shared\\Common\\VSPerfCollectionTools\\vs2019\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\devinit;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\10.0.19041.0\\x64;C:\\Program Files (x86)\\Windows Kits\\10\\bin\\x64;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\\\MSBuild\\Current\\Bin;C:\\Windows\\Microsoft.NET\\Framework64\\v4.0.30319;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\Tools\\;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Program Files\\Microsoft VS Code\\bin;C:\\Strawberry\\c\\bin;C:\\Strawberry\\perl\\site\\bin;C:\\Strawberry\\perl\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\bin;C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.1\\libnvvp;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files\\NVIDIA Corporation\\Nsight Compute 2019.4.0\\;C:\\Program Files\\Docker\\Docker\\resources\\bin;C:\\ProgramData\\DockerDesktop\\version-bin;C:\\Users\\dale\\AppData\\Local\\Microsoft\\WindowsApps;C:\\njs\\wslgit;;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\Llvm\\x64\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\CMake\\bin;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\CommonExtensions\\Microsoft\\CMake\\Ninja;C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\Common7\\IDE\\VC\\Linux\\bin\\ConnectionManagerExe"],["TMP","C:\\Users\\dale\\AppData\\Local\\Temp"],["FRAMEWORKDIR","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKDIR64","C:\\Windows\\Microsoft.NET\\Framework64\\"],["FRAMEWORKVERSION","v4.0.30319"],["FRAMEWORKVERSION64","v4.0.30319"],["UCRTCONTEXTROOT",""],["UCRTVERSION","10.0.19041.0"],["UNIVERSALCRTSDKDIR","C:\\Program Files (x86)\\Windows Kits\\10\\"],["VCINSTALLDIR","C:\\Program Files [... cut ...]
[ctest] 22/22 Test #22: camera_mobilenet_sync ...............   Passed    5.03 sec
[ctest] 
[ctest] 86% tests passed, 3 tests failed out of 22
[ctest] 
[ctest] Total Test time (real) = 106.65 sec
[ctest] 
[ctest] The following tests FAILED:
[ctest] 	  1 - color_camera_node_test (Failed)
[ctest] 	  2 - image_manip_node_test (Failed)
[ctest] 	  4 - neural_network_test (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
[extension] [9213] cmake.ctest finished (returned 8)
```

diablodale, 2021-03-12:
Did it click? I can't remember.
I've been running the ctest repeated as in OP today. I get random failures of test cases, but I haven't seen the no UI of the OP yet.

themarpe, 2021-03-12:
Retested the without USB on Windows - issue stems from 'Debug' configuration halting the execution because uncaught exception is raised. 'Release' configuration terminates before timeout and tests fail.

WRT your previous point - I did get some cases of #79 - but haven't been able to pin it down yet. 
Never had a case where all 18 cases wouldn't show UI. Some in between yes, but not all of them. 

So I think its a combination of both - Device getting into odd USB state, where it cannot be found by OS, therefore this latter observation comes into play as you are testing 'Debug' configuration.

Will add exception handling for examples, to both solve this and show first hand that some functions may throw.

diablodale, 2022-02-03:
@themarpe I recommend we close this issue. There are other issues (and fixes) that are more specific and actionable.
🤔