luxonis/depthai-core #91: Using depthai with another (complex) cmake based project?

rlabs-oss, 2021-03-23:
Added "add_subdirectory(depthai-core EXCLUDE_FROM_ALL)" to 2nd line in RTAB-Map CMakelists. 

Depthai builds fine as is as a static lib. Using that from rtabmap Ive tried symlinking to the depthai and adding it as a subfolder as above. Didnt work. 
/rtabmap/corelib/src/../include/rtabmap/core/camera/CameraDepthAI.h:40:10: fatal error: 'depthai/depthai.hpp' file not found
#include <depthai/depthai.hpp>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.

Previously I built depthai-core then added manual overides on the cmake invocation

cmake -Dnlohmann_json_DIR="/home/m/external-source/depthai-core/build/install/lib/cmake/nlohmann_json" -DXLink_DIR="/home/m/external-source/depthai-core/build/install/lib/cmake/XLink" -Ddepthai-core_DIR="/home/m/external-source/depthai-core/build/install/lib/cmake/depthai" -DWITH_GTSAM=OFF -DWITH_DEPTHAI=ON ..

Which used to work... but not anymore the linking fails
/usr/bin/ld: cannot find -lBZip2::bz2
/usr/bin/ld: cannot find -lFP16::fp16
/usr/bin/ld: cannot find -larchive_static
/usr/bin/ld: cannot find -lspdlog::spdlog

Which raises the question - how is another project with complex cmakelists like rtabmap suppose to use depthai?

Considering depthai is static how do I include the static lib and include into rtabmap so it finds the packages? Currently depthai is installed under build/install and cmake cant find it which lead to the manual directory hackery. 

as you may have gathered Im not a cmake expert.

themarpe, 2021-03-23:
Hi @rlabs-oss.

One option is `add_subdirectory`. Another option is `find_package`. You can checkout both over at `tests/integration/CMakeLists.txt`

I'd suggest the latter to not need to modify your CMake script to much and bringing in the depthai targets, but it does require building depthai separately.

Latest changes support using `find_package` on build directory as well so you can point it there (although if you want to point it to install directory, point it to `-D depthai_DIR=[.../build/install/lib/cmake/depthai]`:
```
# Build depthai first
cmake -S depthai-core -B depthai-core/build
cmake --build depthai-core/build --parallel
# Configure and build your project
cmake -S RTAB-Map -B RTAB-Map/build -D depthai_DIR=$PWD/depthai-core/build
cmake --build RTAB-Map/build --parallel
```

Now for the above error, I think that the issues are present solely because of missing `target_link_libraries` call.
As mentioned, check `tests/integration/CMakeLists.txt` for the usage of that - after you link to `depthai::core` (or `depthai::opencv`) then all the dependencies and include directories, etc.. will be resolved by CMake under the hood.

rlabs-oss, 2021-03-23:
@themarpe Thank you I will try the suggestions.

rlabs-oss, 2021-03-23:
Hunter seems to download the bootloader, if rtabmap is adding depthai as a sub 
-- Downloading depthai bootloader
-- folder: /home/m/external-source/rtabmap/build/depthai-core/resources
-- Found RealSense2: 

yet it still cant find depthai

CMake Error at CMakeLists.txt:539 (FIND_PACKAGE):
  Could not find a package configuration file provided by "depthai" with any
  of the following names:

    depthaiConfig.cmake
    depthai-config.cmake

```
if(WITH_DEPTHAI)
        add_subdirectory("${CMAKE_CURRENT_LIST_DIR}/../depthai-core" depthai-core EXCLUDE_FROM_ALL)
endif()

adding
cmake -Ddepthai_DIR=/home/m/external-source/depthai-core/build/install/lib/cmake/depthai -DWITH_GTSAM=OFF -DWITH_DEPTHAI=ON ..
```

Progresses and then fails with
```
  Some (but not all) targets in this export set were already defined.

  Targets Defined: depthai::core;depthai::opencv

  Targets not yet defined: depthai::depthai-resources;depthai::cmrc-base

```

also fails
cmake prefix on the depthai-core folder fails, and on the build/install folder.

closed 2021-03-23 by rlabs-oss

nickjrz, 2022-02-08:
[image]
[image]

I am currently trying to build rtabmap with depthai and it seems that it succeeds at finding it, see picture attached. But when I check `rtabmap --version` it shows as Depthai = false and driver for OAK-D is disabled on the camera drivers. I am also specifying path of installation for depthai-core on the cmake command, depthai core has been built with depthai shared too. Any insight on what this behavior could be?

themarpe, 2022-02-08:
@nickjrz I think this question would be best suited for https://github.com/introlab/rtabmap repository

CC: @rlabs-oss

nickjrz, 2022-02-08:
@themarpe  sounds good, I will post it on there. Thanks