luxonis/depthai-core #95: ImgFrame.cpp:42:20: error: ‘CV_16FC1’ was not declared in this scope

ans-qureshi, 2021-03-31:
Hi! There seems to be a coding error here, please clarify. Thanks
[image]

alex-luxonis, 2021-03-31:
We only support OpenCV version 4.
It could be that you have version 3 or older installed. Could you try to upgrade it?

Also we should improve the build system checks here, and either error out on the old version detected, or build without OpenCV support.

themarpe, 2021-03-31:
Will add a PR for this now.

ans-qureshi, 2021-03-31:
Hi @alex-luxonis , the version of my OpenCV is 4.2 :/ same problem

themarpe, 2021-03-31:
@ans-qureshi can you point it to your 4.2 OpenCV installation directory?
```
cmake [...] -D OpenCV_DIR=[path/to/install_dir/lib/cmake/opencv4]
```

civerachb-cpr, 2021-03-31:
I had a similar set of errors on Ubuntu 18.04 while trying to build this + the noetic-devel branch of depthai-ros.  I was able to resolve the problems in my case by:

1. Adding this PPA to install OpenCV4: https://launchpad.net/~cran/+archive/ubuntu/opencv-4.2
2. Modifying `/opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake` on lines 94 and 96 to change `/usr/include/opencv` to `/usr/include/opencv4`.

Once those changes were made I was able to get everything compiled correctly.

themarpe, 2021-03-31:
@civerachb-cpr the mentioned changes could have some issues down the line for other ROS libraries/apps that rely on OpenCV version 3 compared to version 4.

Can you retest with just adding the PPA and using the original `cv_bridgeConfig.cmake`, while applying the PR: https://github.com/luxonis/depthai-core/pull/96

This should find and use the correct version (OpenCV 4) if installed on system in regular location

civerachb-cpr, 2021-04-01:
On my system the folder `/usr/include/opencv` does not exist, so reverting the change to `cv_bridge` would still result in a broken build on my system.  I suppose I could symlink `/usr/include/opencv` to point to `/usr/include/opencv4`, but I'm not sure that's necessary for the scope of this test.

themarpe, 2021-04-01:
@ans-qureshi should be fixed by #96. Closing

closed 2021-04-01 by themarpe

ans-qureshi, 2021-04-02:
Yup. Works. @themarpe