luxonis/depthai-core #99: cannot compile with opencv

chopin1998, 2021-04-07:
error log here:

`
[ 98%] Building CXX object CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.o
/home/marco/tmp/build/depthai-core/src/opencv/ImgFrame.cpp: In member function ‘cv::Mat dai::ImgFrame::getFrame(bool)’:
/home/marco/tmp/build/depthai-core/src/opencv/ImgFrame.cpp:42:20: error: ‘CV_16FC1’ was not declared in this scope; did you mean ‘CV_16SC1’?
   42 |             type = CV_16FC1;
      |                    ^~~~~~~~
      |                    CV_16SC1
/home/marco/tmp/build/depthai-core/src/opencv/ImgFrame.cpp:55:20: error: ‘CV_16FC3’ was not declared in this scope; did you mean ‘CV_16SC3’?
   55 |             type = CV_16FC3;
      |                    ^~~~~~~~
      |                    CV_16SC3
make[2]: *** [CMakeFiles/depthai-opencv.dir/build.make:63: CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:274: CMakeFiles/depthai-opencv.dir/all] Error 2

`

themarpe, 2021-04-07:
@chopin1998 Use OpenCV 4 version.
Either switch over to `develop` branch - latest `main` doesn't have explictly set `OpenCV 4` version in yet. 
(You may also use `main`, but specify `-D OpenCV_DIR=[path/to/opencv4]`).

Closing for now.

closed 2021-04-07 by themarpe

chopin1998, 2021-04-07:
not work...

chopin1998, 2021-04-07:
> @chopin1998 Use OpenCV 4 version.
> Either switch over to `develop` branch - latest `main` doesn't have explictly set `OpenCV 4` version in yet.
> (You may also use `main`, but specify `-D OpenCV_DIR=[path/to/opencv4]`).
> 
> Closing for now.

not work...

either use -D OpenCV_DIR nor checkout develop branch...