luxonis/depthai-core #100: Silent failure on high fps acquisition

oysstu, 2021-04-08:
Setting the target fps for the mono cameras on the Oak-D to 70 or above causes the output queues to not generate any data without reporting any errors. The issue persists even if no encoder is used for the mono-stream,  if a color stream is used or not, and if blocking/non-blocking output queues are used.  The CPU usage reported for LeonOS/LeonRT is approximately 15% and 2.5% when the issue occurs, indicating that the pipeline may not have been started successfully.

Let me know if I can provide any additional details.

alex-luxonis, 2021-04-08:
Thanks for the issue report. 
With the current camera sensor configs, the max FPS is (depending on the selected resolution):
- ColorCamera (IMX378/IMX477):
  - 118 for `THE_1080_P`
  - 42 for `THE_4_K`
  - 30 for `THE_12_MP`
- MonoCamera (OV9282/OV9281):
  - 120 for `THE_400_P`
  - 60 for `THE_720_P` or `THE_800_P`

We will add capping in the FW and log warnings, and also properly document this.

oysstu, 2021-04-08:
I see, I assumed that the frame rate in the specifications was at full resolution. It would be good to have an error message if the frame rate is set too high.  Perhaps an option to set the frame rate to the highest possible setting given a specified resolution would be good, or a way to query the device for its max fps.

Luxonis-Brandon, 2021-04-08:
Hi @oysstu ,

Sorry about the confusion here.  Yes, long-term doing full framerate (120FPS) at 1280x800 should be possible.  We have to re-write the driver though to support this.  So we haven't tackled it yet in lieu of other features.

And agreed on a way to query the device.  That will help now and in the future.  

Thanks,
Brandon