luxonis/depthai-core #117: undefined reference to symbol 'pthread_create@@GLIBC_

AndreFCosta608, 2021-04-25:
Hi.

I'm having severe problems running this lib on a CodeBlocks project.
I'm trying to use it as a static library in a C ++ project with CodeBlocks.
However, in the compilation / linking project, the following error occurs:

/ usr / bin / ld: undefined reference to symbol 'pthread_create @@ GLIBC_

I edited the included CmakeLists.txt ...

IF (NOT APPLE AND NOT MSVC)
    SET (PLATFORM_SPECIFIC_LIBS "-lpthread")
    set (CMAKE_THREAD_LIBS_INIT "-lpthread")
    SET (CMAKE_C_FLAGS_RELEASE "$ {CMAKE_C_FLAGS_RELEASE} -pthread")
    SET (CMAKE_CXX_FLAGS "$ {CMAKE_CXX_FLAGS} -pthread")
    set (CMAKE_HAVE_THREADS_LIBRARY 1)
    set (CMAKE_USE_PTHREADS_INIT 1)
    set (THREADS_PREFER_PTHREAD_FLAG ON)
ENDIF ()

Distributes “#include <pthread.h>” to all source files that included thread or memory include…
It also includes "-pthread" and "lpthread" and #include <pthread.h> "in my project files that consume the library.

My environment is CodeBlocks running on Raspibian.

I don't know what else to try.
Any tips?

Thanks.

themarpe, 2021-04-25:
Hi @AndreFCosta608 
I'd suggest you to try out the CMake example: https://github.com/luxonis/depthai-core-example

And use CMake to generate the CodeBlocks project: https://stackoverflow.com/a/37618167

Note that above wasn't done before and might now work as expected. Below steps are more general and should probably apply to CodeBlocks custom build system as well.

Adding depthai as shared library to CodeBlocks project could be done, but pay attention to add necessary include paths: https://github.com/luxonis/depthai-core/issues/116 and follow the steps for building a shared library: https://github.com/luxonis/depthai-core#building (After building install to have all the needed files in [build]/install directory: `
cmake --build . --target install --parallel 8`)

Luxonis-Brandon, 2021-04-25:
(Just wanted to say that I’m really excited to see this in CodeBlocks.)

AndreFCosta608, 2021-04-26:
Hi…
I managed to solve it.
It was enough to include a reference to lib libpthread.so.0 as soon as being the first in the link list.
Also include reference to lib libgthread-2.0.a
There was no need to edit in CmakeLists.txt or include “#include <pthread.h>” in the sources.

Thanks.

closed 2021-04-26 by AndreFCosta608

Luxonis-Brandon, 2021-04-26:
Thanks for circling back @AndreFCosta608 !  And we'd love to share what you make if that's of interest.  I've long wanted to see DepthAI used in CodeBlocks.  :-)