luxonis/depthai-core #121: libusb is required

MikaMaracuja, 2021-04-28:
When compiling under debian buster, I got stuck into

```
CMake Error at XLink.cmake:27 (message):
  libusb is required
Call Stack (most recent call first):
  CMakeLists.txt:8 (include)
```

after running 

`sudo apt-get -y install cmake libusb-1.0-0-dev build-essential`

the compile process went almost through.

themarpe, 2021-04-28:
Hi @MikaMaracuja 
Is this on latest `main` branch?
Can you retry on `develop` - otherwise a known fix which was needed in the past is to create a symlink to /usr/lib for libusb shared library. Although shouldn't be needed anymore. 
I presume this is on Raspberry PI?

MikaMaracuja, 2021-04-28:
Its on debian buster.

closed 2021-04-28 by MikaMaracuja