luxonis/depthai-core #123: Segmentation fault

AndreFCosta608, 2021-04-28:
Helo everybody.

I finally managed to compile my project inside the codeblocks.
Apparently, all depthai-core dependency libraries have been properly compiled and installed.
After a good spanking, my program finally finds libdepthai-core.so.

So I took the example given in: depthai-core / examples / src / camera_video_example.cpp.
However, when I try to run it bursts on the line “dai :: Device d (p);”.
With the following error: "Segmentation fault".

I have already enabled the Raspberry pi 3 camera interface.
3D camera on and connected via USB.

I keep trying to solve it.
Any tips on what could be the problem?

Thanks.

themarpe, 2021-04-29:
@AndreFCosta608 does compiling the example using CMake as described in README work for you?

And WRT to your specific case, can you run the example under `gdb` and post the backtrace?
```
$ gdb camera_video
gdb> run
... crash
gdb> bt
```

AndreFCosta608, 2021-04-30:
Hi.
I managed to solve it.
WORKING!!!!!!!!!!! :)
Apparently it was a flaw in the build.
What I did was to put the method that reads the camera into a try.
I recompiled library by library .. FP16, json, Xlink .. I recompiled again as dynamic libraries.
I noticed that the install did not take all * .so to user / local / lib. So I copied them manually there.
I did a "sudo ldconfig".
And in my project, when doing "Clean" and "Build" I managed to capture 2D images from the central camera.
Thanks.

closed 2021-04-30 by AndreFCosta608

Luxonis-Brandon, 2021-04-30:
AWESOME!  Thanks for letting us know.  So I'm not as much in the code - so apologies if this is a dumb question - but is there anything we need to do on our build system to make this easier for you (or others) in the future?

AndreFCosta608, 2021-04-30:
On 30/04/2021 13:13, Luxonis-Brandon wrote:
> AWESOME! Thanks for letting us know. So I'm not as much in the code - 
> so apologies if this is a dumb question - but is there anything we 
> need to do on our build system to make this easier for you (or others) 
> in the future?

I had a hard time working on C ++ until it worked.
It was arduous weeks of learning.
It was stressful.

First major problem was the dependency libraries.
As depthai-core used other libraries such as XLink, json, FP16 ... And 
they did not come in the package on the part of third parties, I had to 
be able to identify the need for each one, so download it.
Then the versions that I installed through apt-get did not match those 
used in depthai-core.
I had to test versions of the dependency libraries until I found the 
correct ones.

XLink even found the fonts in the Luxonis repository. But it did not 
meet the namespace. The .h sources inside the include should be inside 
XLink. And they weren't. When trying to compile, there was always a 
reference error.

Another big problem. That it took me more than a week. Conflicts with 
multithreaded libraries. I even created an issue on github. Do not 
compile until you include references in the project link for two 
multithreaded OS libraries. And it has a specific order, otherwise it 
doesn't work.

One more problem. The references to the XLink include are scattered 
within the main source of depthai-core, bootloader, 3Parts ... Until I 
realized that I needed to search for all of them and include them one by 
one, I lost a lot of time.

I installed the libraries from my user directory. The buidl of both 
depthai-core and other libraries did not export .so to usr / local / lib.
Linux does not look for libraries in the user directory.
I had to copy all the dependencies manually to usr / local / lib.
Even so. After compiling, my executable could not find the libraries.
Using the command “ldd myProgram” via the list of dependencies and it 
appeared not found for libdepthai-core-so.
He only solved it by executing the command "sudo ldconfig".

My environment is Linux Raspibian in 32bits. Running on a Raspberry pi 
3. Using the CodeBlocks IDE.

My suggestions:
1) Include all sources of dependencies. Even if in a separate directory.
2) In the depthai-core buid, compile and install the dependencies as 
well. Already copy the .so to usr / local / lib and registering in the 
system (ldconfig).
3) In the documentation ("Troubleshooting") mention the multithreaded 
compilation bug indicating that references to multithreaded libraries 
should be added to the project.
4) In the documentation ("Troubleshooting") mention the necessary 
include paths.

Today, after working hours, I will test the epipolar part of the camera.
Then, the control needs to be tested via reverse kinematics of the servo 
motors of the arm.
Then only the fun part of the project will remain. That is the part of 
computer vision and reinforcement of learning.

I didn't go into details now so the email doesn't get too long.
In need of me, I am available.

Thanks.

Luxonis-Brandon, 2021-04-30:
Thank you @AndreFCosta608 for the thorough feedback here.  This is extremely useful for us to help make this less painful in the future (for you and others).

And sorry about all the trouble with integration here.  Will triage with the team and circle back.  (I'm not sufficiently in the code to triage/comment directly.)

Thanks again,
Brandon