luxonis/depthai-core #127: OpenCV support in depthai-core project in C++

AndreFCosta608, 2021-05-01:
Hi.
I have the following error:
error: ‘using element_type = class dai :: ImgFrame’ {aka ‘class dai :: ImgFrame’} has no member named ‘getFrame’; did you mean ‘ImgFrame’?

Searching the documentation, I found that the “getFrame” and “getCVFrame” methods are part of opencv support. Apparently my build did not incorporate this support.
The “libdepthai-opencv.so” was generated.
I keep investigating the problem in search of a solution.
But if anyone has any suggestions I will be very grateful.

themarpe, 2021-05-02:
If doing non-CMake integration you have to manually specify a define `DEPTHAI_OPENCV_SUPPORT` to enable this API and link to depthai-opencv.

alex-luxonis, 2021-05-02:
@themarpe For cases where the library is built without OpenCV support (either not found - with CMake, or other cases like above), should we create some stubs for these functions `getFrame, getCvFrame` to report a more user-friendly compile time error as here: https://stackoverflow.com/a/18814056 , or another option: deprecation warning (not quite the case, but not aware of another way) + runtime error.

themarpe, 2021-05-02:
@alex-luxonis yeah was thinking stubs would be better suited error wise. 
Only issue is that not sure if we can do compile time actually. Would be best though.

Both of the options listed won't be able to change the signature when opencv is available.

Maybe a better suited alternative is "config" header that lists these includes eg. `DEPTHAI_OPENCV_SUPPORT` and is generated and installed as part of the build process. Then other headers requiring these definitions just have to include config header. 
That way users don't have to deal with that.

And if config doesn't specify the support we can "create" (#ifdef) the function with deprecation message that states the missing support.

Will take a look and try integrating something appropriate

AndreFCosta608, 2021-05-02:
Is it necessary to define a kind of environment variable called "DEPTHAI_OPENCV_SUPPORT" with the path to lib "libdepthai-opencv.so"?
This is a big problem for anyone using CodeBlocks. This is invisible to the IDE. At least in the standard build process.
Is it possible to link the reference in the compilation of “libdepthai-core.so” itself?

themarpe, 2021-05-02:
@AndreFCosta608 no, only a compile definition (for now, I'm working on making this part of build configuration, so no need to manually specify this definition) `-DDEPTHAI_OPENCV_SUPPORT`

Then you must link to `libdepthai-opencv.so` instead of `libdepthai-core.so`

For a quick test you can also do `#define DEPTHAI_OPENCV_SUPPORT` before including anything else

AndreFCosta608, 2021-05-02:
I did not understand.
Is this test in my project with CodeBlocks or in the depthai-core cmake file?

themarpe, 2021-05-03:
@AndreFCosta608 I think you can add these defines under the following tab (Project build settings) in CodeBlocks. And there you then add `DEPTHAI_OPENCV_SUPPORT`. After #125 gets merged this won't be necessary anymore.
[image]

AndreFCosta608, 2021-05-03:
This was a great help.

Solved it. Compiling and running !!!

I had hit the spot for the include. But I had misspelled “DEPTHAI_OPENCV_SUPPORT”.
Things from those who have become nights at the project.

100% functional camera.

Thanks.

closed 2021-05-03 by AndreFCosta608