luxonis/depthai-core #130: Cannot stream Depth image when RGB alignment

visionbike, 2021-05-05:
Hi all,

I followed the RGB alignment example and when I wanted to stream "depth" stream (instead "disparity") I got problem:
terminate called after throwing an instance of 'std::runtime_error'
 what():  Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)

I wonder if we have solution for this because I want to collect depth map beside disparity map. Hope receive the respond soon.

Thank you so much!

alex-luxonis, 2021-05-05:
Sorry for the trouble here... I'll look what's going wrong on device side.

Could you use for now depth computed from disparity on host? It's done by passing the disparity through a pre-computed LUT.
I've updated the example to use that here: 
https://github.com/luxonis/depthai-core/compare/host_disp2depth

visionbike, 2021-05-12:
Hi,

When running RGB-D alignment example, I found that the disparity map delays a little bit than RGB image (look clearly when doing some action fast).

video link:
https://1drv.ms/u/s!Aqa2tJmag-nlgaYYce8wiBE-qIdiyQ?e=Kywcad

After changing into extended disparity mode, the performance is better, however the program crashes later.

video link:
https://1drv.ms/u/s!Aqa2tJmag-nlgaYbK0Fo4BBdBihW2Q?e=cFzmlW

Thank you.

Luxonis-Brandon, 2021-05-12:
Hi @visionbike ,

(I really like your Github handle, I hope you're making what I think you're making.)

Thanks for the report, and sorry for trouble.

So in Gen2 (which is what everything is built on now) we have not yet done the temporal sync between the RGB and grayscale cameras, so the RGB and depth is not synced in time.  So @alex-luxonis has been meaning to implement this, but keeps getting pulled by other fixes/implementations/etc.

In Gen1, we have this sync - so we know we can do it.  Gen2 is just a lot more flexible, so the porting effort is non-trivial, as we need to make sure the flexibility of Gen2 is not broken when implementing this sync.

So anyway I think we'll be to jump on this soon.  I'll sync with Alex to see when he thinks he can jump on this.

Thoughts?

Thanks,
Brandon

visionbike, 2021-05-12:
Great! I think I am waiting Gen2 update. Thank you for your fast respond.

Luxonis-Brandon, 2021-05-12:
Thanks.  Oh and on Extended/Subpixel - there is a known issue that when the RGB camera is used, a crash occurs when those are enabled.  We have been trying to debug/workaround that but haven't figured it out yet.  

We're discussing internally, but I'm thinking we should do the time-sync implementation prior to the Extended/Subpixel crash debug/fix.  Thoughts?

visionbike, 2021-05-12:
Definite agree! Thank you!

alex-luxonis, 2021-05-12:
The latency is mainly caused here by an increased time the StereoDepth node needs to run with the alignment mode enabled (we still need to optimize it), and an internal queuing issue which could cause a delay of up to 8 frames. Until this is fixed, could you try lowering the camera FPS,  like:
```
cam->setFps(12);
left->setFps(12);
right->setFps(12);
```

closed 2021-06-26 by visionbike