luxonis/depthai-core #155: getImuToCameraExtrinsics Error

PRono8, 2021-06-11:
Hi,

Thank you for your work. 
I have seen in the documentation the getImuToCameraExtrinsics and getCameraToImuExtrinsics functions. I have tried to use them in a similar way to the getCameraExtrinsics functions in the example code https://github.com/luxonis/depthai-core/blob/main/examples/src/calibration_reader.cpp. But I get the following error: segmentation fault ('Core' dump).

If someone could help me I would really appreciate it. Thank you very much
Cheers,
Jose

Luxonis-Brandon, 2021-06-11:
Hi @PRono8 ,

So @saching13 can correct me if I'm wrong, but my understanding is that all of our existing product do not have IMU to camera calibration performed - and this is why these are not working.  

That said, they shouldn't seg fault, so CC: @saching13 on that.

Sachin is working to write the routines now that will allow calibrating the IMU to the cameras.  And so these functions were actually architected in there as part of that effort - but we currently do not have the capability (to my knowledge) to calibrate the IMU to the cameras.

@saching13 is wrapping up some other things with calibration, and when that is done, he's moving on to IMU calibration, which will involve Vicalib Integration.

Thanks,
Brandon

saching13, 2021-06-11:
Hi @PRono8  ,
Like Brandon mentioned our devices don't ship with IMU calibration. Will be adding that feature soon and provide a code to calibrate that by the users. 

And sorry for Seg Fault. I will check on that and replace it with an appropriate error.

closed 2021-06-17 by saching13

chobitsfan, 2024-11-28:
Hi @Luxonis-Brandon and @saching13 
I am geting "IMU calibration data is not available on device yet." from getImuToCameraExtrinsics in a recently bought OAK-D Lite. Are you sitll working on this? Thank you very much