luxonis/depthai-core #162: Hunter download error while building

ans-qureshi, 2021-06-23:
Hi! I am facing an issue while building depthai-core. This is what I'm getting:

```
root@NanoPi-NEO-Plus2:~/depthai-core# cmake -H. -Bbuild -D BUILD_SHARED_LIBS=ON
-- [hunter] Initializing Hunter workspace (062a19ab13ce8dffa9a882b6ce3e43bdabdf75d3)
-- [hunter]   https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz
-- [hunter]   -> /root/.hunter/_Base/Download/Hunter/0.23.258/062a19a
CMake Error at Build/Hunter-prefix/src/Hunter-stamp/download-Hunter.cmake:159 (message):
  Each download failed!

    error: downloading 'https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

         --- LOG END ---
         error: downloading 'https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

         --- LOG END ---
         error: downloading 'https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

         --- LOG END ---
         error: downloading 'https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

         --- LOG END ---
         error: downloading 'https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

         --- LOG END ---
         error: downloading 'https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz' failed
         status_code: 1
         status_string: "Unsupported protocol"
         log:
         --- LOG BEGIN ---
         Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

         --- LOG END ---
         
    

make[2]: *** [Hunter-prefix/src/Hunter-stamp/Hunter-download] Error 1
make[1]: *** [CMakeFiles/Hunter.dir/all] Error 2
make: *** [all] Error 2

[hunter ** INTERNAL **] Build project failed
[hunter ** INTERNAL **] [Directory:/root/depthai-core]

------------------------------ ERROR ------------------------------
    https://docs.hunter.sh/en/latest/reference/errors/error.internal.html
-------------------------------------------------------------------

CMake Error at cmake/HunterGate.cmake:87 (message):
Call Stack (most recent call first):
  cmake/HunterGate.cmake:97 (hunter_gate_error_page)
  cmake/HunterGate.cmake:345 (hunter_gate_internal_error)
  cmake/HunterGate.cmake:509 (hunter_gate_download)
  CMakeLists.txt:17 (HunterGate)

-- Configuring incomplete, errors occurred!

```

Luxonis-Brandon, 2021-06-23:
Sorry for the trouble here.  Will likely need @themarpe 's help on this.

themarpe, 2021-06-23:
Reposting the response from Community Discord:

@ans-qureshi seems like libcurl is outdated. Can you try updating it? Note: most likely offical apt of Ubuntu16 doesn't carry it, so you'll have to get it somewhere else, but updating it might cause issues for some other programs that also really on it. While not highly likely, still proceed with caution
```
wget https://curl.se/download/curl-7.77.0.tar.gz
./configure
make
sudo make install
```

The above should install the latest version. If it still won't work, some poeple recommend removing system curl `sudo apt-get remove libcurl`

Luxonis-Brandon, 2021-06-23:
Thanks, @themarpe !

closed 2021-07-12 by ans-qureshi