luxonis/depthai-core #172: 16-bit Float support

bobdavies2000, 2021-07-13:
If you clone this repo, run CMake, and build the resulting .sln, you will see an error. OpenCV has no CV_16FC1/C3 support and this code assumed it did.  If you would like to give it a try, this code with show the error when running on Windows 10.

if exist depthai-core (rmdir depthai-core /s)
“c:\Program Files\Git\bin\git.exe” clone “GitHub - luxonis/depthai-core: DepthAI C++ Library”
cd depthai-core
git submodule update --init --recursive
cd ..
“C:\Program Files\CMake\bin\Cmake.exe” -DCMAKE_CONFIGURATION_TYPES=Debug;Release -DDEPTHAI_BUILD_EXAMPLES=ON -DOpenCV_DIR=…/opencv/Build -S depthai-core -B depthai-core/Build
msbuild.exe depthai-core/Build/depthai.sln /p:Configuration=Debug
msbuild.exe depthai-core/Build/depthai.sln /p:Configuration=Release

alex-luxonis, 2021-07-13:
Could you check the OpenCV version used, DepthAI needs version 4. `CV_16FC1/C3` are available there: https://docs.opencv.org/4.5.3/d1/d1b/group__core__hal__interface.html#gac133c060dd6c36fda8e162068ef50b50
but missing in version 3 or older.

bobdavies2000, 2021-07-13:
I am using the latest OpenCV 4.5.3...

>

alex-luxonis, 2021-07-13:
That's strange, as it's available in the headers:
https://github.com/opencv/opencv/blob/4.5.3/modules/core/include/opencv2/core/hal/interface.h#L130

Maybe somehow the wrong installation is being used?

I see three dots here: `-DOpenCV_DIR=.../opencv/Build`
Looks like that's some Windows thing, equivalent to `../..` ?

bobdavies2000, 2021-07-13:
Alex - I just downloaded the latest version of the depthai-core - updated
15 hours ago.  In the link inputs for each project, there is now a
directive to include "opencv_world.lib" which breaks.  It is easily fixed
but it must be done for each project.  The CMake is putting it there but
there is no opencv_world.lib in my OpenCV build.  CMake is a great tool but
for Windows 10 developers the #pragma link input is a much better way to
control the link statements in a project in Visual Studio because it is
just included like all the other #include statements.  I have attached what
I use for my OpenCV work.  With all the projects needing the same link
statements it may help your CMake and build process.  In this case, all the
projects may be updated by editing one file.

#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_aruco453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_barcode453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_bgsegm453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_bioinspired453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_calib3d453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_ccalib453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_core453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_datasets453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_dnn453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_dnn_objdetect453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_dnn_superres453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_dpm453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_face453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_features2d453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_flann453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_fuzzy453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_gapi453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_hfs453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_highgui453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_imgcodecs453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_imgproc453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_img_hash453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_intensity_transform453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_line_descriptor453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_mcc453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_ml453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_objdetect453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_optflow453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_phase_unwrapping453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_photo453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_plot453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_quality453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_rapid453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_reg453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_rgbd453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_saliency453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_shape453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_stereo453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_stitching453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_structured_light453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_superres453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_surface_matching453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_text453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_tracking453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_video453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_videoio453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_videostab453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_wechat_qrcode453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_xfeatures2d453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_ximgproc453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_xobjdetect453.lib")
#pragma comment(lib, "OpenCV/Build/lib/Release/opencv_xphoto453.lib")

bobdavies2000, 2021-07-13:
After commenting out the 16-bit float cases in ImgFrame.cpp, the
depthai-opencv library builds correctly.  However, when the depthai-opencv
library is used in any of the other depthai-core projects - like
edge_detector - the link is now failing with the following error messages -
it is missing allocate/deallocate/error.  I bring this up because it might
shed some light on why the CV_16FC1/C3 are missing.  Could I be building
OpenCV wrong?  I am a long-time C++ user of this library and I can share
how I built OpenCV...

>edge_detector.obj : error LNK2019: unresolved external symbol "private:
char * __cdecl cv::String::allocate(unsigned __int64)" (?allocate@***@***.***@
@AEAAPEAD_K@Z) referenced in function "public: __cdecl
cv::String::String(char const *)" ***@***.***@@QEAA@PEBD@Z)
1>depthai-opencvd.lib(ImgFrame.obj) : error LNK2001: unresolved external
symbol "private: char * __cdecl cv::String::allocate(unsigned __int64)"
(?allocate@***@***.***@@AEAAPEAD_K@Z)
1>edge_detector.obj : error LNK2019: unresolved external symbol "private:
void __cdecl cv::String::deallocate(void)" (?deallocate@***@***.***@@AEAAXXZ)
referenced in function "public: __cdecl cv::String::~String(void)"
***@***.***@@QEAA@XZ)
1>depthai-opencvd.lib(ImgFrame.obj) : error LNK2001: unresolved external
symbol "private: void __cdecl cv::String::deallocate(void)"
(?deallocate@***@***.***@@AEAAXXZ)
1>edge_detector.obj : error LNK2019: unresolved external symbol "void
__cdecl cv::imshow(class cv::String const &,class
cv::debug_build_guard::_InputArray const &)" ***@***.***@@***@***.***
@***@***.******@***.***@@Z) referenced in function main
1>depthai-opencvd.lib(ImgFrame.obj) : error LNK2019: unresolved external
symbol "void __cdecl cv::error(int,class cv::String const &,char const
*,char const *,int)" ***@***.***@@***@***.***@PEBD1H@Z) referenced in
function "public: __cdecl cv::Mat::Mat(class cv::Mat &&)" ***@***.***@@QEAA@
$$QEAV01@@Z)
1>C:\_src\OpenCVB\depthai-core\Build\examples\Debug\edge_detector.exe :
fatal error LNK1120: 4 unresolved externals

themarpe, 2021-07-13:
@bobdavies2000 have you tried using the prebuilt Windows OpenCV already? Last I've tested it worked as expected (I think it was 4.5.2 though, but I think nothing this major should've changed)

bobdavies2000, 2021-07-13:
I will give that a try...Thanks...

>

bobdavies2000, 2021-07-13:
It is failing the same way with the pre-built libraries.  Same error.

>

bobdavies2000, 2021-07-13:
I have figured out the problem.  It is rooted in the CMake script used to
prepare the depthai-core tree.  Even though the script explicitly calls for
CMake to use -DOpenCV_DIR=../opencv/Build, the command line for CMake
searches nearby directories for a version of OpenCV and found an old
version without the CV_16FC1/C3 definitions.  It only became clear when
using CMake-gui and the OpenCV_Dir was incorrectly defined.  Changing it in
CMake-gui fixes the problem with the luxonis code.  Sorry for all the
trouble but it helped to get past the problem with CMake.

>

themarpe, 2021-07-13:
Thanks for letting us know. Odd that the initial definition of `-D OpenCV_DIR=../opencv/Build` didn't work. Did you apply the same path in cmake-gui or was it an absolute one in that case?

bobdavies2000, 2021-07-13:
Yes.  When I put in the same value into the CMake-gui, it works.  CMake
should not have searched for OpenCV.  I am surprised and will get to the
bottom of the problem with CMake because I need to automate the install for
the depthai-core.

>

closed 2021-09-03 by themarpe