luxonis/depthai-core #175: Breakpoints with Oak-D C++ support

bobdavies2000, 2021-07-15:
All the example code provided with depthai-core (just downloaded) is working on my Windows 10 (latest).  If I set a breakpoint anywhere, it correctly hits that breakpoint and I am able to inspect variables.  I can even single step through the code.  However, if I hit "Continue" the execution, it always fails.  The error is below.  It doesn't matter which example I choose to test with, hit a breakpoint and it is unable to continue execution.

Have you seen this before?  

Bob Davies

Unhandled exception at 0x00007FFEC0A94ED9 in edge_detector.exe: Microsoft C++ exception: std::runtime_error at memory location 0x0000000B5C6FE5C8.

The location is here in DataQueue.hpp:

    template <class T>
    std::shared_ptr<T> get() {
        if(!running) throw std::runtime_error(exceptionMessage.c_str());   <<<<<<<<<<<<<<<<<<running is false so exception 
        std::shared_ptr<ADatatype> val = nullptr; <<<<<<<<< next instruction...
        if(!queue.waitAndPop(val)) {
            throw std::runtime_error(exceptionMessage.c_str());
            return nullptr;
        }
        return std::dynamic_pointer_cast<T>(val);
    }

It also happens in tryGet, depending on the example app, with the same value for running (false.)

bobdavies2000, 2021-07-15:
I should have added that I am using the latest OpenCV 4.5.3.  Why would the running flag have gotten turned off by the breakpoint?

alex-luxonis, 2021-07-15:
This happens because of the device watchdog, that the host needs to feed periodically. It has a current timeout of 1.5s as defined here:
https://github.com/luxonis/depthai-shared/blob/7729ed1/include/depthai-shared/xlink/XLinkConstants.hpp#L17
Note the value cannot be changed directly there, as it's also used when building the firmware blob (internally by us).

That's one more reason we need to make the watchdog time configurable or have it disabled.

About the `running` flag getting unset, it's because the device resets, and reading from the XLink stream will throw an exception (that caught in `readingThread`), but makes the thread exit, and setting `running = false`.
https://github.com/luxonis/depthai-core/blob/555ce2e/src/device/DataQueue.cpp#L35-L79

bobdavies2000, 2021-07-15:
Thanks for the prompt response.  I have looked through the bios for any watchdog setting and couldn't find any.  How did you disable the watchdog timer?  I am not running the rtx64 package - just straight windows 10.  I wanted to update the file you pointed me at but you say that does nothing.  Any recommendations on how to circumvent this problem until the software gives us that ability?

alex-luxonis, 2021-07-15:
Ah, sorry, I wasn't clear. It's the watchdog enabled on the Luxonis camera (MyriadX SoC). 

We'll probably need to add some API to configure the watchdog when sending the pipeline to the camera.
So no solution for now, until this gets implemented.

Luxonis-Brandon, 2021-07-15:
(Ah, just saw Alex's response while responding)
We'll have to implement the capability to disable the watchdog that is currently-hard-baked into our firmware.  We have this on our roadmap, here:
https://github.com/orgs/luxonis/projects/4#card-63859654
And likely we will be able to start on this probably late this month or early next.

alex-luxonis, 2021-07-15:
And here is where the host keeps the watchdog alive:
https://github.com/luxonis/depthai-core/blob/main/src/device/Device.cpp#L538-L553

bobdavies2000, 2021-07-15:
Perhaps it was just wishful thinking but I tried to comment out that code -
let the main task think the watchdog is running.  Naturally it doesn't
work.  I guess I will have to just limp along until this gets resolved.
Thanks for your help.

>

themarpe, 2021-07-16:
@bobdavies2000 @alex-luxonis 
I'll add this as part of #137. Then its up to being exposed. Easiest way would be an env variable of sorts: `DEPTHAI_WATCHDOG=off` alongside API in the code. 

Note that with watchdog disabled, a device will most likely have to be power cycled / reset between runs

themarpe, 2021-09-02:
@bobdavies2000 just as a heads up - you may test this out already in `develop_embedded` branch. Set environmental variable `DEPTHAI_WATCHDOG=0` to disable watchdog. Note that device might need a power cycle / reset if application doesn't exit cleanly

vizicist, 2021-12-29:
@themarpe 
@alex-luxonis
I'd really like to try this DEPTHAI_WATCHDOG=0 feature, but I don't see the develop_embedded branch you mentioned.  I see a connect_timeout_override branch where @alex-luxonis is working on WATCHDOG things.  Is that branch usable for disabling the watchdog timeout?   If so, how can it be used?

alex-luxonis, 2021-12-29:
@vizicist The branch was merged in the meantime, now `DEPTHAI_WATCHDOG` can be used on `main`.

I also just merged to `develop` the PR https://github.com/luxonis/depthai-core/pull/312, but it's mostly for addressing some issues we had with PoE devices.

For debugging, it should be enough to set `DEPTHAI_WATCHDOG=0`, but if you're debugging the initialization sequence, it could be useful to set also the newly added (on develop for now) `DEPTHAI_BOOTUP_TIMEOUT` and `DEPTHAI_CONNECT_TIMEOUT` to some high value, to avoid a timeout.

PR #202 (2021-09-07): Embedded and other improvements
Related PR: https://github.com/luxonis/depthai-shared/pull/32

Combines following PRs for easier integration and testing of combined features:

- **Device configuration** - https://github.com/luxonis/depthai-core/pull/137
- **Bootloader** fixes and improvements - https://github.com/luxonis/depthai-core/pull/190

Updated **XLink** dependency with flash booted state handling: https://github.com/luxonis/XLink/pull/15 

Added **Backward** library to handle printing user friendly stack traces instead of plain segfaults

Closes: #175
closed 2022-03-08 by themarpe