luxonis/depthai-core #177: Pointcloud

bobdavies2000, 2021-07-16:
Is there a way to get a point cloud out of the depthai-core interface?  None of the examples showed how to do that.

Luxonis-Brandon, 2021-07-16:
So DepthAI does not generate a point cloud by default.  We have examples of using Open3D to take the depth data and project this into a point cloud, but nothing native (at least at this time).

bobdavies2000, 2021-07-16:
Brandon - the Open3D example would be great. Is it on Github?

Bob Davies

>

Luxonis-Brandon, 2021-07-16:
Looking.  I think we only have it for Python.  Not sure if C++ would be similar and/or easily portable.  CC: @saching13

saching13, 2021-07-16:
Using Open3D code would be pretty much similar.
And we don't have an official example that converts the depth to pointcloud but I can point you to some examples. 
1. You can do it easily using opencv function [here](https://docs.opencv.org/4.5.2/d9/d0c/group__calib3d.html#ga1bc1152bd57d63bc524204f21fde6e02)
2. Use PCL Library
3. [Here](https://github.com/luxonis/depthai-core/blob/84fe1e164d6b505cc00b8d82dacbaaf8026fc24d/example/main.cpp#L12) is the raw code which converts depth to PointCloud using Eigen that I had written for some internal tests.

bobdavies2000, 2021-07-17:
Sachin - thanks for the response.  I will try to use the OpenCV function
reprojectImage that you suggested.

Bob Davies

>

bobdavies2000, 2021-07-17:
Actually, the Eigen solution looks better.  I will give that one a try first....

saching13, 2021-07-17:
> Actually, the Eigen solution looks better. I will give that one a try first....

Sounds good. Just a heads up. Make sure to Cherry Pick the `DepthAlignment` class and use it on top of gen2 pipeline. 
The link contains old gen1 version of depthai.

bobdavies2000, 2021-07-18:
Sachin - what is the best way to "get_right_homography" using the latest
depthai-core API's?

Bob Davies

>

saching13, 2021-07-20:
> Sachin - what is the best way to "get_right_homography" using the latest depthai-core API's? Bob Davies
> […](#)

It needs to be done similar to the `H_left` [here](https://github.com/luxonis/depthai-python/blob/a675ac1a939f6b7fc6e02be9b27fbc2224ae4b68/examples/calibration_reader.py#L28)

Erol444, 2023-03-30:
We are adding on-device pointcloud generation support in the near future. More [information here](https://github.com/luxonis/depthai-core/issues/791).
Thanks, Erik

closed 2023-03-30 by Erol444