luxonis/depthai-core #183: [SOLVED] Shared library build: error linking on `libspdlog.a`, recompile with -fPIC

matlabbe, 2021-07-22:
Machine: Ubuntu 20.04 / ROS Noetic

Folling the build instructions, if ou do `cmake -H. -Bbuild` first, then change idea later to use shared library with `cmake -H. -Bbuild -D BUILD_SHARED_LIBS=ON`, you may get this error:
```bash
$ git submodule update --init --recursive

# Hunter's deps built static
$ cmake -H. -Bbuild
$ cmake --build build

# Hunter's deps not rebuilt for shared library
$ cmake -H. -Bbuild -D BUILD_SHARED_LIBS=ON
$ cmake --build build

[ 96%] Linking CXX shared library libdepthai-core.so
/usr/bin/ld: /home/mathieu/.hunter/_Base/062a19a/64dcd41/5051a8e/Install/lib/libspdlog.a(spdlog.cpp.o): relocation R_X86_64_TPOFF32 against `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/depthai-core.dir/build.make:796: libdepthai-core.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:513: CMakeFiles/depthai-core.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
Solution: delete hunter directory then redo cmake with `BUILD_SHARED_LIBS=ON`:
```
# cleanup dependencies, to force rebuild them
$ rm -rf ~/.hunter

# Hunter's deps are now built for shared library
$ cmake -H. -Bbuild -D BUILD_SHARED_LIBS=ON

$ cmake --build build
[ 96%] Linking CXX shared library libdepthai-core.so
[ 96%] Built target depthai-core
# should be ok now!
```

Just wanted to make an issue even if already solved in case someone else has the same problem.

closed 2021-07-22 by matlabbe

rlabs-oss, 2021-08-29:
Tried this and see "can not be used when making a shared object; recompile with -fPIC.
Attempting to use the OAK-D again with RTAB-Map. The depth-ai compiles in static build mode, rtabmap doesnt find it and cmakes as normal. I did managed to build it earlier in the year but things have changed since then.

themarpe, 2021-08-30:
@rlabs-oss can you share more details - the steps you've taken after removing the Hunter cache directory as well as any relevant output of the commands that you've executed.

rlabs-oss, 2021-08-30:
removing the hunter cache and previous build directory worked. 

[ 96%] Linking CXX shared library libdepthai-core.so
[ 96%] Built target depthai-core
Scanning dependencies of target depthai-opencv
[ 98%] Building CXX object CMakeFiles/depthai-opencv.dir/src/opencv/ImgFrame.cpp.o
[100%] Linking CXX shared library libdepthai-opencv.so
[100%] Built target depthai-opencv

The problem is after building shared libs, rtabmap is not detecting them.

themarpe, 2021-08-30:
@rlabs-oss correct, this is desired behavior. We've opted for installing to a build local directory by default instead of system directory, to not clutter the system with older versions, etc...

One can also specify the `-D CMAKE_INSTALL_PREFIX=/usr/local` which will cause the installation to be done in the specified folder. This also requires `sudo` when executing `install` target