luxonis/depthai-core #186: MinGW WinAMD64 Hunter XLink Release Unknown Error

onthegrid007, 2021-07-23:
TWIMC,

I have tried countless ways now to get this working (again). I deleted the compiled library by accident, don't ask...
Source code is at ~/Desktop/depthai-core
git cloned from THIS repo 20min ago with --init and --recursive as flags

I have tried cleaning dirs to see if there was some old random toolchain-related code anywhere.
I'm using GitBashPortable on Windows so I have cleaned all the following directories:
/c/.hunter
~/.hunter
%temp%
prefetch

I even for once tried to follow the given comp. instructions in the README.txt
Nothing.
I am stuck.
It won't even give the make error. mingw32-make just crashes with Error: 1 when hunter is compiling XLink-Release.

Here is the terminal log:

```
LocalAdmin@COMPUTER MINGW64 ~/Desktop/depthai-core (main)
$ mkdir win64-build

LocalAdmin@COMPUTER MINGW64 ~/Desktop/depthai-core (main)
$ cd win64-build/

LocalAdmin@COMPUTER MINGW64 ~/Desktop/depthai-core/win64-build (main)
$ cmake .. -G "MinGW Makefiles"
-- [hunter] Initializing Hunter workspace (062a19ab13ce8dffa9a882b6ce3e43bdabdf75d3)
-- [hunter]   https://github.com/cpp-pm/hunter/archive/v0.23.258.tar.gz
-- [hunter]   -> C:/Users/LocalAdmin/.hunter/_Base/Download/Hunter/0.23.258/062a19a
-- The CXX compiler identification is GNU 8.1.0
-- The C compiler identification is GNU 8.1.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/mingw-w64/x86_64-8.1.0-posix-sjlj-rt_v6-rev0/mingw64/bin/g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/mingw-w64/x86_64-8.1.0-posix-sjlj-rt_v6-rev0/mingw64/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Setting build type to 'Debug' as none was specified.
-- Found Git: C:/Program Files/Git/mingw64/bin/git.exe (found version "2.32.0.windows.1")
-- [hunter] Calculating Toolchain-SHA1
-- [hunter] Calculating Config-SHA1
-- [hunter] HUNTER_ROOT: C:/Users/LocalAdmin/.hunter
-- [hunter] [ Hunter-ID: 062a19a | Toolchain-ID: cbb7500 | Config-ID: b63ed45 ]
-- [hunter] NLOHMANN_JSON_ROOT: C:/Users/LocalAdmin/.hunter/_Base/062a19a/cbb7500/b63ed45/Install (ver.: 3.9.1)
-- [hunter] Building nlohmann_json
loading initial cache file C:/Users/LocalAdmin/.hunter/_Base/062a19a/cbb7500/b63ed45/cache.cmake
loading initial cache file C:/Users/LocalAdmin/.hunter/_Base/062a19a/cbb7500/b63ed45/Build/nlohmann_json/args.cmake
-- The C compiler identification is GNU 8.1.0
-- The CXX compiler identification is GNU 8.1.0
-- Check for working C compiler: C:/Program Files/mingw-w64/x86_64-8.1.0-posix-sjlj-rt_v6-rev0/mingw64/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files/mingw-w64/x86_64-8.1.0-posix-sjlj-rt_v6-rev0/mingw64/bin/g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/LocalAdmin/.hunter/_bin/vnkJA/nlohmann_json
[  6%] ←[34m←[1mCreating directories for 'nlohmann_json-Release'←[0m
[ 12%] ←[34m←[1mPerforming download step (download, verify and extract) for 'nlohmann_json-Release'←[0m
-- Downloading...
   dst='C:/Users/LocalAdmin/.hunter/_Base/Download/nlohmann_json/3.9.1/f8a20a7/v3.9.1.tar.gz'
   timeout='none'
[... 258 lines cut ...]
mingw32-make.exe[2]: *** [CMakeFiles\XLink-Release.dir\build.make:85: XLink-Release-prefix/src/XLink-Release-stamp/XLink-Release-build] Error 2
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:84: CMakeFiles/XLink-Release.dir/all] Error 2
mingw32-make.exe: *** [Makefile:90: all] Error 2

[hunter ** FATAL ERROR **] Build step failed (dir: C:/Users/LocalAdmin/.hunter/_Base/062a19a/cbb7500/b63ed45/Build/XLink
[hunter ** FATAL ERROR **] [Directory:C:/Users/LocalAdmin/Desktop/depthai-core/cmake]

------------------------------ ERROR -----------------------------
    https://hunter.readthedocs.io/en/latest/reference/errors/error.external.build.failed.html
------------------------------------------------------------------

CMake Error at C:/Users/LocalAdmin/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_error_page.cmake:12 (message):
Call Stack (most recent call first):
  C:/Users/LocalAdmin/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_fatal_error.cmake:20 (hunter_error_page)
  C:/Users/LocalAdmin/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_download.cmake:623 (hunter_fatal_error)
  C:/Users/LocalAdmin/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_add_package.cmake:53 (hunter_download)
  cmake/depthaiDependencies.cmake:9 (hunter_add_package)
  CMakeLists.txt:111 (include)

-- Configuring incomplete, errors occurred!
See also "C:/Users/LocalAdmin/Desktop/depthai-core/win64-build/CMakeFiles/CMakeOutput.log".
```

Am I just losing my mind right now or is there actually something broken...
Help is greatly appreciated.

alex-luxonis, 2021-07-23:
Seems to be a problem when using MinGW to build, potentially introduced by the Ethernet/PoE support in `2.7.0.0`:
```
C:\Users\LocalAdmin\.hunter\_Base\062a19a\cbb7500\b63ed45\Build\XLink\Source\pc\protocols\tcpip_host.c:268:18: error: conflicting types for 'tcpip_close_socket'
 tcpipHostError_t tcpip_close_socket(SOCKET sock)
                  ^~~~~~~~~~~~~~~~~~
...
C:\Users\LocalAdmin\.hunter\_Base\062a19a\cbb7500\b63ed45\Build\XLink\Source\pc\protocols\tcpip_host.h:73:18: note: previous declaration of 'tcpip_close_socket' was here
 tcpipHostError_t tcpip_close_socket(int sockfd);
 ```
I think we only tested building with Visual Studio.
CC: @themarpe

themarpe, 2021-07-23:
@onthegrid007 
Looks like MinGW properly differentiates between `SOCKET` and `int` (or just some ifdefs not working correctly under MinGW).
 
Officially we only support MSVC. That said I did some support for MinGW in the past and if you are willing to create a PR to XLink repository with needed fixes, I'll gladly take a look at it.

Easiest way for local development is to point depthai core to a cloned directory of XLink by setting a CMake variable `-D DEPTHAI_XLINK_LOCAL=[path/to/xlink]`

wstnturner, 2021-11-27:
I've made a PR for allowing XLink to build under MinGW (x64): https://github.com/luxonis/XLink/pull/24

When this XLink project is referenced by depthai-core (as specified above, e.g., -D DEPTHAI_XLINK_LOCAL=../XLink), The Entirety of the core library can be built under MinGW, for example:

```
cmake -S. -Bbuild -D BUILD_SHARED_LIBS=ON -D DEPTHAI_XLINK_LOCAL=../XLink  -D DEPTHAI_ENABLE_BACKWARD=OFF -G "MinGW Makefiles"
cmake --build build
```

Caveat, DEPTHAI_ENABLE_BACKWARD=OFF must be set to prevent other windows'isms from intruding into the build.

PR #290 (2021-12-02): Updated XLink with MinGW fixes
Closes: #186
closed 2021-12-03 by themarpe