luxonis/depthai-core #208: Use of imagemanip operations with the regular video stream (still / video) results in fatal error

oysstu, 2021-09-06:
Hi,

Using the ImageManipulation node with the video/still stream results in the following error:
`Fatal error. Please report to developers. Log: 'ImageManipHelper' '60'`

This occurs even if the ImgFrame::Type is set to NV12

Best regards,
oysstu

themarpe, 2021-09-06:
Hi @oysstu 
Sorry for the inconvenience. ImageManip doesn't (yet) support NV12 input. 
We are working on it and hope to have a fix for it soon.

themarpe, 2021-12-27:
@oysstu `image_manip_refactor` branch now includes improvements to ImageManip which now handle NV12 as input as well.
Along that it also supports color space conversion, so one can input other formats as well and produce NV12 output (as well as others, RGB,BGR,planar,interleaved,gray,...)

Note: due to hw restrictions and implementation detail, inputting planar formats is faster than interleaved (eg. YUV420 vs NV12, BGR planar vs BGR interleaved, etc...)
But converting to planar/interleaved doesn't make a difference. 
(Above could sometimes be the reason to use `isp` instead of `video` output for some operations)

closed 2022-08-09 by oysstu

oysstu, 2022-08-09:
@themarpe I've closed this, seeing as it's implemented and merged already. With respect to what you wrote above; how to best judge which output stream to use for imagemanip? Let's say that the imagemanip node is a combination of cropping, resizing and for example conversion to GRAY8.

themarpe, 2022-08-09:
@oysstu 
For ColorCamera that'd be `isp` over `video`. `video` is NV12 (interleaved), while `isp` output is YUV420 (planar).

Also, currently only `still` and `video` are interleaved (or `preview` if set to interleaved). So you may compute across pipeline in planar formats, until needed (eg for VideoEncoder) and only then convert to NV12