luxonis/depthai-core #242: [Feature Request] Support for NN models with non-square resolutions

OmarALFA, 2021-11-02:
Attempting to use a detection network with a non-square aspect ratio (e.g. 640x352) produces the following error: 

`[14442C1041155FD700] [255.697] [system] [critical] Fatal error. Please report to developers. Log: 'PlgDetectionParser' '108'`

Luxonis-Martin on the Discord explained that it's not currently supported, but that I can open an issue to request it. 

Thanks and best regards.

Luxonis-Brandon, 2021-11-02:
Non-square NN resolutions are supported.  There must be some confusion in communication on Discord here; sorry about that.

Examples of a variety of non-square models are used here via the API:
[https://docs.luxonis.com/en/latest/pages/tutorials/pretrained_openvino/#trying-other-models](https://docs.luxonis.com/en/latest/pages/tutorials/pretrained_openvino/#trying-other-models)

For example `face-detection-adas-0001` there is `384x672` resolution.

szabi-luxonis, 2021-11-02:
> Non-square NN resolutions are supported. There must be some confusion in communication on Discord here; sorry about that.
>
> Examples of a variety of non-square models are used here via the API: [...]

He is referring to parsing for non-square yolo output is not supported on-device.

OmarALFA, 2021-11-03:
> Non-square NN resolutions are supported. There must be some confusion in communication on Discord here; sorry about that.
>
> Examples of a variety of non-square models are used here via the API: [...]

My apologies for the ambiguity. 

I've edited this issue with more details and to reflect @szabi-luxonis's clarification.

szabi-luxonis, 2021-11-05:
> My apologies for the ambiguity.
>
> I've edited this issue with more details and to reflect @szabi-luxonis's clarification.

Is it possible to share the model? Would help validating parsing support for non-squared model.
If not, sharing the parsing code on host is equally helpful.

szabi-luxonis, 2021-11-07:
Tested w/ network/dataset provided by @OmarALFA
640x352 YOLO network:

[image]

OmarALFA, 2021-11-08:
I built the new library off [develop](https://github.com/luxonis/depthai-core/tree/develop). Can confirm this resolves the issue on my end! Non-square yolov4-tiny NN no longer produces the error! 

Note, that you will need to adjust the anchor masks depending on the NN resolution, in the case of 640x352: 

```cpp
detectionNetwork->setAnchorMasks({ {"side40", {0, 1, 2}}, {"side20", {3, 4, 5}} });
``` 

For other resolutions, the formula is as follows: 

1. with width = inputWidth/16
2. with width = inputWidth/32

e.g.
640/16 = 40 --> "side40" 
640/32 = 20 --> "side20"

This applies to tiny yolo. For full yolo, there will be a third:
3. with width = inputWidth/8

These notes were provided by @szabi-luxonis. Thank you!

closed 2021-11-08 by OmarALFA