luxonis/depthai-core #244: vcpkg integration for ease of use

saulthu, 2021-11-06:
It'd be great if this could be added to vcpkg, then it's very easy to build and use on Windows e.g. use a vcpkg manifest file in Visual Studio to add the dependency, then all include paths and linking is automatic. I could try to make this, but I'm not good with cmake.

themarpe, 2021-11-07:
Hi @saulthu - I agree, having the ability to add DepthAI as dependency using vcpkg would ease integration quite a lot. I've looked into the way how vcpkg handles packages and it seems like it should be possible without much issues. 

The steps I'd take on creating the vcpkg package for it would be to build and install DepthAI using vcpkg and use `find_package` to add it. 

There might be some issues regarding DepthAI dependencies, as in this case, they would be build by Hunter in the background and could clash with user libraries. For example using ZLib in your project could clash with internal ZLib from DepthAI. There are 2 options to handle this case
 1. Ensure that this kind of dependencies are the same, and share them between application and library (one can check `cmake/Hunter/config.cmake` to see which these are)
 2. Make visibility of this kind of dependencies hidden, to ensure a "self-contained" DepthAI library, which isn't influenced by use of external libraries.

I'd lean towards option 2, as apart from slightly bigger library size, it ensures that dependencies used are same and tested against. Also being an opensource library, I don't think it hurts repairability/maintainability as one can rebuild with different dependencies without having to rely on `LD_PRELOAD` or modifying used shared libraries. (Link to case where this issue occured: https://github.com/luxonis/depthai-core/pull/239#issuecomment-952349184)

**TLDR**
So I think the way to go is to take a "similar" library that handles CMake install well (like: https://github.com/microsoft/vcpkg/tree/master/ports/nlohmann-json) and just reuse its main parts to create a port for DepthAI. This should be adequate for start - the above mentioned dependency issues can be resolved later if they show up at all.

If you'd be up to testing this locally, seeing if it works well and doesn't present any issues with the rest of vcpkg, that'd be of much help! Also feel free to ask any questions on this topic or if you encounter any issues.

saulthu, 2021-11-07:
Hey @themarpe, sounds like you know a lot more about the requirements that I do. I do think it'd be nice, if possible, to use the existing vcpkg libs e.g. opencv, zlib, etc., since they're already ported to vcpkg, and use them as vcpkg dependencies for depthai. I hear you that only certain versions are tested against though.

saulthu, 2021-11-07:
I'm very willing to test out versions on Windows with vcpkg and Visual Studio.

themarpe, 2021-11-08:
Thanks @saulthu! 
In that case, could you create a minimum working port for DepthAI and for starters disabling OpenCV support (so opencv isn't required as a dependency), so it should essentially boil down to the above mentioned port. 

And afterwards I'll tinker with getting OpenCV and other public dependencies (nlohmann json & libnop) to be managed by vcpkg instead of Hunter.