luxonis/depthai-core #246: fails to cmake config+build on Windows using cmake and vscode (hunter pain)

diablodale, 2021-11-08:
depthai-core fails to config (and therefore can't build) on a clean repo and using VSCode.
Cascading failures in hunter.

### Setup

* Microsoft Windows [Version 10.0.19043.1288]
* VS2019 specifics below
* depthai-core main (currently it is c1b697afa432c8d620b097ba57e20a0ffd2ec52d)

### Repro

1. install VS2019 community 16.11.5 with: Desktop Development c++, linux dev with c++. In installation details for Desktop Development include C++ CMake tools for Windows.
2. install vscode x64 for windows
3. install git for windows
4. `git clone https://github.com/luxonis/depthai-core` into an empty folder.
5. `git checkout c1b697afa432c8d620b097ba57e20a0ffd2ec52d` so to have less variations in the setup.
6. run VSCode
7. Add the extensions for: C/C++, CMake Tools. Both of these are made by Microsoft. Restart vscode if prompted.
8. File/open the `depthai-core` folder. Select `yes` to trust.
9. At bottom right, select `NOT NOW` and `NOT THIS WORKSPACE` at the configure project popup.
10. At bottom center, you should see "No Kit Selected". Click that and select "Visual Studio Community 2019 Release - amd64"
11. Bottom center should update with no errors to show "Visual Studio Community 2019 Release - amd64"
12. Bottom left should show the checked out commit c1b697af
13. Note, since `BUILD_SHARED_LIBS` is not declared anywhere, it should be a static library build
14. in the left nav, click the CMake icon. At the top of the left nav beside the "CMAKE: PROJECT OUTLINE" click the first icon for Configure All Projects.

Result

```
[main] Configuring folder: depthai-core 
[proc] Executing command: "C:\Program Files\CMake\bin\cmake.exe" --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -Hc:/repos-nobackup/depthai-core -Bc:/repos-nobackup/depthai-core/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The CXX compiler identification is MSVC 19.29.30136.0
[cmake] -- The C compiler identification is MSVC 19.29.30136.0
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Found Git: C:/njs/wslgit/git.EXE (found version "2.33.1") 
[cmake] -- [hunter] Calculating Toolchain-SHA1
[cmake] -- [hunter] Calculating Config-SHA1
[cmake] -- [hunter] HUNTER_ROOT: C:/.hunter
[cmake] -- [hunter] [ Hunter-ID: 062a19a | Toolchain-ID: 07acd96 | Config-ID: 05fa1a3 ]
[cmake] 
[cmake] [hunter ** FATAL ERROR **] Incorrect MSVC setup:
[cmake] -- Configuring incomplete, errors occurred!
[cmake] See also "C:/repos-nobackup/depthai-core/build/CMakeFiles/CMakeOutput.log".
[cmake] [hunter ** FATAL ERROR **]   At least one of the following should be an absolute path
[cmake] [hunter ** FATAL ERROR **]   CMAKE_VS_DEVENV_COMMAND:()
[cmake] [hunter ** FATAL ERROR **]   CMAKE_VS_MSBUILD_COMMAND:()
[cmake] [hunter ** FATAL ERROR **] [Directory:C:/repos-nobackup/depthai-core/cmake]
[cmake] 
[cmake] ------------------------------ ERROR -----------------------------
[cmake]     https://hunter.readthedocs.io/en/latest/reference/errors/error.vs.devenv.html
[cmake] ------------------------------------------------------------------
[cmake] 
[cmake] CMake Error at C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_error_page.cmake:12 (message):
[cmake] Call Stack (most recent call first):
[cmake]   C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_fatal_error.cmake:20 (hunter_error_page)
[cmake]   C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_setup_msvc.cmake:148 (hunter_fatal_error)
[cmake]   C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_finalize.cmake:117 (hunter_setup_msvc)
[cmake]   C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_add_package.cmake:23 (hunter_finalize)
[cmake]   cmake/depthaiDependencies.cmake:7 (hunter_add_package)
[cmake]   CMakeLists.txt:118 (include)
[... 87 lines cut ...]
[cmake] 
[cmake] 
[cmake] FAILED: libarchive-luxonis-Release-prefix/src/libarchive-luxonis-Release-stamp/libarchive-luxonis-Release-configure 
[cmake] cmd.exe /C "cd /D C:\.hunter\_bin\veis2\libarchive-luxonis\libarchive-luxonis-Release-prefix\src\libarchive-luxonis-Release-build && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -GNinja -CC:/.hunter/_Base/062a19a/07acd96/05fa1a3/cache.cmake -CC:/.hunter/_Base/062a19a/07acd96/05fa1a3/Build/libarchive-luxonis/args.cmake -DCMAKE_RELEASE_POSTFIX= -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=C:/.hunter/_Base/062a19a/07acd96/05fa1a3/Build/libarchive-luxonis/Install -DCMAKE_TOOLCHAIN_FILE=C:/.hunter/_Base/062a19a/07acd96/05fa1a3/Build/libarchive-luxonis/toolchain.cmake -DCMAKE_MAKE_PROGRAM=C:/PROGRA~2/MICROS~2/2019/COMMUN~1/Common7/IDE/COMMON~1/MICROS~1/CMake/Ninja/ninja.exe -GNinja C:/.hunter/_Base/062a19a/07acd96/05fa1a3/Build/libarchive-luxonis/Source && "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" -E touch C:/.hunter/_bin/veis2/libarchive-luxonis/libarchive-luxonis-Release-prefix/src/libarchive-luxonis-Release-stamp/libarchive-luxonis-Release-configure"
[cmake] ninja: build stopped: subcommand failed.
[cmake] 
[cmake] [hunter ** FATAL ERROR **] Build step failed (dir: C:/.hunter/_Base/062a19a/07acd96/05fa1a3/Build/libarchive-luxonis
[cmake] [hunter ** FATAL ERROR **] [Directory:C:/repos-nobackup/depthai-core/cmake]
[cmake] 
[cmake] ------------------------------ ERROR -----------------------------
[cmake]     https://hunter.readthedocs.io/en/latest/reference/errors/error.external.build.failed.html
[cmake] ------------------------------------------------------------------
[cmake] 
[cmake] CMake Error at C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_error_page.cmake:12 (message):
[cmake] Call Stack (most recent call first):
[cmake]   C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_fatal_error.cmake:20 (hunter_error_page)
[cmake]   C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_download.cmake:623 (hunter_fatal_error)
[cmake]   C:/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/modules/hunter_add_package.cmake:53 (hunter_download)
[cmake]   cmake/depthaiDependencies.cmake:13 (hunter_add_package)
[cmake]   CMakeLists.txt:118 (include)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[cmake] See also "C:/repos-nobackup/depthai-core/build/CMakeFiles/CMakeOutput.log".
```

themarpe, 2021-11-09:
@diablodale does the build work correctly if done outside VSCode? (eg. following specified build commands in README.md)

Seems like an environment issue on first glance.

diablodale, 2021-11-10:
Testing...

diablodale, 2021-11-11:
I isolated and fixed the cmake harness of depthai-core to enables building depthai-core from vscode or command line.
I'll make a PR.

Errant things included depth-ai's cmake code, a lack-of-features, related #72, and/or due to opencv's inconsistent transitive library declarations.

### How to use

Hunter is limited and has to be told about the build environment directly -- it doesn't respect or infer it from a parent environment like VSCode. This is done by specifying the `CMAKE_VS_MSBUILD_COMMAND` cmake variable in custom toolchain file (because Hunter ***only*** respects a toolchain file).

Quick howto for vscode

1. create a toolchain file, e.g `my-toolchain.cmake`
2. In it set the build environment, surounding with an `if()` because of Hunter breakage as related to `project()`
   ```cmake
   if(NOT DEFINED HUNTER_MSVC_ARCH)
       set(CMAKE_VS_MSBUILD_COMMAND "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe")
   endif()
   ```
3. Put any other things in your toolchain you might want like compiler flags, shared library or not, CMAKE_INSTALL_PREFIX, etc.
4. set the toolchain file in `.vscode/settings.json`
   ```json
   "cmake.configureSettings": {
       "CMAKE_TOOLCHAIN_FILE": "${workspaceFolder}/my-toolchain.cmake",
   }
   ```

Because of more Hunter pain, you can NOT `find_package()` in the toolchain file. If you need this for test, examples, etc. to build, then you can set the cmake variable `DEPTHAI_DEPENDENCY_INCLUDE` in the toolchain. And later in the configure process when it is safer, the cmake infrastructure will read that var and include it as a cmake file. Here is an example of another line within your toolchain.

```cmake
if(NOT DEFINED HUNTER_MSVC_ARCH)
    set(CMAKE_VS_MSBUILD_COMMAND "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/MSBuild/Current/Bin/MSBuild.exe")
endif()
set(DEPTHAI_DEPENDENCY_INCLUDE "${CMAKE_SOURCE_DIR}/cmake/my-extra-include.cmake")
```

closed 2021-11-22 by szabi-luxonis