luxonis/depthai-core #249: `noexcept` declarations incorrect or missing -- leads to crashes, code bloat, obscuring code behavior

diablodale, 2021-11-09:
The codebase and its dependencies have a range of exception declaration issues

* declared `noexcept` yet calling APIs that can throw exceptions -- this is unsafe and can lead to crashes
* not declared `noexcept` yet not calling any APIs that can throw exceptions -- this is inefficient, can lead to code bloat for compiler to include handling code not actually needed, cascades up the call chain requiring all callers in chain to handle exceptions that will never occur, etc.
* Default constructor may not throw. Declare it 'noexcept' (f.6) -- this is probably handled silently by the compiler but not declaring it as such obscures actual code behavior and therefore coders, editors, etc. don't provide as good code assistance as possible. And has the same cascading caller chain errors of bullet 2.

The top offender is `dependencies\include\nlohmann`. But I can also see them in Luxonis code like `depthai-shared\common\Rect.hpp`.

### Setup

* Microsoft Windows [Version 10.0.19043.1288]
* MSVS 2019 Community v16.11.5 (though since these are C++ language issues, is likely present in other compilers)
* depthai-core windows prebuilt distribution `depthai-core-v2.11.1-win64.zip`. There are likely issues beyond those attached as I have only used the precompiled Windows distribution (can't yet full compile due to other open depthai issues).

### Repro

1. Enable warnings and code analyzer e.g. in MSVC it is with `/W4  /analyze:plugin EspXEngine.dll` and env var `esp.extensions=cppcorecheck.dll' https://docs.microsoft.com/en-us/cpp/code-quality/using-the-cpp-core-guidelines-checkers?view=msvc-170
2. Compile an app that uses the `depthai-core-v2.11.1-win64` and simple code that cascades to headers like `common/Rect.hpp` or `pipeline/Node.hpp`. Most any code, even just a dai device, will do this.

### Result

88+ errors of exception handling issues. Here is a sampling
[depthai-noexcept.txt](https://github.com/luxonis/depthai-core/files/7507958/depthai-noexcept.txt)

### Expected

Zero warnings of `C26447: The function is declared 'noexcept' but calls function 'xxxx()' which may throw exceptions (f.6).`
These can lead to crashes.

It is preferred to not have the other types of warnings. They should not cause crashes, but they do lead to code bloat or incorrect/lack of clarity on code behavior for the developer.

themarpe, 2021-11-09:
@diablodale all of the mentioned issues `which may throw exceptions` are coming from nlohmann json library.
It seems like that is the case, and there are some issues with it, but checking out the first issue reveals the following:
```
class exception : public std::exception
{
  public:
    /// returns the explanatory string
    JSON_HEDLEY_RETURNS_NON_NULL
    const char* what() const noexcept override
    {
        return m.what();
    }

    /// the id of the exception
    const int id;

  protected:
    JSON_HEDLEY_NON_NULL(3)
    exception(int id_, const char* what_arg) : id(id_), m(what_arg) {}

    static std::string name(const std::string& ename, int id_)
    {
        return "[json.exception." + ename + "." + std::to_string(id_) + "] ";
    }

  private:
    /// an exception object as storage for error messages
    std::runtime_error m;
};
```

The function in question being `what()` - it calls its own member `m`, which is of type `std::runtime_error`.

https://en.cppreference.com/w/cpp/error/runtime_error

Docs show that `std::runtime_error::what()` is indeed noexcept since C++11. 

Are you compiling with C++14 set? Otherwise this seems to be either a bug in the anaylizer (more likely) or in Microsoft STL (less likely)

themarpe, 2021-11-09:
Also we are currently still using `nlohmann json 3.9.1`, where `3.10.4` is already out. It might be that some issue were fixed by a later version. Would be nice giving this a try by just swapping out the mentioned header for this one: https://github.com/nlohmann/json/releases/download/v3.10.4/json.hpp

diablodale, 2021-11-09:
My app is compiled c++17 as I use features like `std::optional`

For the specific `what()` issue...MSVC itself has a bug about `what()` comformance and now are dealing with the legacy/vtable issues of it. https://developercommunity2.visualstudio.com/t/VC:-std::exception::what-has-wrong-sig/785027
Their suggestion for that specific `what` bug is
> Suppressing the Core Guidelines warning C26447 is the most reasonable workaround, I think

themarpe, 2021-11-09:
I've checked the latest nlohmann version about the second and third issue that came up:
```
include\nlohmann\json.hpp(4801) : warning C26447: The function is declared 'noexcept' but calls function 'rdstate()' which may throw exceptions (f.6).
include\nlohmann\json.hpp(4801) : warning C26447: The function is declared 'noexcept' but calls function 'clear()' which may throw exceptions (f.6).
```

Which seems to be calling a function that may throw in a destructor. And some other cases. 

These should be raised to nlohmann json, as `noexcept` that can throw is indeed unsafe.

For points 2. and 3. on our codebase, I wouldn't go about them unless some sort of gain is to be had there. 
The performance gains available to be had in our library by introducing `noexcept` are essentially 0. While there is work to do it and maintain it properly (which can more easily than not cause the reverse issues of having a `noexcept` where it shouldn't be).

As the drawbacks drastically outweigh the benefits, I wouldn't pursue this changes.

I'd only see about updating the `nlohmann json` library if they address this on their end.

diablodale, 2021-11-12:
#### Point 1 (crashes)
I request your team escalate the unsafe/crashing scenarios to nlohmann. Your team is the right role to escalate issue or PR as you are the consumer/user. I'm happy to +1 or comment on any issue/PR you send.

#### Point 2 (efficiency)
Ok

#### Point 3 (restrictive, unclear)
I disagree here for code your team controls and is public to users (like Point3f).
https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#c44-prefer-default-constructors-to-be-simple-and-non-throwing
...and in common use cases like making a class that has a member of std::vector<dai::Point3f> causes a chain of forced assert handling on default construct of that class (which cascades to default construct of vector then dai::Point3f when not necessary and sometimes difficult to wrap try/catches in places.

I request your team investigate the 4 so far found. And it is probable even better to write `= default` instead of `{}` so that the default constructor gains additional abilities/qualifications. And this relates also to https://github.com/luxonis/depthai-core/issues/251

```
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\common\Point3f.hpp(17) : warning C26455: Default constructor may not throw. Declare it 'noexcept' (f.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\common\Point2f.hpp(17) : warning C26455: Default constructor may not throw. Declare it 'noexcept' (f.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\common\Size2f.hpp(17) : warning C26455: Default constructor may not throw. Declare it 'noexcept' (f.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\common\Rect.hpp(20) : warning C26455: Default constructor may not throw. Declare it 'noexcept' (f.6).
```

themarpe, 2021-11-16:
I agree on Point 3 regarding non default constructors.
We'll update this.

Will also post my findings to nlohmann.

themarpe, 2021-11-16:
Rechecking the nlohmann warnings, and doing a repro in godbolt - it seems that there are many functions with this warning. Even in various std libraries, etc...
https://godbolt.org/z/KoKWG4qP6

I'm actually not sure if this warning is actually relevant if library ensures that the function won't throw (eg. by knowing the state of underlying function, etc...)

So I'll skip on raising this as an issue.

diablodale, 2021-11-16:
Are you swapping issues? The godbolt you provides shows mostly `can be declared 'noexcept'`
Those are not our nlohmann discussion.
Our discussion for nlohmann is the opposite.   That something is declared noexcept but actually DOES risk exceptions which can lead to crashes.
80 out of 1750 godbolt warnings/errors are the potential crash locations
21 of those 80 are nlohmann

Some amount of the msvc warnings of potential crashes are due to template parameters given to them, and cascades of nlohmann usage. Only coed analysis by nlohmann can surface the actual case.

Microsoft has bugs in their STL which I link above. Dependent libraries can change. etc. Our roles in library providers (you depthai, microsoft stl, nlohmann json, etc.) is to continue the progress for more accurate, less crashing, better documented code.

FYI, I always compile W4 and analyzer and do not see widespread STL issues like in this godbolt. It is due to nlohmann's usage of the STL in an "unsafe" manner that is cascading all these warnings.

themarpe, 2021-11-16:
The latest example you've posted was fixed in latest (3.10.4) release. The function isnt' `noexcept` anymore.

What I meant is that for the time being I won't be diving into nlohmann lib about their `noexcept` usage, as its not a pressing matter.
I do believe however that there are cases where things can break. Fortunately this doesn't immediately affects us currently.

Regarding the `noexcept` DepthAI warnings, will address this. 
Also any recommendations for similar flags that raise `noexcept` warnings for Clang & GCC?

diablodale, 2021-11-17:
ok. `main` still shows nlohmann 3.9.1. I'll ignore the json issues for now and when/if I notice depthai incorporates 3.10.4, then I'll open a separate issue on the newer version as needed.

I rarely build for linux so I'm not a wiz for switches. Here's some chatter on it https://stackoverflow.com/questions/59676462/enabling-warnings-when-a-noexcept-function-attempts-to-call-non-noexcept-functio

There are some switches for the MSVC analyzer that can be useful to remove noise while getting deep analysis.
https://docs.microsoft.com/en-us/cpp/code-quality/using-the-cpp-core-guidelines-checkers?view=msvc-170#non-msbuild-projects

* compiler options `/analyze:plugin EspXEngine.dll` to enable it
* env var (set it for the cmake build environ) `esp.extensions=CppCoreCheck.dll`
* env var(cmake build) `caexcludepath=C:\Program Files;C:\Program Files (x86)` or wherever the MSVC, WinSDK, and STL are located

themarpe, 2021-11-23:
Addressed in https://github.com/luxonis/depthai-shared/pull/65

closed 2021-11-23 by themarpe