luxonis/depthai-core #250: several depthai variables are uninitialized -- may/not lead to coding errors or crashes

diablodale, 2021-11-09:
In the precompiled depthai windows distribution, I have found 17 cases of variables not being initialized. This leaves those variables in an undefined state with unknown data in them. This may be ok, or may be bad. Only a code review can discern.

The fix is to always initialize variables to a known state.

### Setup

* all compilers and OS
* Seen in the headers of precompiled Windows distrib:  `depthai-core-v2.11.1-win64`. I have not analyzed the entire codebase as I can't yet compile the whole depthai SDK.

### Repro

1. Enable warnings and code analyzer e.g. in MSVC it is with `/W4  /analyze:plugin EspXEngine.dll` and env var `esp.extensions=cppcorecheck.dll' https://docs.microsoft.com/en-us/cpp/code-quality/using-the-cpp-core-guidelines-checkers?view=msvc-170
2. Compile an app that uses the `depthai-core-v2.11.1-win64` and simple code that cascades to headers like `common/Rect.hpp` or `pipeline/Node.hpp`. Most any code, even just a dai device, will do this.

### Result

In my case of using just the headers, I found 17 failures to initialize variables. Most are in Luxonis code.

```
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\properties\IMUProperties.hpp(164) : warning C26495: Variable 'dai::IMUSensorConfig::sensorId' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawNNData.hpp(43) : warning C26495: Variable 'dai::TensorInfo::numDimensions' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawNNData.hpp(43) : warning C26495: Variable 'dai::TensorInfo::order' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawNNData.hpp(43) : warning C26495: Variable 'dai::TensorInfo::dataType' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawNNData.hpp(43) : warning C26495: Variable 'dai::TensorInfo::offset' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawSpatialLocations.hpp(46) : warning C26495: Variable 'dai::SpatialLocations::depthMin' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawSpatialLocations.hpp(46) : warning C26495: Variable 'dai::SpatialLocations::depthAverage' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawSpatialLocations.hpp(46) : warning C26495: Variable 'dai::SpatialLocations::depthMax' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawSpatialLocations.hpp(46) : warning C26495: Variable 'dai::SpatialLocations::depthAveragePixelCount' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawTracklets.hpp(55) : warning C26495: Variable 'dai::Tracklet::age' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawTracklets.hpp(55) : warning C26495: Variable 'dai::Tracklet::srcImgDetection' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawTracklets.hpp(55) : warning C26495: Variable 'dai::Tracklet::id' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawTracklets.hpp(55) : warning C26495: Variable 'dai::Tracklet::status' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawTracklets.hpp(55) : warning C26495: Variable 'dai::Tracklet::label' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\include\depthai-shared\datatype\RawTrackedFeatures.hpp(41) : warning C26495: Variable 'dai::TrackedFeature::id' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\lib\cmake\depthai\dependencies\include\nlohmann\json.hpp(18509) : warning C26495: Variable 'nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer,std::vector<unsigned char,std::allocator<unsigned char> > >::m_type' is uninitialized. Always initialize a member variable (type.6).
[build] C:\repos-nobackup\depthai-core-v2.11.1-win64\lib\cmake\depthai\dependencies\include\nlohmann\json.hpp(18509) : warning C26495: Variable 'nlohmann::basic_json<std::map,std::vector,std::basic_string<char,std::char_traits<char>,std::allocator<char> >,bool,__int64,unsigned __int64,double,std::allocator,nlohmann::adl_serializer,std::vector<unsigned char,std::allocator<unsigned char> > >::m_value' is uninitialized. Always initialize a member variable (type.6).

```

### Expected

No errors of uninitialized variables.

szabi-luxonis, 2021-11-09:
Thanks for the report.
We have [clang-tidy](https://github.com/luxonis/depthai-core/runs/4133173505?check_suite_focus=true) as part of CI, but we didn't really get into solving the warnings reported.
If you are willing to contribute, pull requests are welcome, otherwise,  we will get into these in the shortest time possible.
In this case, the warnings aren't harmful, since these are data received from device which are always initialized.

diablodale, 2021-11-09:
To verify then...all 15 of the variables in Luxions code are guaranteed to be set/initialized by other/internal code before any possibility of them being read?

If this is true via code review, then I can workaround/ignore this specific code error on those specific files to move forward with a compile.

I won't be able to fix these in any reasonable time period myself as I would need to understand the corpus of depthai. At the moment, I am focused on getting compiled/working alpha before the end of the year and OAK-D-lite ships.

szabi-luxonis, 2021-11-09:
> To verify then...all 15 of the variables in Luxions code are guaranteed to be set/initialized by other/internal code before any possibility of them being read?

Yes, you can safely ignore it.

diablodale, 2021-11-12:
ok. I'll ignore the 15. Is this also errant possibilty in the 2 from your 3rd party lohnmann json?

closed 2024-04-01 by diablodale