luxonis/depthai-core #252: depthai precompiled distrib for windows has undisclosed requirements

diablodale, 2021-11-10:
I am incrementally discovering that the depthai Windows precompiled distrib has very specific (and incompatible with my needs) setup requirements. It requires:

* use of depthai in shared (aka DLL) mode
* OpenCV 4.5.3 only
* OpenCV built with world only (not individual DLLs)
* OpenCV build in DLL/shared mode only
* I have yet to determine what threading library it uses (windows, concrt, tbb, etc.)

Here is one failure on running a hello world EXE built with the precompiled Windows distrib `depthai-core-v2.11.1-win64.zip` which exposes some of the hidden requirements
[image]

I am not able to use the DepthAI windows distribution built DLL, that requires OpenCV DLLs, and the old 4.5.3 version. I have fixed OpenCV's memory allocation, threading, TLS, and OpenCL infrastructure beyond even what I already contributed to OpenCV.

### Requests

1. All requirements for the depthai Windows precompiled distrib be clearly listed so that I and other devs know at the beginning everything. At a minimum, these locations should have that information:
   https://github.com/luxonis/depthai-core/blob/main/README.md  
   https://docs.luxonis.com/projects/api/en/latest/install/#windows-10  
   README put inside the distrib `depthai-core-v2.11.1-win64.zip`
2. The Windows precompiled distrib be available in static (build shared=no).
3. Support linking to OpenCV static libraries. This might be controversial. I'm happy to squash any rumors and old info.

Until this or the related issue #246 is resolved successfully (so I can build depthai myself), I am blocked from any further DepthAI development on Windows as I can't compile hello world. :-(

themarpe, 2021-11-17:
Hi @diablodale 

The precompiled Windows library builds both core and opencv support, which targets "latest" prebuilt opencv available. One may use depthai-core and use any opencv library to go along.

I haven't looked into making it version agnostic. Do you have any suggestions on this part? In that case that would make it such that the different library linked would have to have same ABI. While another option would be to to have this code in a public header, so no opencv dependency would be needed by depthai in advance, but would trickle down to consumers if opencv support was desired - where only API has to be the same.  

Reason for providing a shared library (currently) is that it covers both being used by a CMake project as well as relatively easily being consumed by manually specifying include directories and libraries in eg Visual Studio. (in static case, all the dependents of DepthAI have to be correctly specified as well) 

Noted on requests, will update the README and add some extra information on that front into the prebuilts.
Regarding linking to static OpenCV I think this will come freely with a static build, but in that case the static OpenCV build provided will have to correctly specify all of its dependencies, etc...

diablodale, 2021-11-17:
OpenCV has some processes in place to maintain ABI but when OS loaders look for specific file names like in OP, its becomes a need for both ABI *and* filenames. There is also the ambiguity from *where* that specific filename needs to come.
From https://github.com/opencv/opencv/releases ?
From https://opencv.org/releases/ which uses sourceforge?
From sourceforge directly?
From xyz.myhacker.com?
winget?
vcpkg?

I don't know of a trivial method to make depthai agnostic to ABI versions of opencv.

With effort, you might be able to get a fixed range of opencv versions (aka filesnames) all having the same ABI to work. That effort would be different for different OSs. On Windows you could delayload the opencv DLL, then hook the delayload function to scan for the range of filenames you support and try to load them one by one. Then assume the loader will match the function names automatically as per normal DLL loading. If that doesn't work, then you could again one-by-one search for valid DLL filenames, then manually load a DLL, and manually get function pointers to all the OpenCV functions you need, then call the OpenCV functions by those pointers.

All somewhat related to our discussion back in March cautioning about versioning and ABI topics as it relates to OpenCV and its dependencies.

diablodale, 2021-11-19:
On Windows, keep in mind how OpenCV itself is compiled for STATIC linking. I hope this is not a concern for DLL linking.

OpenCV is compiled with the `/EHa` option.
https://github.com/opencv/opencv/blob/17234f82d025e3bbfbf611089637e5aa2038e7b8/cmake/OpenCVCompilerOptions.cmake#L406

Depthai-core uses `/EHsc`.

Yet Microsoft warns that all compilation units linked into the same binary must use the same `/EH` switch.
https://docs.microsoft.com/en-us/cpp/build/reference/eh-exception-handling-model?view=msvc-170#structured-and-standard-c-exception-handling

So in the static linking scenario, it is a violation of Microsoft's warning.

themarpe, 2021-11-20:
Thanks - will take a look at this for potential static prebuild. 

And for dynamic, the filenames for DLLs is kinda cumbersome as well. Will see about doing a public header way instead.