luxonis/depthai-core #254: fails to cmake config & build on Windows

Augustus-demon, 2021-11-11:
I've tried every single way to build this project ,including cmake gui ,cmake command line and VS Code with cmake tools ,but it always leads to a debug info and the process cannot go forward.

# example in git bash
'''
$ cmake -S. -Bbuild
-- Building for: Visual Studio 15 2017
-- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.19042.
-- The CXX compiler identification is MSVC 19.16.27045.0
-- The C compiler identification is MSVC 19.16.27045.0
-- Check for working CXX compiler: G:/VS/VS2017/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe
-- Check for working CXX compiler: G:/VS/VS2017/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: G:/VS/VS2017/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: G:/VS/VS2017/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: G:/Git/mingw64/bin/git.exe (found version "2.23.0.windows.1")
-- [hunter] Calculating Toolchain-SHA1
-- [hunter] Calculating Config-SHA1
-- [hunter] HUNTER_ROOT: C:/Users/Js/.hunter
-- [hunter] [ Hunter-ID: 062a19a | Toolchain-ID: 1c8709f | Config-ID: 05fa1a3 ]
-- [hunter] NLOHMANN_JSON_ROOT: C:/Users/Js/.hunter/_Base/062a19a/1c8709f/05fa1a3/Install (ver.: 3.9.1)
'''
After this very line of debug info appears, the process cannot go any further or terminate with any error.
Cannot come up with any idea of solution for this situation .

Augustus-demon, 2021-11-11:
It seems that it's just too slow .after 3 hours ,it finally resumes and blocks in another node, we will see.

themarpe, 2021-11-16:
Hi @Augustus-demon.
This seems odd - as you've said it always get stuck in same spots? And even if you exit with CTRL+C and retry the command?

Can you post a more detailed log using the following define added: 
```
cmake -S. -Bbuild -DHUNTER_STATUS_DEBUG=ON
```

Augustus-demon, 2021-11-16:
Thanks. This turns out to be my computer's fault, as I said ,It's so slow that I misunderstands that there's some problems.

closed 2021-11-16 by themarpe