luxonis/depthai-core #255: link error with release v2.11.1 win64

Augustus-demon, 2021-11-11:
I need the opencv support so I chose the win64 library, the "depthai-opencv.lib" works, but the depthai-core.lib seems not,
# Code 
```
dai::Pipeline pipeline;

// Define source and output
auto camRgb = pipeline.create<dai::node::ColorCamera>();
auto xoutVideo = pipeline.create<dai::node::XLinkOut>();

xoutVideo->setStreamName("video");

// Properties
camRgb->setBoardSocket(dai::CameraBoardSocket::RGB);
camRgb->setResolution(dai::ColorCameraProperties::SensorResolution::THE_1080_P);
camRgb->setVideoSize(1920, 1080);

xoutVideo->input.setBlocking(false);
xoutVideo->input.setQueueSize(1);

// Linking
camRgb->video.link(xoutVideo->input);

// Connect to device and start pipeline
dai::Device device(pipeline);

auto video = device.getOutputQueue("video");

while (true) {
	auto videoIn = video->get<dai::ImgFrame>();

	// Get BGR frame from NV12 encoded video frame to show with opencv
	// Visualizing the frame on slower hosts might have overhead
	cv::imshow("video", videoIn->getCvFrame());

	int key = cv::waitKey(1);
	if (key == 'q' || key == 'Q') {
		return ;
	}
}
```

# Errors :
```
1>main.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl dai::node::ColorCamera::setResolution(enum dai::ColorCameraProperties::SensorResolution)" (?setResolution@ColorCamera@node@dai@@QEAAXW4SensorResolution@ColorCameraProperties@3@@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl dai::node::ColorCamera::setVideoSize(int,int)" (?setVideoSize@ColorCamera@node@dai@@QEAAXHH@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl dai::node::ColorCamera::setBoardSocket(enum dai::CameraBoardSocket)" (?setBoardSocket@ColorCamera@node@dai@@QEAAXW4CameraBoardSocket@3@@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl dai::node::ColorCamera::ColorCamera(class std::shared_ptr<class dai::PipelineImpl> const &,__int64)" (??0ColorCamera@node@dai@@QEAA@AEBV?$shared_ptr@VPipelineImpl@dai@@@std@@_J@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl dai::node::XLinkOut::setStreamName(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?setStreamName@XLinkOut@node@dai@@QEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl dai::node::XLinkOut::XLinkOut(class std::shared_ptr<class dai::PipelineImpl> const &,__int64)" (??0XLinkOut@node@dai@@QEAA@AEBV?$shared_ptr@VPipelineImpl@dai@@@std@@_J@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "private: __int64 __cdecl dai::PipelineImpl::getNextUniqueId(void)" (?getNextUniqueId@PipelineImpl@dai@@AEAA_JXZ)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl dai::Node::Output::link(class dai::Node::Input const &)" (?link@Output@Node@dai@@QEAAXAEBVInput@23@@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl dai::Node::Input::setQueueSize(int)" (?setQueueSize@Input@Node@dai@@QEAAXH@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: void __cdecl dai::Node::Input::setBlocking(bool)" (?setBlocking@Input@Node@dai@@QEAAX_N@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl dai::Pipeline::Pipeline(void)" (??0Pipeline@dai@@QEAA@XZ)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: class std::shared_ptr<class dai::DataOutputQueue> __cdecl dai::Device::getOutputQueue(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &)" (?getOutputQueue@Device@dai@@QEAA?AV?$shared_ptr@VDataOutputQueue@dai@@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@4@@Z)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: virtual __cdecl dai::Device::~Device(void)" (??1Device@dai@@UEAA@XZ)
1>main.obj : error LNK2001: 无法解析的外部符号 "public: __cdecl dai::Device::Device(class dai::Pipeline const &)" (??0Device@dai@@QEAA@AEBVPipeline@1@@Z)
```
these symbols seems included in depthai-core.lib, because these link error doesn't  appear when I use the win32 library .

themarpe, 2021-11-16:
Hi @Augustus-demon 
Did you link to both dephtai-opencv & depthai-core? 
Regarding the error message, ignoring opencv support if you link to just depthai-core on win64 and win32 only win32 doesn't produce the errors?

Augustus-demon, 2021-11-16:
Yes, as you said, I linked both. But I turn to previous version of win64 with opencv supported libs (which is version 2.10.0), and that works. I haven't tried what you said, I'll give it a try tomorrow and give further feedbacks. By the way , I tried the latest release v2.12.0 today which doesn't work either.

themarpe, 2021-11-16:
@Augustus-demon looks like win64 prebuilt library was partially overriden with win32. Working on a fix, will upload a new prebuilts after the fix is in place and you can test those afterwards

themarpe, 2021-11-16:
@Augustus-demon can you try these out - https://github.com/luxonis/depthai-core/suites/4374122789/artifacts/115108332
(note: temporary artifacts of the CI build, will get delete after a while)

If all works well over on your end I'll create a release:)

Augustus-demon, 2021-11-17:
@themarpe  I tried only link the win64 core library and generate these errors too, which could almost confirm the problem of win64-core library. Besides, the library from link you gave seems no problem. Thanks a lot 👍 .
By the way ,can win32 library support opencv, perhaps needs to open some marcos when buildding ? Appreciate it if  you give some tips , : )

themarpe, 2021-11-17:
@Augustus-demon great to hear! I've made the release (v2.12.1) with fixed and updated prebuilt libraries:)

The "opencv support" actually just brings the `setFrame` and `getFrame` functions of `ImgFrame` currently. You can use win32 prebuilt DepthAI library and use other opencv functions without any issues (but cannot call the above functions).

Other way is to build DepthAI from sources, targeting Win32 and with OpenCV which targets Win32 as well. Main reason that Win32 doesn't have opencv support along it is that there aren't any win32 prebuilts for opencv anymore (for newer 4.x version). So one would have to build OpenCV from sources as well.

In todays time, I'd highly suggest going with x64 in more or less all the cases

Augustus-demon, 2021-11-17:
Yes, I'm total willing to use x64, however the previous project is based on win32, :( . 
Maybe I'll check on other ways ,thanks anyway . : )
so I'll close this issue soon.

closed 2021-11-17 by Augustus-demon