luxonis/depthai-core #260: CMake build error

DanielSepeda, 2021-11-12:
Hello, I'm currently getting the following build error.

 CMake Error at /home/falcon/.hunter/_Base/062a19a/672a026/2127330/Install/lib/cmake/ZLIB/ZLIBConfig.cmake:36 (add_library):
  add_library cannot create ALIAS target "ZLIB::ZLIB" because target
  "ZLIB::zlib" is IMPORTED.

My full output is: cmake -S. -B build ..
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] HUNTER_ROOT set using HOME environment variable
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] HUNTER_ROOT: /home/falcon/.hunter
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] Settings (initialize):
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   HunterGate done (NO)
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   Cache init (YES)
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] Single-configuration generator
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] Using default cache server
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] List of cache servers:
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   * https://github.com/ingenue/hunter-cache
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] Settings (finalize):
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   HunterGate done (NO)
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   Cache init (YES)
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] Variables from HunterGate:
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   HUNTER_GATE_ROOT: /home/falcon/.hunter
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   HUNTER_GATE_VERSION: 0.23.258
-- [hunter *** DEBUG *** 2021-11-12T13:59:48]   HUNTER_GATE_SHA1: 062a19ab13ce8dffa9a882b6ce3e43bdabdf75d3
-- [hunter] Calculating Toolchain-SHA1
-- [hunter *** DEBUG *** 2021-11-12T13:59:48] [/home/falcon/Downloads/depthai-core/build/_3rdParty/Hunter/toolchain] > "/usr/bin/cmake" "-DTOOLCHAIN_INFO_FILE=/home/falcon/Downloads/depthai-core/build/_3rdParty/Hunter/toolchain/toolchain.info.NOLF" "-DHUNTER_SELF=/home/falcon/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked" "-GUnix Makefiles" "-H/home/falcon/Downloads/depthai-core/build/_3rdParty/Hunter/toolchain" "-B/home/falcon/Downloads/depthai-core/build/_3rdParty/Hunter/toolchain/_builds" "-DCMAKE_MAKE_PROGRAM=/usr/bin/make" "-DCMAKE_RELEASE_POSTFIX="
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/gcc-8
-- Check for working CXX compiler: /usr/bin/gcc-8 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/falcon/Downloads/depthai-core/build/_3rdParty/Hunter/toolchain/_builds
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Already exists: /home/falcon/.hunter/_Base/062a19a/672a026/toolchain.info
-- [hunter] Calculating Config-SHA1
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Cache data is up-to-date
-- [hunter] HUNTER_ROOT: /home/falcon/.hunter
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] HUNTER_TOOLCHAIN_ID_PATH: /home/falcon/.hunter/_Base/062a19a/672a026
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] HUNTER_CONFIGURATION_TYPES: Release
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] HUNTER_BUILD_SHARED_LIBS: 
-- [hunter] [ Hunter-ID: 062a19a | Toolchain-ID: 672a026 | Config-ID: 2127330 ]
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] load: /home/falcon/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/projects/nlohmann_json/hunter.cmake
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] nlohmann_json versions available: [2.1.1-p0;2.1.1-p1;3.0.0;3.0.1;3.1.0;3.1.1;3.1.2;3.2.0;3.3.0;3.4.0;3.5.0;3.6.0;3.6.1;3.7.0;3.7.1;3.7.2;3.7.3;3.8.0]
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Package 'nlohmann_json' CONFIGURATION_TYPES: Release
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Package 'nlohmann_json' is cacheable: YES
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Install to: /home/falcon/.hunter/_Base/062a19a/672a026/2127330/Install
-- [hunter] NLOHMANN_JSON_ROOT: /home/falcon/.hunter/_Base/062a19a/672a026/2127330/Install (ver.: 3.9.1)
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Package already installed: nlohmann_json
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] load: /home/falcon/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/projects/nlohmann_json/hunter.cmake ... end
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] File not found (injected package): /home/falcon/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/projects/XLink/hunter.cmake
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Injected package 'XLink'
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] XLink versions available: []
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Package 'XLink' CONFIGURATION_TYPES: Release
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Package 'XLink' is cacheable: YES
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Install to: /home/falcon/.hunter/_Base/062a19a/672a026/2127330/Install
-- [hunter] XLINK_ROOT: /home/falcon/.hunter/_Base/062a19a/672a026/2127330/Install (ver.: luxonis-2021.3-develop)
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] Package already installed: XLink
-- [hunter *** DEBUG *** 2021-11-12T13:59:49] load: /home/falcon/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/projects/BZip2/hunter.cmake
-- [hunter *** DEBUG *** 2021-11-12T13:59:50] BZip2 versions available: [1.0.6-p1;1.0.6-p2;1.0.6-p3;1.0.6-p4;1.0.8-p0]
-- [hunter *** DEBUG *** 2021-11-12T13:59:50] Package 'BZip2' CONFIGURATION_TYPES: Release
-- [hunter *** DEBUG *** 2021-11-12T13:59:50] Package 'BZip2' is cacheable: YES
[... 74 lines cut ...]
-- File already downloaded (resources): depthai-device-fwp-2021725697c9b570383b0597951a9d7b162e6182.tar.xz
-- Downloading depthai bootloader
-- folder: /home/falcon/Downloads/depthai-core/build/resources
-- maturity: release
-- commit_version_arg: 0.0.15
-- bootloader shared commit: 126f344155346836a6a2d67b0993c161b6b48729
-- File already downloaded (resources): depthai-bootloader-shared-commit-hash-0.0.15.txt
-- depthai-bootloader-shared between device and host MATCH!. (device: 126f344155346836a6a2d67b0993c161b6b48729, host: 126f344155346836a6a2d67b0993c161b6b48729
-- Downloading and checking depthai-bootloader-fwp.tar.xz
-- File already downloaded (resources): depthai-bootloader-fwp-0.0.15.tar.xz
-- LIST OF RESOURCE COMPILED FILES: /home/falcon/Downloads/depthai-core/build/resources/depthai-device-fwp-2021725697c9b570383b0597951a9d7b162e6182.tar.xz;/home/falcon/Downloads/depthai-core/build/resources/depthai-bootloader-fwp-0.0.15.tar.xz
-- OpenCV and required libraries (opencv_core;opencv_imgproc) found. OpenCV Support enabled
-- [hunter *** DEBUG *** 2021-11-12T13:59:51] load: /home/falcon/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/projects/Catch/hunter.cmake
-- [hunter *** DEBUG *** 2021-11-12T13:59:51] Catch versions available: [2.11.0;2.9.1;2.7.0;2.6.0;2.5.0;2.2.2;2.2.1;2.0.1;1.8.2-p0;1.5.9]
-- [hunter *** DEBUG *** 2021-11-12T13:59:51] Package 'Catch' CONFIGURATION_TYPES: Release
-- [hunter *** DEBUG *** 2021-11-12T13:59:51] Package 'Catch' is cacheable: YES
-- [hunter *** DEBUG *** 2021-11-12T13:59:51] Install to: /home/falcon/.hunter/_Base/062a19a/672a026/2127330/Install
-- [hunter] CATCH_ROOT: /home/falcon/.hunter/_Base/062a19a/672a026/2127330/Install (ver.: 2.11.0)
-- [hunter *** DEBUG *** 2021-11-12T13:59:51] Package already installed: Catch
-- [hunter *** DEBUG *** 2021-11-12T13:59:51] load: /home/falcon/.hunter/_Base/Download/Hunter/0.23.258/062a19a/Unpacked/cmake/projects/Catch/hunter.cmake ... end
-- BUILD_DATETIME: 2021-11-12 21:59:51 +0000, BUILD_COMMIT: c1b697afa432c8d620b097ba57e20a0ffd2ec52d, BUILD_COMMIT_DATETIME: 2021-10-25 22:42:02 +0200
-- Configuring incomplete, errors occurred!

Attached below is my cmake cache
[CMakeCache.txt](https://github.com/luxonis/depthai-core/files/7530847/CMakeCache.txt)

szabi-luxonis, 2021-11-13:
@DanielSepeda this issue is fixed in `develop` branch.
We are releasing a new version v2.12.0 today, which will incorporate fixes related to `zlib` from https://github.com/luxonis/depthai-core/pull/237

DanielSepeda, 2021-11-15:
@szabi-luxonis Okay it seems switching to the `develop` branch allowed CMake to resolve zlib aliasing issue. However, I am now getting an error during my ccmake configuration.
Here is the error from my CMakeError.log

`Determining if the pthread_create exist failed with the following output:
Change Dir: /home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_8bd76/fast"
/usr/bin/make -f CMakeFiles/cmTC_8bd76.dir/build.make CMakeFiles/cmTC_8bd76.dir/build
make[1]: Entering directory '/home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_8bd76.dir/CheckSymbolExists.c.o
/usr/bin/gcc-8   -fPIC    -o CMakeFiles/cmTC_8bd76.dir/CheckSymbolExists.c.o   -c /home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_8bd76
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8bd76.dir/link.txt --verbose=1
/usr/bin/gcc-8 -fPIC     -rdynamic CMakeFiles/cmTC_8bd76.dir/CheckSymbolExists.c.o  -o cmTC_8bd76 
CMakeFiles/cmTC_8bd76.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_8bd76.dir/build.make:97: recipe for target 'cmTC_8bd76' failed
make[1]: *** [cmTC_8bd76] Error 1
make[1]: Leaving directory '/home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_8bd76/fast' failed
make: *** [cmTC_8bd76/fast] Error 2

File /home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_1612f/fast"
/usr/bin/make -f CMakeFiles/cmTC_1612f.dir/build.make CMakeFiles/cmTC_1612f.dir/build
make[1]: Entering directory '/home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1612f.dir/CheckFunctionExists.c.o
/usr/bin/gcc-8   -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_1612f.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_1612f
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1612f.dir/link.txt --verbose=1
/usr/bin/gcc-8 -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_1612f.dir/CheckFunctionExists.c.o  -o cmTC_1612f -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_1612f.dir/build.make:97: recipe for target 'cmTC_1612f' failed
make[1]: *** [cmTC_1612f] Error 1
make[1]: Leaving directory '/home/falcon/falc/peregrine/master/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_1612f/fast' failed
make: *** [cmTC_1612f/fast] Error 2`

I think cmake is having some difficulties locating CUDA_SDK since it states CUDA_SD_ROOT_DIR=CUDA_SDK_ROOT_DIR-NOTFOUND in my configuration file.

[CMakeError.log](https://github.com/luxonis/depthai-core/files/7539810/CMakeError.log)
[CMakeCache.txt](https://github.com/luxonis/depthai-core/files/7539822/CMakeCache.txt)

DanielSepeda, 2021-11-15:
@szabi-luxonis should I close this issue (#260) since #237 resolved the aliasing issue, and open a new issue to address the pthread undefined_reference error given above?

szabi-luxonis, 2021-11-15:
> @szabi-luxonis should I close this issue (#260) since #237 resolved the aliasing issue, and open a new issue to address the pthread undefined_reference error given above?

Try:
`find /usr/lib -name "libpthread*"`
or `find /lib -name "libpthread*"`
Is there any ?

If not try running `sudo apt-get install build-essential` and try again.

DanielSepeda, 2021-11-15:
@szabi-luxonis with command `find /usr/lib -name "libpthread*"`
I get: 
```
/usr/lib/x86_64-linux-gnu/libpthread.a
/usr/lib/x86_64-linux-gnu/libpthread.so
/usr/lib/x86_64-linux-gnu/libpthread_nonshared.a
```
After deleting my build folder and restarting fresh. It seems to build successfully during `cmake ..` and `cmake -S . -B build ..` but when I try `make install` 

I get the following error when building SteroDepthConfig.cpp.o

```
[ 11%] Building CXX object CMakeFiles/depthai-core.dir/src/pipeline/datatype/StereoDepthConfig.cpp.o
/home/falcon/Downloads/depthai-core/src/pipeline/datatype/StereoDepthConfig.cpp: In member function ‘void dai::StereoDepthConfig::setExtendedDisparity(bool)’:

/home/falcon/Downloads/depthai-core/src/pipeline/datatype/StereoDepthConfig.cpp:50:26: error: ‘struct dai::RawStereoDepthConfig::AlgorithmControl’ has no member named ‘enableExtended’; did you mean ‘enableSubpixel’?
     cfg.algorithmControl.enableExtended = enable;
                          ^~~~~~~~~~~~~~
                          enableSubpixel

/home/falcon/Downloads/depthai-core/src/pipeline/datatype/StereoDepthConfig.cpp: In member function ‘float dai::StereoDepthConfig::getMaxDisparity() const’:
/home/falcon/Downloads/depthai-core/src/pipeline/datatype/StereoDepthConfig.cpp:63:29: error: ‘struct dai::RawStereoDepthConfig::AlgorithmControl’ has no member named ‘enableExtended’; did you mean ‘enableSubpixel’?
     if(cfg.algorithmControl.enableExtended) maxDisp *= 2;
                             ^~~~~~~~~~~~~~
                             enableSubpixel

CMakeFiles/depthai-core.dir/build.make:1142: recipe for target 'CMakeFiles/depthai-core.dir/src/pipeline/datatype/StereoDepthConfig.cpp.o' failed

```

szabi-luxonis, 2021-11-15:
@DanielSepeda update git submodules
'git submodule update --init --recursive'

DanielSepeda, 2021-11-15:
> @DanielSepeda update git submodules 'git submodule update --init --recursive'

@szabi-luxonis So close. Using the command `git submodule update --recursive `allowed me to successfully `make install`.
Now when I am integrating depthai into my own project's CMakeLists.txt with the following configurations:
```
set(depthai_DIR "../../depthai-core/build")
find_package(depthai CONFIG REQUIRED)
target_link_libraries(${PROJECT_NAME} PRIVATE depthai::opencv)
```

I get the same error from the beginning of this issue.

```
CMake Error at /usr/local/lib/cmake/depthai/dependencies/lib/cmake/ZLIB/ZLIBConfig.cmake:36 (add_library):
  add_library cannot create ALIAS target "ZLIB::ZLIB" because target
  "ZLIB::zlib" is IMPORTED.
Call Stack (most recent call first):
  /home/falcon/Downloads/depthai-core/build/depthaiDependencies.cmake:34 (find_package)
  /home/falcon/Downloads/depthai-core/build/depthaiConfig.cmake:14 (include)
  src/Camera/Oakie/CMakeLists.txt:16 (find_package)

```
As soon I insert the line `find_package(depthai CONFIG REQUIRED)` the build fails with the above error. (Note: I have cleaned my own build folder after _successfully_ installing depthai)

DanielSepeda, 2021-11-15:
@szabi-luxonis UPDATE: it finally worked. I had to set `CMAKE_INSTALL_PREFIX` to `/usr/local` in my cmake configuration. After rebuilding my project successfully built and linked with depthai. Thank you for your help!

szabi-luxonis, 2021-11-15:
@DanielSepeda thanks for circling back.
FWIW install instructions are described in readme: https://github.com/luxonis/depthai-core#installing
If you consider issue solved, it can be closed.

closed 2021-11-15 by DanielSepeda