luxonis/depthai-core #265: Run time failure when running c++ code but not python code

DanielSepeda, 2021-11-16:
It seems I keep receiving the following error every time I construct a `dai::Device() object`. I'm currently running the RGB Preview example from _https://docs.luxonis.com/projects/api/en/latest/samples/ColorCamera/rgb_preview/_  
I can confirm that I installed depth-ai SDK correctly, and have integrated into my project since there were no compilation errors during the build process (I'm running Ubuntu 18).
I can also confirm that the camera is connected and reachable since I can run different python examples and they are able connect to the camera.
This is the following **runtime** error I get when I run the above example
```
terminate called after throwing an instance of 'nanorpc::core::exception::logic'
  what():  [json.exception.type_error.304] cannot use at() with null
Stack trace (most recent call last):
#11   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#10   Object "./falcon", at 0x5583e0d137e9, in _start
#9    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdb3cb31bf6, in __libc_start_main
#8    Object "./falcon", at 0x5583e0d13a9e, in main
#7    Object "./falcon", at 0x5583e0e33371, in dai::Device::Device(dai::Pipeline const&, dai::UsbSpeed)
#6    Object "./falcon", at 0x5583e0e4280e, in dai::DeviceBase::tryStartPipeline(dai::Pipeline const&)
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fdb3d1be958, in __cxa_rethrow
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fdb3d1be670, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fdb3d1be605, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fdb3d1b3256, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdb3cb50920, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdb3cb4efb7, in gsignal
Aborted (Signal sent by tkill() 19117 1000)
Aborted (core dumped)
```
Oddly, this runtime error is consistent across different C++ example programs (e.g. https://docs.luxonis.com/projects/api/en/latest/samples/IMU/imu_accelerometer_gyroscope/ and  https://docs.luxonis.com/projects/api/en/latest/samples/StereoDepth/depth_preview/ ) with various pipeline and device initialization. However, the compilation process yields no errors.

Any suggestions?

themarpe, 2021-11-16:
Hi @DanielSepeda - can you post a log with `DEPTHAI_LEVEL=trace` env variable set?
Does this exception happen always or occasionally?

DanielSepeda, 2021-11-16:
@themarpe Below is the log with `DEPTHAI_LEVEL=trace`

```
[2021-11-16 12:29:33.821] [debug] Library information - version: 2.12.0, commit: 5a5b2f0ae279dc82cdd93da371751c1c5a208afe from 2021-11-13 03:05:23 +0200, build: 2021-11-15 21:00:32 +0000
[2021-11-16 12:29:33.826] [debug] Initialize - finished
[2021-11-16 12:29:33.969] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.15.tar.xz' open: 6ms, archive read: 137ms
[2021-11-16 12:29:34.362] [debug] Resources - Archive 'depthai-device-fwp-3bb469fa08b7b0abc4353cd27219521dfd763a9d.tar.xz' open: 6ms, archive read: 534ms
[2021-11-16 12:29:35.135] [debug] Device - OpenVINO version: 2021.4
[2021-11-16 12:29:40.888] [debug] Booting FW with Bootloader. Version 0.0.15, Time taken: 5392ms
[2021-11-16 12:29:40.888] [debug] DeviceBootloader about to be closed...
[2021-11-16 12:29:40.889] [debug] XLinkResetRemote of linkId: (0)
[2021-11-16 12:29:42.139] [debug] DeviceBootloader closed, 1250
[2021-11-16 12:29:42.150] [trace] RPC: [1,1,10182484315255513117,[0]]
[2021-11-16 12:29:42.156] [trace] RPC: [1,1,5804304869041345055,[1.0]]
[2021-11-16 12:29:42.163] [trace] RPC: [1,1,17425566508637143278,null]
[2021-11-16 12:29:42.164] [trace] Log vector decoded, size: 3
[10.1.1.165] [93.684] [system] [info] Memory Usage - DDR: 0.12 / 358.82 MiB, CMX: 2.05 / 2.50 MiB, LeonOS Heap: 24.39 / 78.63 MiB, LeonRT Heap: 2.88 / 23.84 MiB
[10.1.1.165] [93.684] [system] [info] Temperatures - Average: 37.77 °C, CSS: 38.65 °C, MSS 37.48 °C, UPA: 36.77 °C, DSS: 38.18 °C
[10.1.1.165] [93.684] [system] [info] Cpu Usage - LeonOS 5.49%, LeonRT: 0.56%
[2021-11-16 12:29:42.316] [trace] RPC: [1,1,16527326580805871264,[{"connections":[{"node1Id":0,"node1Output":"preview","node2Id":1,"node2Input":"in"}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"xlinkChunkSize":-1},"nodes":[[0,null],[1,null]]}]]
[2021-11-16 12:29:42.638] [debug] Device about to be closed...
[2021-11-16 12:29:42.638] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[2021-11-16 12:29:42.639] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[2021-11-16 12:29:42.639] [debug] XLinkResetRemote of linkId: (0)
[2021-11-16 12:29:42.904] [debug] Device closed, 265
terminate called after throwing an instance of 'nanorpc::core::exception::logic'
  what():  [json.exception.type_error.304] cannot use at() with null
Stack trace (most recent call last):
#12   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#11   Object "./falcon", at 0x55c37144c9f9, in _start
#10   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9d041f5bf6, in __libc_start_main
#9    Object "./falcon", at 0x55c37144cb22, in main
#8    Object "./falcon", at 0x55c371563e11, in Oakie::ConnectCamera()
#7    Object "./falcon", at 0x55c37156c889, in dai::Device::Device(dai::Pipeline const&, dai::UsbSpeed)
#6    Object "./falcon", at 0x55c37157bd26, in dai::DeviceBase::tryStartPipeline(dai::Pipeline const&)
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9d04882958, in __cxa_rethrow
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9d04882670, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9d04882605, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f9d04877256, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9d04214920, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9d04212fb7, in gsignal
Aborted (Signal sent by tkill() 3484 1000)
Aborted (core dumped)
```

This exception **always happens** when I construct the `depthai::Device() object`

themarpe, 2021-11-16:
@DanielSepeda thanks - can you run you application with `gdb` and print the stack trace.
```
gdb ./falcon
```
And type `bt` afterwards to print out the stack trace (it'll be more detailed)

Also, can you reproduce your issue in a stock example compiled under `depthai-core` repo using CMake option `DEPTHAI_BUILD_EXAMPLES=ON`? Maybe try bringing your app over one of the examples if it isn't too complex.

Also does you application use any kind of specific dependencies? (nanorpc or nlohamann json?)

DanielSepeda, 2021-11-16:
@themarpe Okay I will try to get the program out into a stock example with a bare minimum CMake and re-run with _gdb_ and _bt_ 
Yes, my application does use the nlohmann json library. We were using version 3.2. Could this be the reason for the error?

themarpe, 2021-11-16:
@DanielSepeda 
Having a different nlohmann might be the reason for this issues. Just for a quick test, if your app allows, can you update the nlohmann you use to 3.9.1?

szabi-luxonis, 2021-11-17:
@DanielSepeda can you try adding the following [here](https://github.com/luxonis/depthai-core/blob/main/cmake/Hunter/config.cmake#L5):
```
    CMAKE_ARGS
        CMAKE_CXX_VISIBILITY_PRESET=hidden
        CMAKE_C_VISIBILITY_PRESET=hidden
```

then clear hunter cache and rebuild. (`rm -rf ~/.hunter`)

themarpe, 2021-11-17:
@szabi-luxonis nlohmann json library is header only and no compilation step is needed when "building (configuring)". I think this is a no operation in this case. Could be added to DepthAI (if built separately) or to the final app that links DepthAI. Still, care has to be taken as public headers use it as well, which adds symbols to both DepthAI and app.

DanielSepeda, 2021-11-17:
> @DanielSepeda Having a different nlohmann might be the reason for this issues. Just for a quick test, if your app allows, can you update the nlohmann you use to 3.9.1?

This fixed my issue! I had to explicitly set the program to build with nlohmann json 3.9.1 in my CMakeLists.txt as
```
set(nlohmann_json_DIR "../../.hunter/_Base/062a19a/672a026/ce0762d/Install/lib/cmake/nlohmann_json")
find_package(nlohmann_json 3.9.1 REQUIRED)
```

closed 2021-11-17 by DanielSepeda

themarpe, 2021-11-17:
@DanielSepeda great, thanks for letting us know!
If you'd be up, could you test setting this requirement in our depthai `cmake/depthaiDependencies` under find_package?)
Wondering whether find_package will then grab correct version or complain that 3.9.1 is required

This would allow us to add this under our umbrella, so users of library don't have to know/deal with this potential issue

DanielSepeda, 2021-11-17:
When building as is without altering `cmake/depthaiDepencies`; The `find_package(nlohmann_json ${_QUIET} CONFIG REQUIRED)` call finds **nlohmann_json version 3.9.1**. 
See the following output log.
```
 Found nlohmann_json: /home/falcon/.hunter/_Base/062a19a/672a026/ce0762d/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version
 "3.9.1")
```
Now if I set `find_package(nlohmann_json 3.2 ${_QUIET} CONFIG REQUIRED)` from within `cmake/depthaiDepencies`
it silently finds version 3.2.0 but still builds with no errors.
```
 Found nlohmann_json: /home/falcon/.hunter/_Base/062a19a/672a026/ce0762d/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found suitable
 version "3.9.1", minimum required is "3.2.0")
```

themarpe, 2021-11-17:
@DanielSepeda 
How about the other way around, so you app doesn't specify nlohmann version (or does but 3.2.0) and you set **3.9.1** in `depthaiDependencies.cmake`