luxonis/depthai-core #271: build no error but get unknown error when using

Augustus-demon, 2021-11-22:
I can successfully build my depth-core.lib in win32 format but when I try to link it ,these errors occured.
```
1>depthai-core.lib(XLinkStream.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(DeviceBootloader.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(AssetManager.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(DataQueue.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(DeviceBase.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(Resources.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(ColorCamera.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(MonoCamera.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(Pipeline.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(Device.obj) : error LNK2001: 无法解析的外部符号 "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl fmt::v7::detail::vformat(class fmt::v7::basic_string_view<char>,struct fmt::v7::format_args)" (?vformat@detail@v7@fmt@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@D@23@Uformat_args@23@@Z)
1>depthai-core.lib(Resources.obj) : error LNK2001: 无法解析的外部符号 "void __cdecl fmt::v7::detail::vformat_to<char>(class fmt::v7::detail::buffer<char> &,class fmt::v7::basic_string_view<char>,class fmt::v7::basic_format_args<class fmt::v7::basic_format_context<class fmt::v7::detail::buffer_appender<char>,char> >,class fmt::v7::detail::locale_ref)" (??$vformat_to@D@detail@v7@fmt@@YAXAAV?$buffer@D@012@V?$basic_string_view@D@12@V?$basic_format_args@V?$basic_format_context@V?$buffer_appender@D@detail@v7@fmt@@D@v7@fmt@@@12@Vlocale_ref@012@@Z)
1>depthai-core.lib(XLinkConnection.obj) : error LNK2001: 无法解析的外部符号 "void __cdecl fmt::v7::detail::vformat_to<char>(class fmt::v7::detail::buffer<char> &,class fmt::v7::basic_string_view<char>,class fmt::v7::basic_format_args<class fmt::v7::basic_format_context<class fmt::v7::detail::buffer_appender<char>,char> >,class fmt::v7::detail::locale_ref)" (??$vformat_to@D@detail@v7@fmt@@YAXAAV?$buffer@D@012@V?$basic_string_view@D@12@V?$basic_format_args@V?$basic_format_context@V?$buffer_appender@D@detail@v7@fmt@@D@v7@fmt@@@12@Vlocale_ref@012@@Z)
1>depthai-core.lib(DeviceBootloader.obj) : error LNK2001: 无法解析的外部符号 "void __cdecl fmt::v7::detail::vformat_to<char>(class fmt::v7::detail::buffer<char> &,class fmt::v7::basic_string_view<char>,class fmt::v7::basic_format_args<class fmt::v7::basic_format_context<class fmt::v7::detail::buffer_appender<char>,char> >,class fmt::v7::detail::locale_ref)" (??$vformat_to@D@detail@v7@fmt@@YAXAAV?$buffer@D@012@V?$basic_string_view@D@12@V?$basic_format_args@V?$basic_format_context@V?$buffer_appender@D@detail@v7@fmt@@D@v7@fmt@@@12@Vlocale_ref@012@@Z)
1>depthai-core.lib(StereoDepth.obj) : error LNK2001: 无法解析的外部符号 "void __cdecl fmt::v7::detail::vformat_to<char>(class fmt::v7::detail::buffer<char> &,class fmt::v7::basic_string_view<char>,class fmt::v7::basic_format_args<class fmt::v7::basic_format_context<class fmt::v7::detail::buffer_appender<char>,char> >,class fmt::v7::detail::locale_ref)" (??$vformat_to@D@detail@v7@fmt@@YAXAAV?$buffer@D@012@V?$basic_string_view@D@12@V?$basic_format_args@V?$basic_format_context@V?$buffer_appender@D@detail@v7@fmt@@D@v7@fmt@@@12@Vlocale_ref@012@@Z)
1>depthai-core.lib(Initialization.obj) : error LNK2001: 无法解析的外部符号 "void __cdecl fmt::v7::detail::vformat_to<char>(class fmt::v7::detail::buffer<char> &,class fmt::v7::basic_string_view<char>,class fmt::v7::basic_format_args<class fmt::v7::basic_format_context<class fmt::v7::detail::buffer_appender<char>,char> >,class fmt::v7::detail::locale_ref)" (??$vformat_to@D@detail@v7@fmt@@YAXAAV?$buffer@D@012@V?$basic_string_view@D@12@V?$basic_format_args@V?$basic_format_context@V?$buffer_appender@D@detail@v7@fmt@@D@v7@fmt@@@12@Vlocale_ref@012@@Z)
1>depthai-core.lib(DataQueue.obj) : error LNK2001: 无法解析的外部符号 "void __cdecl fmt::v7::detail::vformat_to<char>(class fmt::v7::detail::buffer<char> &,class fmt::v7::basic_string_view<char>,class fmt::v7::basic_format_args<class fmt::v7::basic_format_context<class fmt::v7::detail::buffer_appender<char>,char> >,class fmt::v7::detail::locale_ref)" (??$vformat_to@D@detail@v7@fmt@@YAXAAV?$buffer@D@012@V?$basic_string_view@D@12@V?$basic_format_args@V?$basic_format_context@V?$buffer_appender@D@detail@v7@fmt@@D@v7@fmt@@@12@Vlocale_ref@012@@Z)
1>depthai-core.lib(DeviceBase.obj) : error LNK2001: 无法解析的外部符号 "void __cdecl fmt::v7::detail::vformat_to<char>(class fmt::v7::detail::buffer<char> &,class fmt::v7::basic_string_view<char>,class fmt::v7::basic_format_args<class fmt::v7::basic_format_context<class fmt::v7::detail::buffer_appender<char>,char> >,class fmt::v7::detail::locale_ref)" (??$vformat_to@D@detail@v7@fmt@@YAXAAV?$buffer@D@012@V?$basic_string_view@D@12@V?$basic_format_args@V?$basic_format_context@V?$buffer_appender@D@detail@v7@fmt@@D@v7@fmt@@@12@Vlocale_ref@012@@Z)
1>depthai-core.lib(DeviceBootloader.obj) : error LNK2001: 无法解析的外部符号 "public: __thiscall spdlog::details::log_msg::log_msg(struct spdlog::source_loc,class fmt::v7::basic_string_view<char>,enum spdlog::level::level_enum,class fmt::v7::basic_string_view<char>)" (??0log_msg@details@spdlog@@QAE@Usource_loc@2@V?$basic_string_view@D@v7@fmt@@W4level_enum@level@2@1@Z)
1>depthai-core.lib(DataQueue.obj) : error LNK2001: 无法解析的外部符号 "public: __thiscall spdlog::details::log_msg::log_msg(struct spdlog::source_loc,class fmt::v7::basic_string_view<char>,enum spdlog::level::level_enum,class fmt::v7::basic_string_view<char>)" (??0log_msg@details@spdlog@@QAE@Usource_loc@2@V?$basic_string_view@D@v7@fmt@@W4level_enum@level@2@1@Z)
```
Did any of the dependencies missing which could cause this ? because there is several ssl connect errors during the cmake process when downloading something ,seems named basic-deps.DONE or something . After 10 failed attemps , it went through and cmake is end with success. Does those files necessary?

Augustus-demon, 2021-11-22:
I temporarily use the win32 core library build by you which works well , but still wanna figure THIS out .

themarpe, 2021-11-22:
Hi @Augustus-demon 
So the prebuilt win32 library works okay, but building it from source and linking doesn't work (above error)?
Did you build as shared or as static library? If you built as static, then you need to link to all of the dependencies as well (can be found under lib/cmake/depthai/dependencies). For integration with non CMake build systems (like non CMake Visual Studio solution), its highly recommended to go with shared library approach instead, as only the final depthai library has to be properly linked.
Check the Readme for more information

Augustus-demon, 2021-11-22:
OMG !  !  ! that's pretty shamed, I used the static way yesterday and found it's not what I need , changed to the integration way but I forgot to rebuild so the lib files remained the previous ,which cause this , what an idiot !  thanks anyway.
(Can issue be deleted ? I could not bear this kind of stupid mistakes. :(

themarpe, 2021-11-22:
@Augustus-demon not a problem at all, happy to help.
If you don't mind, I'd rather modify the title to reflect the issue at hand, so that anyone with similar linking error can see what could be the issue and can try recompiling as well - I think it'll help more people, than cause issue:) 
Is that okay with you?

Augustus-demon, 2021-11-23:
All right , in that case,happy to help. : )