luxonis/depthai-core #272: XLink not installed with `cmake install` (only static issue?)

diablodale, 2021-11-22:
For a static build of depthai, using `DEPTHAI_XLINK_LOCAL`... a cmake install
* does not install the XLink cmake files 😕
* does not install the XLink static libraries 😕
* does install the xlink includes 🙂

### Setup

* Microsoft Windows [Version 10.0.19044.1348]
* VS2019 Community v16.11.7
* cmake v3.19.1
* depthai-core v2.12.1 + my local fixes

### Repro

1. set depthai for a static build, x64, and opencv support
2. set depthai to use a local xlink with `DEPTHAI_XLINK_LOCAL`. Use the same commit as hunter would.
3. set depthai to install to a known location e.g. `CMAKE_INSTALL_PREFIX=c:/myinstalldir`
4. cmake configure, build, and then install

### Result

Either try to find_package() and then compile an app that would use this depthai install
-or-
search `c:/myinstalldir` for the xlink cmake file `XLinkConfig.cmake` and its static libraries `XLink.lib`/`XLinkd.lib`
You will find none of these files.

### Expected

Your app finds depthai and all its dependencies (like XLink) and cleanly builds.
-or-
You find XLink's cmake files and static libraries in the given install location `c:/myinstalldir`

### Notes

I see all the needed files in the depthai build directory `...\depthai-core\build\XLink`. This suggests their might be only a small change needed for cmake install process to use those and then do the install.

diablodale, 2021-11-22:
Hacking some on this in depthaiDependencies.cmake (probably wrong approach)...
```cmake
# XLink
if(NOT CONFIG_MODE)
    if(DEPTHAI_XLINK_LOCAL)
        add_subdirectory("${DEPTHAI_XLINK_LOCAL}" ${CMAKE_CURRENT_BINARY_DIR}/XLink EXCLUDE_FROM_ALL)
    else()
        find_package(XLink ${_QUIET} CONFIG REQUIRED)
    endif()
    if(NOT BUILD_SHARED_LIBS)
        list(APPEND targets_to_export XLink)
    endif()
endif()
```

I've got a few steps further, but app compile fails with depthai because it can't find `XLink/XLinkPublicDefines.h`
That header was not cmake-installed by depthai yet both `XLinkConnection.hpp` and `XLinkStream.hpp` need that include themselves.

diablodale, 2021-11-22:
I've got a local hack almost working. I think are three things in play...

* `depthai-core/depthaiDependencies.cmake` includes a `DEPTHAI_XLINK_LOCAL` xlink with the `EXCLUDE_FROM_ALL` flag. I think that flag excludes that added thing from the ALL built target. And likely from the ALL install target. Therefore xlink isn't installed.
* You might be able to get it to install with `cmake --install . --component XLink` but the cmake code in the XLink project skips all installation steps when a component is defined. It doesn't realize the distinction between a component of depthai -vs- a component of XLink.
* Finally, after hacking the above with workarounds, i have the includes and libraries installing. The entire `XLink/include` directory is installed...all 15 include files. Probably only one needs to be installed `XLinkPublicDefines.h`

themarpe, 2021-11-22:
I think removing the `EXCLUDE_FROM_ALL` is the way to go - although there might be way to specify it explicitly in installation part of core instead

closed 2022-01-06 by szabi-luxonis