luxonis/depthai-core #275: received signal SIGPIPE when starting a pipeline

mdtoom, 2021-11-23:
When running the example provided by https://github.com/luxonis/depthai-core-example with depthai-core version 2.13 (instead of 2.10), the program stops when opening the device (`dai::Device device(pipeline);`).
When using version 2.10 it works as expected.
I am also having this issue with a custom pipeline created in c++, that's why I am posting the issue on this repository.
The camera used is an OAK-D POE with a bootloader that has been assigned a static IP address using the following python code:
```
import depthai as dai

STATIC_IP = "192.168.10.100"
NETMASK = "255.255.255.0"
GATEWAY = "192.168.10.1"

print("Getting device")
(f, bl) = dai.DeviceBootloader.getFirstAvailableDevice()
bootloader = dai.DeviceBootloader(bl)

print("Reading config")
config = bootloader.readConfig()
config.setStaticIPv4(STATIC_IP, NETMASK, GATEWAY)

print("Flash config with static ip")
bootloader.flashConfig(config)
```
I don't know how to get back to the default config so I can exclude this from being the cause.

the output of GDB:
```
[Thread 0x7fffe9de6700 (LWP 38138) exited]

Thread 5 "Scheduler00Thr" received signal SIGPIPE, Broken pipe.
[Switching to Thread 0x7fffea5e7700 (LWP 38137)]
__libc_send (flags=<optimized out>, len=84, buf=0x555556484cc0 <schedulerState+3456>, fd=9)
    at ../sysdeps/unix/sysv/linux/send.c:28
28	../sysdeps/unix/sysv/linux/send.c: No such file or directory.
(gdb) backtrace
#0  __libc_send (flags=<optimized out>, len=84, buf=0x555556484cc0 <schedulerState+3456>, fd=9)
    at ../sysdeps/unix/sysv/linux/send.c:28
#1  __libc_send (fd=9, buf=0x555556484cc0 <schedulerState+3456>, len=84, flags=0)
    at ../sysdeps/unix/sysv/linux/send.c:23
#2  0x0000555555a5dcc8 in XLinkPlatformWrite ()
#3  0x0000555555a5b0e6 in dispatcherEventSend ()
#4  0x0000555555a59c94 in eventSchedulerRun ()
#5  0x00007ffff7ba1609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#6  0x00007ffff7760293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

Does anyone else have this issue or knows what causes it?

mdtoom, 2021-11-23:
I want to add that the issue also occurs with version 12.1, but not with 12.0. I looks like the xlink dependency is updated at version 12.1, so maybe that has some side effects? I am using ubuntu 20.04.

alex-luxonis, 2021-11-23:
Could you try removing the Hunter cache: `rm -r ~/.hunter`
and then rebuilding the depthai library / C++ app. Sometimes the updates to Hunter dependencies seem to not be picked up properly.

We'll try to reproduce the issue as well. 
Thanks!

mdtoom, 2021-11-23:
Hi Alex, thanks for the quick reply. Removing the Hunter cache does not solve the issue.

themarpe, 2021-11-23:
Thanks for report @mdtoom 
Will investigate and try to issue a fix by the end of the day

themarpe, 2021-11-24:
@mdtoom - We've found the issue, it was a regression from our end in XLink library.
Fixed by: https://github.com/luxonis/depthai-core/pull/279

Can you test it out on your end as well?

mdtoom, 2021-11-24:
Confirming that this fixes the issue. Thanks!

PR #279 (2021-11-24): XLink regression fix
Fixes regression of XLink signaling a `SIGPIPE` (XLink PR: https://github.com/luxonis/XLink/pull/23)
Closes: #275
closed 2021-11-24 by mdtoom