luxonis/depthai-core #278: regression in v2.13?  examples via ctest fail with exception

diablodale, 2021-11-23:
I think v2.13 has caused a regression in the examples run by ctest.
This afternoon with v2.12.1, I could run the full 61 test suite with no failures.
Then I merged in `main` v2.13
Now when I run the test suite, I get many USB failures which lead to shutdown. So far I've seen it in two tests `rgb_camera_control` and sometimes `object_tracker`. It is suspicous since v2.13 introduced changes in `rgb_camera_control`

```
...
... many USB failures
...
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] usb_bulk_write:472
[ctest] 10: WinUsb_WritePipe failed with error:=22
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -2) | event XLINK_WRITE_RESP
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] sendEvents:1034	Event sending failed[0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] wperror:105	WinUsb_WritePipe: System err 22
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] usb_bulk_write:472
[ctest] 10: WinUsb_WritePipe failed with error:=22
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -2) | event XLINK_WRITE_RESP
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] sendEvents:1034	Event sending failed[0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] wperror:105	WinUsb_WritePipe: System err 22
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] uStack trace (most recent call last)sb_bulk_write:472
[ctest] 10: WinUsb_WritePipe failed with error:=22
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -2) | event XLINK_WRITE_RESP
[ctest] 10: [0m
[ctest] 10: [35mE: [global] [         0] [Scheduler00Thr] sendEvents:1034	Event sending failed[0m
[ctest] 10: :
[ctest] 10: #8    Object "", at 00007FFAC9657034, in BaseThreadInitThunk
[ctest] 10: #7    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp", line 17, in mainCRTStartup [00007FF7A80E4D8E]
[ctest] 10: #6    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 331, in __scrt_common_main [00007FF7A80E4A5E]
[ctest] 10: #5    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 288, in __scrt_common_main_seh [00007FF7A80E4B9E]
[ctest] 10: #4    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 79, in invoke_main [00007FF7A80E4CF9]
[ctest] 10: #3    Source "C:\njs\depthai-core\examples\ColorCamera\rgb_camera_control.cpp", line 101, in main [00007FF7A7547A02]
[ctest] 10:          98:     int sensMax = 1600;
[ctest] 10:          99:
[ctest] 10:         100:     while(true) {
[ctest] 10:       > 101:         auto previewFrames = previewQueue->tryGetAll<dai::ImgFrame>();
[ctest] 10:         102:         for(const auto& previewFrame : previewFrames) {
[ctest] 10:         103:             cv::Mat frame(previewFrame->getHeight(), previewFrame->getWidth(), CV_8UC3, previewFrame->getData().data());
[ctest] 10:         104:             cv::imshow("preview", frame);
[ctest] 10: #2    Source "C:\njs\depthai-core\include\depthai\device\DataQueue.hpp", line 252, in dai::DataOutputQueue::tryGetAll<dai::ImgFrame> [00007FF7A754D5C7]
[ctest] 10:         249:     std::vector<std::shared_ptr<T>> tryGetAll() {
[ctest] 10:         250:         if(!running) throw std::runtime_error(exceptionMessage.c_str());
[ctest] 10:         251:
[ctest] 10:       > 252:         std::vector<std::shared_ptr<T>> messages;
[ctest] 10:         253:         queue.consumeAll([&messages](std::shared_ptr<ADatatype>& msg) {
[ctest] 10:         254:             // dynamic pointer cast may return nullptr
[ctest] 10:         255:             // in which case that message in vector will be nullptr
[ctest] 10: #1    Object "", at 00007FFAAA8CB460, in CxxThrowException
[ctest] 10: #0    Object "", at 00007FFAC7E54F69, in RaiseException
[ctest] 10: [2021-11-23 19:51:15.293] [debug] Device about to be closed...
[ctest] 10: [2021-11-23 19:51:15.865] [debug] Watchdog thread exception caught: Couldn't write data to stream: '__watchdog' (X_LINK_ERROR)
[ctest] 10: [2021-11-23 19:51:17.336] [debug] XLinkResetRemote of linkId: (0)
[ctest] 10: [2021-11-23 19:51:17.336] [debug] Device closed, 2043
[ctest] 10: [2021-11-23 19:51:17.337] [debug] DataInputQueue (control) closed
[ctest] 10: [2021-11-23 19:51:17.337] [debug] DataInputQueue (config) closed
[ctest] 10: -- After process executed,  produced the following exit code: -529697949
[ctest] 10: CMake Error at C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake:42 (message):
[ctest] 10:    produced an error (-529697949) while running
[ctest] 10:
[ctest] 10:
[ctest] 10/61 Test #10: rgb_camera_control ..................***Failed   10.21 sec
```

### Setup

* Microsoft Windows [Version 10.0.19044.1348]
* OAK-D without rgb cali in EEPROM
* VS 20219 Community v16.11.7
* depthai-core `main` 8ab5bf954e3b65f6ec0933e372777aac10b81900 + fix 276 + ownership/thread fixes

### Repro

1. cmake configure for x64, Debug, static library
2. cmake build
3. ctest -j1 -C Debug -T test --output-on-failure -VV

### Result

One or more tests fail. So far I have seen tests 10 `rgb_camera_control` and sometimes 38 `object_tracker`.
I captured test 10 failure 
[test10-fail.txt](https://github.com/luxonis/depthai-core/files/7590480/test10-fail.txt)

### Expected

All tests pass with no failures.

###

diablodale, 2021-11-23:
I deleted my build/ directory and reconfig, and rebuilt so I had all new bits.
Still failed with
[test10-fail2.txt](https://github.com/luxonis/depthai-core/files/7590768/test10-fail2.txt)

The behavior is the test starts, the small preview window appears and it is black and stays black and nothing happens for many seconds and then the app exits.

diablodale, 2021-11-23:
Perhaps contributing is what's seen in the crash logs. Line 101 is a direct call to `tryGetAll` which can throw. But there is no try/catch in the app. So anytime there is a burp/fail in the queue or communication, the app will crash.

The log also shows line 252 (likely the line before it) is `if(!running) throw()`. So a thread somewhere has changed running=false.

diablodale, 2021-11-23:
This issue is tricky to repro. Maybe its just 👻 me...but it seems to often reproduce on the first few times the OAK-D is used *after* unplugging from power and usb...then reattaching power and usb and then running `rgb_camera_control`

If I somehow get `rgb_camera_control` to work, then it continues to work forever. I've run it 50 times in a loop with no problems.
But if I disconnect power+usb, then I can reproduce the crash and exception.
So far, that seems the pattern.

I got a bit further in debugging now that I can somewhat reliably reproduce it.
1. app crashes because there is no try/catch around line 103 `previewQueue->tryGetAll<dai::ImgFrame>()`
2. tryGetAll()  is throwing on line 250 of `DataQueue.hpp` because running= `false`.
3. The exception message that is getting lost but i can see in the debugger is
   ```
   "Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'preview' (X_LINK_ERROR)'"
   ```
4. exceptionMessage is set on line 84 of `Dataqueue.cpp` as part of a catch. The matching try is the spawned thread that is calling `stread.readRaw()` and parsing it.

My guess is that `readRaw()` is throwing which cascades the above behaviors.
There are several other threads also running but at the moment the debugger stopped them, I don't readily see anything interesting. Several threads waiting and/or on semaphores.

Did something change as part of depthai-core v12.3 that alters the communication to/from the device?

themarpe, 2021-11-24:
AFAIK Nothing in terms of communication changed - no XLink or host side changes
Using same test to reproduce, you do not observe same behavior on 2.12?

Just so I get the picture -  does the failing example do anything or gets stuck for 10s and fails afterwards?
If only that example doesn't work, and it continues afterwards, might be a timing issue to check (going from one test to another too soon, device isn't yet ready).

Will test on my end as well.

diablodale, 2021-11-24:
I did not observe it on v2.12.1.   (I did not use 2.12).  2.12.1 has been my base for all the work I've done for the last ~week.

After my fixes applied (most of which are now merged) it passed all 60 tests consistently. After I finished the `initialize()` fix, I then merged in v2.13 and then rebased my remaining fixes on it. Ran the suite and now get the fails.

As above, when I unplugged power and usb from OAK-D. Then reattached them. Several seconds pass so I move my hands, click things in UI...then directly run test 10 `rgb_camera_control` it fails.
The behavior is the test starts, the small preview window appears and it is black and stays black and nothing happens for many seconds and then the app exits. All along the log is showing all the errors and eventually the exit occurs with its error code.

I have seen that running tests via ctest have different behaviors. I can get the fail using ctest as you can see above. But I can't debug because of the many layers between the actual EXE and my debugger. Therefore, I am now running the EXE directly with my debugger and when I do the power+usb cycle I can get the fail also and now can walk things in the debugger.

diablodale, 2021-11-24:
I read there is a coming v2.13.1 that has a change in XLink.
I also see that my XLink fix from 17 Nov was merged into XLink `develop` but that hasn't been merged into `master`.
I see your recent work is being directly merged into master, so some fixes in `develop` are getting in. 🤔

alex-luxonis, 2021-11-24:
I can reproduce the problem with `rgb_camera_control`, in some cases after a quick unplug/replug of USB & power. Only the preview window gets shown (and I'm getting a valid frame there), then it stalls. All subsequent runs (without power-cycling) do work.
Looks like a firmware crash, probably referencing some uninitialized memory.

diablodale, 2021-11-24:
Sounds similar. To refine my repro scenario.
1. Unplug OAK-D from power and usb ,wait 10 seconds
2. Plug OAK-D power, wait 10 seconds
3. Plug OAK-D usb, wait 10 seconds
4. Run `rgb_camera_control.exe`

Reliably reproduces the crash on my hardware.

alex-luxonis, 2021-12-01:
@diablodale This should be fixed in 2.13.3, PR: https://github.com/luxonis/depthai-core/pull/288
It was related to the VideoEncoder init in firmware.

diablodale, 2021-12-01:
Hi. Unfortunately, v2.13.3 fails for me.
First in loop runs. Then it fails on the second run. I can consistantly reproduce the fail on 2nd run.

```
...
... the first run is working...
...
[ctest] 10: [14442C10C1A1C6D200] [105.774] [system] [info] Memory Usage - DDR: 112.26 / 358.55 MiB, CMX: 2.17 / 2.50 MiB, LeonOS Heap: 23.01 / 78.59 MiB, LeonRT Heap: 3.96 / 23.84 MiB
[ctest] 10: [14442C10C1A1C6D200] [105.774] [system] [info] Temperatures - Average: 27.21 ┬░C, CSS: 28.86 ┬░C, MSS 26.90 ┬░C, UPA: 26.16 ┬░C, DSS: 26.90 ┬░C
[ctest] 10: [14442C10C1A1C6D200] [105.774] [system] [info] Cpu Usage - LeonOS 18.66%, LeonRT: 3.25%
[ctest] 10: -- After process executed,  produced the following exit code: Process terminated due to timeout
[ctest]     Test #10: rgb_camera_control ...............   Passed   20.85 sec
[ctest]     Start 10: rgb_camera_control
[ctest] 
[ctest] 10: Test command: "C:\Program Files\CMake\bin\cmake.exe" "-DTIMEOUT_SECONDS=20" "-P" "C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake" "C:/njs/depthai-core/build/examples/rgb_camera_control.exe"
[ctest] 10: Environment variables: 
[ctest] 10:  DEPTHAI_LEVEL=debug
[ctest] 10:  DEPTHAI_WATCHDOG=5000
[ctest] 10: Test timeout computed to be: 1500
[ctest] 10: -- found -P
[ctest] 10: -- arguments: C:/njs/depthai-core/build/examples/rgb_camera_control.exe;
[ctest] 10: [2021-12-01 19:44:55.634] [debug] Library information - version: 2.13.3, commit: 262a8f09e97f83a4bcbdd2c37f9e84762f33b633 from 2021-12-01 19:22:41 +0100, build: 2021-12-01 18:36:04 +0000
[ctest] 10: [2021-12-01 19:44:55.637] [debug] Initialize - finished
[ctest] 10: [2021-12-01 19:44:55.785] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.15.tar.xz' open: 6ms, archive read: 143ms
[ctest] 10: [2021-12-01 19:44:56.329] [debug] Resources - Archive 'depthai-device-fwp-48fca4a443d841221c94c70d5c05e7e946168636.tar.xz' open: 5ms, archive read: 689ms
[ctest] 10: Stack trace (most recent call last):
[ctest] 10: #9    Object "", at 00007FFAC9657034, in BaseThreadInitThunk
[ctest] 10: #8    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp", line 17, in mainCRTStartup [00007FF737ED50AE]
[ctest] 10: #7    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 331, in __scrt_common_main [00007FF737ED4D7E]
[ctest] 10: #6    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 288, in __scrt_common_main_seh [00007FF737ED4EBE]
[ctest] 10: #5    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 79, in invoke_main [00007FF737ED5019]
[ctest] 10: #4    Source "C:\njs\depthai-core\examples\ColorCamera\rgb_camera_control.cpp", line 72, in main [00007FF737337736]
[ctest] 10:          69:     stillEncoder->bitstream.link(stillMjpegOut->input);
[ctest] 10:          70: 
[ctest] 10:          71:     // Connect to device and start pipeline
[ctest] 10:       >  72:     dai::Device device(pipeline);
[ctest] 10:          73: 
[ctest] 10:          74:     // Get data queues
[ctest] 10:          75:     auto controlQueue = device.getInputQueue("control");
[ctest] 10: #3    Source "C:\njs\depthai-core\src\device\Device.cpp", line 26, in dai::Device::Device [00007FF737BC7CA5]
[ctest] 10:          23: // Common explicit instantiation, to remove the need to define in header
[ctest] 10:          24: constexpr std::size_t Device::EVENT_QUEUE_MAXIMUM_SIZE;
[ctest] 10:          25: 
[ctest] 10:       >  26: Device::Device(const Pipeline& pipeline) : DeviceBase(pipeline.getOpenVINOVersion()) {
[ctest] 10:          27:     tryStartPipeline(pipeline);
[ctest] 10:          28: }
[ctest] 10: #2    Source "C:\njs\depthai-core\src\device\DeviceBase.cpp", line 256, in dai::DeviceBase::DeviceBase [00007FF737CAE02F]
[ctest] 10:         254:     // If no device found, throw
[ctest] 10:         255:     if(!found) throw std::runtime_error("No available devices");
[ctest] 10:       > 256:     init(version, false, "");
[ctest] 10:         257: }
[ctest] 10:         258: 
[ctest] 10:         259: DeviceBase::DeviceBase(OpenVINO::Version version, const char* pathToCmd) {
[ctest] 10: #1    Object "", at 00007FFAC208B460, in CxxThrowException
[ctest] 10: #0    Object "", at 00007FFAC7E54F69, in RaiseException
[ctest] 10: -- After process executed,  produced the following exit code: -529697949
[ctest] 10: CMake Error at C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake:42 (message):
[ctest] 10:    produced an error (-529697949) while running
[ctest] 10: 
[ctest] 10: 
[ctest]     Test #10: rgb_camera_control ...............***Failed    5.98 sec
[ctest] -- found -P
[ctest] -- arguments: C:/njs/depthai-core/build/examples/rgb_camera_control.exe;
[ctest] [2021-12-01 19:44:55.634] [debug] Library information - version: 2.13.3, commit: 262a8f09e97f83a4bcbdd2c37f9e84762f33b633 from 2021-12-01 19:22:41 +0100, build: 2021-12-01 18:36:04 +0000
[ctest] [2021-12-01 19:44:55.637] [debug] Initialize - finished
[ctest] [2021-12-01 19:44:55.785] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.15.tar.xz' open: 6ms, archive read: 143ms
[ctest] [2021-12-01 19:44:56.329] [debug] Resources - Archive 'depthai-device-fwp-48fca4a443d841221c94c70d5c05e7e946168636.tar.xz' open: 5ms, archive read: 689ms
[ctest] Stack trace (most recent call last):
[ctest] #9    Object "", at 00007FFAC9657034, in BaseThreadInitThunk
[... cut ...]
[ctest] -- After process executed,  produced the following exit code: -529697949
[ctest] CMake Error at C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake:42 (message):
[ctest]    produced an error (-529697949) while running
[ctest] 
[ctest] 
[ctest] 
[ctest] 
[ctest] 0% tests passed, 1 tests failed out of 1
[ctest] 
[ctest] Total Test time (real) =  26.84 sec
[ctest] 
[ctest] The following tests FAILED:
[ctest] 	 10 - rgb_camera_control (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
```

diablodale, 2021-12-01:
FYI, to achieve the looping in ctest, I run with it params like
```
    "cmake.ctestArgs": [
        "--tests-information",
        "10,10",
        "--repeat",
        "until-fail:30",
        "-VV",
    ],
```

alex-luxonis, 2021-12-01:
Hi @diablodale,
This seems to be caused by:
> [ctest] 10:  DEPTHAI_WATCHDOG=5000

-- the env var set to a large value (5 seconds). Having that exported, I can reproduce the failure of almost every 2nd test ran, in the full suite. It happens because the app is forcibly closed when the test timeout expires, without a proper device reset. The device will auto-reset when the watchdog timeout expires, and by that time the host considers there are `"No available devices"`.
Pressing `q` with an imshow window in focus before the test timeout expires will make it gracefully terminate, so the failure won't be seen.

Can you also try rerunning ctest after unsetting `DEPTHAI_WATCHDOG`.

Thanks!

diablodale, 2021-12-01:
After removing the DEPTHAI_WATCHDOG env var, I do not see errors/failures like the OP.
I ran a full 61 test suite and a loop of 30x on the single test `rgb_camera_control`.
I believe the OP and this specific issue is resolved. 🙂

I will open a separate issue on the startup/timing errors. After removing the watchdog env var, I still got the same startup failures on 6 of 61 tests. All of them look the same.

closed 2021-12-04 by diablodale