luxonis/depthai-core #282: feature request: `ImgFrame::operator bool` and `ImgFrame::valid/empty()`

diablodale, 2021-11-26:
I request two APIs on `dia::ImgFrame` (or its base classes) that express when the ImgFrame contains a frame of actionable image data.
That image might not have a sequence or timestamp...but because it has width+height+data+type...it is actionable.
Enables easy usage -- a bool operator, and an long-form api.

```c++
dai::ImgFrame myframe;
if (myframe) {
  // operate on the frame
}

if (myframe.valid()) { // or empty(), is_valid()
   // operate on the frame
}
```

The code to implement these is trivial. One will implement and the other will call its sibling. More interesting is the rule which causes a `true` return.
First suggestion...
```c++
explicit dai::ImgFrame::operator bool() const {
    return (getType() != Type::NONE) && (getWidth() > 0) && (getHeight() > 0) && !img.data.empty();
}
```

themarpe, 2021-11-27:
I like it - what was the main use case where you saw that having this would be beneficial to your situation?

diablodale, 2021-11-27:
I have an intel TBB pipeline for lots of data processing both on the CPU and via OpenCL. At various stages of processing, I want to validate that image data I have received can be processed -- parameter validation. Since DepthAI allows for rich processing of data (including custom python) on the device....it could result in the image data sent to the host to be intentionally empty or uintentionally malformed.

pseudocode...
```c++
output_t tbb_pipelinestage_12(shared_ptr<dai::ImgFrame> input) {
    // validate the shared_ptr is valid *and* the ImgFrame is valid
    if (input && input->valid()) {
        // process the input frame data
        return imgprocessing(input);
    }
}
```

diablodale, 2022-02-03:
I've got some bandwidth again. Do you want to batch this with your zerocopy work?
If not, I can add `ImgFrame::operator bool` and `ImgFrame::empty()` on the existing impl.

themarpe, 2022-02-10:
Thanks @diablodale - I think separate will be fine for now, as I'm not sure on the timeline for other work currently

diablodale, 2022-02-15:
@themarpe , while writing a test case 😲🤭 I discovered that when `ImgFrame` is default constructed, it has a type of `YUV422i`.
My expectation is `None`

This is a sideaffect of `struct dai::RawImgFrame::Specs` being {} initialized therefore the value of the Type field is `0` aka `Type::YUV422i`.

It would be nice if `Type::NONE == 0`. But I believe it is too late for that. As a fallback, I suggest that the declaration of `struct dai::RawImgFrame::Specs` initialize it to NONE.

In `RawImgframe.hpp` change it to be
```c++
struct Specs {
    Type type = Type::NONE;
    unsigned int width;     // width in pixels
    unsigned int height;    // height in pixels
    unsigned int stride;    // defined as distance in bytes from pix(y,x) to pix(y+1,x)
    unsigned int bytesPP;   // bytes per pixel (for LUT types 1)
    unsigned int p1Offset;  // Offset to first plane
    unsigned int p2Offset;  // Offset to second plane
    unsigned int p3Offset;  // Offset to third plane

    DEPTHAI_SERIALIZE(Specs, type, width, height, stride, bytesPP, p1Offset, p2Offset, p3Offset);
};
```

I can include that one-line change in this PR's work if you approve my research+fix. 🤔

closed 2022-02-23 by themarpe