luxonis/depthai-core #283: `config.hpp` disables OpenCV support even though libraries are compiled with it

diablodale, 2021-11-26:
The currently generated `config.hpp` disables OpenCV support even though the library was built with such support. There is a logic error within it. Please note, the file itself remarks "Generated by CMake build tool." so this might actually be a cmake issue.

### Setup

* Microsoft Windows [Version 10.0.19044.1348]
* VS2019 Community v16.11.7
* cmake 3.19.1
* depthai-core v2.13.2

### Repro

1. config and build depthai-core with: opencv support, x64, static library
2. cmake install to a directory
3. write an depthai app that uses depthai opencv support like `mycvframe = mydaiframe.getFrame(false)`
4. configure cmake to build your app and use find_package(depthai) to that install directory
5. build

### Result

Build will fail with errors like
```
[build] ..\dp.oak\SensorOak.cpp(10): error C2679: binary '=': no operator found which takes a right-hand operand of type 'void' (or there is no acceptable conversion)
[build] ..\dp.oak\SensorOak.cpp(10): note: while trying to match the argument list '(cv::Mat, void)'
```

The compiler is complaining that the left side (cv::mat) wants a cv::mat. But the right hand side "...getFrame()' returns void.
The issue is `ImgFrame.hpp` is using the opencv-disabled clause of the APIs...which return void.
This is due to `DEPTHAI_HAVE_OPENCV_SUPPORT` not being defined
Which is due to the errant logic in depthai/build/config.hpp

```c++
/*
 * Generated by CMake build tool.
 */
#pragma once

// This build supports OpenCV integration?
#define DEPTHAI_HAVE_OPENCV_SUPPORT

// Build specific settings overwrite
#ifdef DEPTHAI_TARGET_CORE
#ifndef DEPTHAI_TARGET_OPENCV
#undef DEPTHAI_HAVE_OPENCV_SUPPORT
#endif
#endif
```

`DEPTHAI_TARGET_CORE` is `1`
and `DEPTHAI_TARGET_OPENCV` is not defined.
Therefore the header defines it on line 7 and then undefines it on line 12.

### Expected

OpenCV support in depthai to be available and clean compile

### Workaround

Hack `config.hpp` in the depthai install directory to remove the errant logic.

themarpe, 2021-11-27:
@diablodale do you link against `depthai::core` or `depthai::opencv`?
OpenCV support is only enabled when linking against `depthai::opencv` target.

For reference, the installed targets and their differences (`depthaiTargets.cmake`):
```
# Create imported target depthai::core
add_library(depthai::core STATIC IMPORTED)

set_target_properties(depthai::core PROPERTIES
  INTERFACE_COMPILE_DEFINITIONS "DEPTHAI_TARGET_CORE"
  INTERFACE_COMPILE_FEATURES "cxx_std_14"
  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;${_IMPORT_PREFIX}/include/depthai-shared/3rdparty"
  INTERFACE_LINK_LIBRARIES "\$<LINK_ONLY:depthai::depthai-resources>;nlohmann_json::nlohmann_json;XLink;\$<LINK_ONLY:Threads::Threads>;\$<LINK_ONLY:BZip2::bz2>;\$<LINK_ONLY:FP16::fp16>;\$<LINK_ONLY:archive_static>;\$<LINK_ONLY:spdlog::spdlog>;\$<LINK_ONLY:ZLIB::zlib>;\$<LINK_ONLY:Backward::Backward>"
)

# Create imported target depthai::opencv
add_library(depthai::opencv STATIC IMPORTED)

set_target_properties(depthai::opencv PROPERTIES
  INTERFACE_COMPILE_DEFINITIONS "DEPTHAI_TARGET_OPENCV"
  INTERFACE_LINK_LIBRARIES "opencv_core;opencv_imgproc;depthai::core"
)
```

diablodale, 2021-11-27:
Oh yes. Relevant part of my CMakeFiles.txt
```
target_link_libraries(dp.oak
    PRIVATE depthai::core depthai::opencv
    ...
```
Linking is not the problem.
the cmake build+install process is generating an incorrect header. Or it is setting incorrect defines which cause the header to undef.
Which causes the compilation step (before linking) to fail because the header defines the wrong signatures.

diablodale, 2021-11-27:
Sidebar...just a few minutes ago got my first "hello world" of OAK-D working with my app. Depth data...
[image]

themarpe, 2021-11-27:
@diablodale 
Can you post the installed `lib/cmake/depthai/depthaiTargets.cmake` file?

> the cmake build+install process is generating an incorrect header. Or it is setting incorrect defines which cause the header to undef.

Its seems like it doesn't have the correct define set up - `depthai::opencv` should bring with it `DEPTHAI_TARGET_OPENCV` define, which nullyfies the final `#undef` of opencv support.

If `depthaiTargets.cmake` file looks correct, maybe try linking only to `depthai::opencv` although most likely won't change anything...

diablodale, 2021-11-28:
Closing issue -- no repro.

I can't replicate this anymore, and I don't know why. The needed defines are happening and the headers now enabling the right signatures.
😵🤷‍♀️

closed 2021-11-28 by diablodale