luxonis/depthai-core #284: Unreachable code in `dai::DataOutputQueue::get()` -- unclear intention

diablodale, 2021-11-26:
Unreachable code in `dai::DataOutputQueue::get()` -- the intention is unclear. Needs a review.
On line 175-6, do you want to throw, or return nullptr. Can't do both.

https://github.com/luxonis/depthai-core/blob/efa3deb7c17e223238696416cd45d2d3eb35674f/include/depthai/device/DataQueue.hpp#L171-L179

### Setup

* all OS
* all compilers
* depthai-core v2.13.2

### Repro

MSVC compiler told me it is unreachable.
Verified by code review. It is unreachable.

### Result

The code will throw. But it is unclear if that is the dev's intention.

### Expected

Code matches dev's intention.

themarpe, 2021-11-27:
`return nullptr` is redundant in this case - the underlying queue was destructed in this case. 
If the underlying queue was destructed, then the caller is given the reason why and gets out of the blocked call.

diablodale, 2022-02-03:
verified it is merged into `develop` codebase

closed 2022-02-03 by diablodale