luxonis/depthai-core #291: (re)start and watchdog timing is fragile; leads to device not found and startup failures

diablodale, 2021-12-01:
When the device is stopped and started, the device will sometimes not be found or fail to start.

While testing the fix for issue #278 I've isolated a reproducible failure for devices to be found and/or startup. From the discussion in the other issue, it might be related to timing, watchdog, etc. The current approach seems to be fragile.

### Setup

* Microsoft Windows [Version 10.0.19044.1348]
* MSVS 2019 Community v16.11.7
* depthai-core v2.13.3 built x64, debug, static lib

### Repro

1. cmake config and build depthai-core for x64, debug, static lib, all examples, all tests
2. Run ctest with additional params to loop the tests, e.g. `--repeat until-fail:3 -VV`

### Result

```
[ctest] The following tests FAILED:
[ctest] 	 23 - imu_gyroscope_accelerometer (Failed)
[ctest] 	 53 - system_information (Failed)
[ctest] 	 54 - disparity_encoding (Failed)
[ctest] 	 55 - rgb_encoding (Failed)
[ctest] 	 57 - rgb_mono_encoding (Failed)
[ctest] 	 58 - encoding_max_limit (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
```

All the failures are similar. Here is full output of ctest
[fail on loop.zip](https://github.com/luxonis/depthai-core/files/7637116/fail.on.loop.zip)

And two examples from that log showing similarity. Notice the tests sometimes pass but fail on later loops

```
...
[ctest] 23: Accelerometer [m/s^2]: x: -0.575 y: 10.036 z: -0.306 
[ctest] 23: Gyroscope timestamp: 5086 ms
[ctest] 23: Gyroscope [rad/s]-- After process executed,  produced the following exit code: Process terminated due to timeout
[ctest]     Test #23: imu_gyroscope_accelerometer .........   Passed   10.74 sec
[ctest]       Start 23: imu_gyroscope_accelerometer
[ctest] 
[ctest] 23: Test command: "C:\Program Files\CMake\bin\cmake.exe" "-DTIMEOUT_SECONDS=10" "-P" "C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake" "C:/njs/depthai-core/build/examples/imu_gyroscope_accelerometer.exe"
[ctest] 23: Environment variables: 
[ctest] 23:  DEPTHAI_LEVEL=debug
[ctest] 23:  UBSAN_OPTIONS=halt_on_error=1
[ctest] 23: Test timeout computed to be: 1500
[ctest] 23: -- found -P
[ctest] 23: -- arguments: C:/njs/depthai-core/build/examples/imu_gyroscope_accelerometer.exe;
[ctest] 23: [2021-12-01 21:20:31.744] [debug] Library information - version: 2.13.3, commit: ce43051c2ea16088437eb377f18f9ee15214229d from 2021-12-01 20:51:22 +0100, build: 2021-12-01 19:56:05 +0000
[ctest] 23: [2021-12-01 21:20:31.748] [debug] Initialize - finished
[ctest] 23: [2021-12-01 21:20:31.878] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.15.tar.xz' open: 5ms, archive read: 126ms
[ctest] 23: [2021-12-01 21:20:32.328] [debug] Resources - Archive 'depthai-device-fwp-48fca4a443d841221c94c70d5c05e7e946168636.tar.xz' open: 4ms, archive read: 578ms
[ctest] 23: Stack trace (most recent call last):
[ctest] 23: #9    Object "", at 00007FFAC9657034, in BaseThreadInitThunk
[ctest] 23: #8    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_main.cpp", line 17, in mainCRTStartup [00007FF63F33D5EE]
[ctest] 23: #7    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 331, in __scrt_common_main [00007FF63F33D2BE]
[ctest] 23: #6    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 288, in __scrt_common_main_seh [00007FF63F33D3FE]
[ctest] 23: #5    Source "d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl", line 79, in invoke_main [00007FF63F33D559]
[ctest] 23: #4    Source "C:\njs\depthai-core\examples\IMU\imu_gyroscope_accelerometer.cpp", line 36, in main [00007FF63EB47068]
[ctest] 23:          33:     imu->out.link(xlinkOut->input);
[ctest] 23:          34: 
[ctest] 23:          35:     // Pipeline is defined, now we can connect to the device
[ctest] 23:       >  36:     dai::Device d(pipeline);
[ctest] 23:          37: 
[ctest] 23:          38:     bool firstTs = false;
[ctest] 23: #3    Source "C:\njs\depthai-core\src\device\Device.cpp", line 26, in dai::Device::Device [00007FF63EFEA385]
[ctest] 23:          23: // Common explicit instantiation, to remove the need to define in header
[ctest] 23:          24: constexpr std::size_t Device::EVENT_QUEUE_MAXIMUM_SIZE;
[ctest] 23:          25: 
[... 55 lines cut ...]
[ctest] 53:          24: constexpr std::size_t Device::EVENT_QUEUE_MAXIMUM_SIZE;
[ctest] 53:          25: 
[ctest] 53:       >  26: Device::Device(const Pipeline& pipeline) : DeviceBase(pipeline.getOpenVINOVersion()) {
[ctest] 53:          27:     tryStartPipeline(pipeline);
[ctest] 53:          28: }
[ctest] 53: #2    Source "C:\njs\depthai-core\src\device\DeviceBase.cpp", line 256, in dai::DeviceBase::DeviceBase [00007FF6A10EA5DF]
[ctest] 53:         254:     // If no device found, throw
[ctest] 53:         255:     if(!found) throw std::runtime_error("No available devices");
[ctest] 53:       > 256:     init(version, false, "");
[ctest] 53:         257: }
[ctest] 53:         258: 
[ctest] 53:         259: DeviceBase::DeviceBase(OpenVINO::Version version, const char* pathToCmd) {
[ctest] 53: #1    Object "", at 00007FFAC208B460, in CxxThrowException
[ctest] 53: #0    Object "", at 00007FFAC7E54F69, in RaiseException
[ctest] 53: -- After process executed,  produced the following exit code: -529697949
[ctest] 53: CMake Error at C:/njs/depthai-core/examples/cmake/ExecuteTestTimeout.cmake:42 (message):
[ctest] 53:    produced an error (-529697949) while running
[ctest] 53: 
[ctest] 53: 
[ctest]     Test #53: system_information ..................***Failed    5.22 sec
```

### Expected

No failures to find devices on restart

Luxonis-Brandon, 2021-12-01:
Thank you for the discovery and thorough documentation!  We will dig in.

diablodale, 2021-12-02:
Perhaps this is a situation where the host suddenly disconnects (e.g. EXE is force killed) and therefore does not send a friendly "device please shutdown" message. I've seen code for host-side watchdog. Makes me suspect there is also a device-side watchdog. And in the case of the host disappearing, the device probably keeps running until the device-side watchdog times-out and leads to a device self-shutdown/reset.

I looked at the USB device list in Windows. When not "running", my OAK-D appears as a MyraidX device. When it is "running" it appears as a Luxonis device. This explains the Windows "beep-boop" sound on OAK stop/start...the devices are changed/found.

Perhaps when the OAK is a "Luxonis", the host-side device detection code fails to find it and issues these "device not found" errors. Also guessing...when the host suddenly disappears, there is a period of time when the device is still a "Luxonis" before the device-side watchdog causes the device itself to shutdown and reset to a MyraidX.

If this gap in device detection only happens when the host suddendly disappears without issuing the friendly "device please shutdown" message...I think it is reasonable to have a fixed/bound limit for that no-detect-gap. I see in `DeviceBase.cpp` the `DEPTHAI_WATCHDOG` env var affects the host-side watchdog. I also see a host-side `WAIT_FOR_CONNECT_TIMEOUT` constant. But, I don't see a device-side watchdog setting.

I would hope there is an USB-level event that occurs when the host-side app disappears which causes the OS to clean/shutdown all the (USB) resources of the app. That something "USB" is triggered on both sides of the USB connection. And that a device-side USB event would trigger the device to self-shutdown/reset. But if such doesn't exist, then a watchdog might have been your approach and this issue is more about configuration/constant tuning.

themarpe, 2021-12-02:
@diablodale 
`DEPTHAI_WATCHDOG` is more of a development variable, as it controls the device watchdog. There are also some limitations of values that can be set.

This event in our case is lack of WD, regarding anything more low level, I'm actually not sure about anything straight forward from user space. From libusb case, there is a "claim interface", but that deals with OS and whether it allows other processes to access that interface.

Regarding above errors, it seems like they occur because next example starts sooner than the device is available.
Testing examples is limited in that sense as it doesn't really care about this scenario as much - its like an easy to use thing to iterate through all of them. Main tests should take care of starting up and shutting down properly.

As of right now, best case for using example testing in this manner is some sort of delay between tests. Best spot is either ctest if it supports that or `ExecuteTestTimeout.cmake`with a post-test delay

diablodale, 2021-12-02:
When you write "it controls the device watchdog"...do you mean a watchdog that runs on the device? Or a watchdog on the host which pings the device? From what I can see in the SDK code, it is the latter.

The fail scenarios being exposed in these test cases are not something isolated to tests. It is not realistic to require all systems that use depthai and Luxonis devices to "properly" shutdown their connections. Won't happen and that's a setup for failure. In the real world...unpredictable things happen. Apps suddenly exit, ether+usb cables are accidently unplugged, etc.

I'm raising awareness on real-world problems...using the test cases as examples. The resolution isn't to tweak ctest because the real-world doesn't use ctest or these examples. The real-world has connectivity issues, and app crashes. And when these happen...depthai and Luxonis devices needs to be findable and restartable.🧐

themarpe, 2021-12-02:
> When you write "it controls the device watchdog"...do you mean a watchdog that runs on the device? Or a watchdog on the host which pings the device? From what I can see in the SDK code, it is the latter.

Its both, sets it on host as a half of that for a safe ping period, and the device side with the actual value.

I didn't go through all of the failed cases, but from glimpse it seems they fail with "No device found" exception?
If that is the case, then this is just an observation of current HW setup, otherwise, if its an actual crash that cant be handled gracefully, we have to make sure it works and can be handled.

(also Windows seem to not print the exception messages by default, which makes the examples a bit more cryptic in this case. Preferably all examples would gracefully handle exceptions and print them out)

diablodale, 2021-12-03:
I ask for your patience in digging into details on this issue. I think I'm seeing a pattern because I am reproducing similar failures using different app/tools. So I need to understand a bit more to sort what I'm seeing.

I think (please correct me) I see on lines

https://github.com/luxonis/depthai-core/blob/0cc6284767ae21e8747d26562cb058d181e0ca34/src/device/DeviceBase.cpp#L464-L471

it parses the `DEPTHAI_WATCHDOG` or uses the default `device::XLINK_WATCHDOG_TIMEOUT` to set...

* `config.preboot.watchdogTimeoutMs` that is "Prepend preboot config" from `getDeviceFirmware()`. This value will get to the device when the firmware is uploaded to the device. The firmware code is Luxonis closed-source.
* `watchdogTimeout` that is host-side and used in later spawned threads

From what I can understand these two host-side and device-side watchdog polling also interplay with device searches. When an app causes directly or indirectly a scan/search for a device, then there is also

https://github.com/luxonis/depthai-core/blob/0cc6284767ae21e8747d26562cb058d181e0ca34/include/depthai/device/DeviceBase.hpp#L46

Would not that mean that if... example...

1. env var DEPTHAI_WATCHDOG=5000
2. causes host app to ping at 2500ms, device-side to ping/listen at 5000ms
3. host app crashes
4. host app "immediately" starts
5. host app scans for 3 seconds `DEFAULT_SEARCH_TIME`
6. ...but device is in "running" mode and waiting 5 seconds
7. so host-side app times out to find a device to which it can connect

I acknowledge the device firmware and implementation of its device-side watchdog is close-source. Can you share the intention/rules for how the device-side watchdog `config.preboot.watchdogTimeoutMs` runs?
Does it ping or listen?
What is the action after `config.preboot.watchdogTimeoutMs` is exceeded?

themarpe, 2021-12-07:
Sorry for late response.

Correct on above - the watchdog value that gets set to device and host is used to periodically "keep alive" the device FW.
The device side watchdog takes the given value (or the default) and it uses that to setup the internal watchdog. This internal watchdog if not "bumped" every x milliseconds, will cause a device reset. Device listens for incoming XLink packets, which indicate a bump. This is what the watchdog thread does - sends down a packet to bump the devices internal watchdog.

> From what I can understand these two host-side and device-side watchdog polling also interplay with device searches. When an app causes directly or indirectly a scan/search for a device, then there is also

> Would not that mean that if... example...

The above scenario would play out as described - the default search time wouldn't work in that case.
The default search time isn't meant to address this case, but it might be a "nice to have" if that was increased as well. 
This is an artifact of "testing the examples" as they aren't designed to properly handle this scenarios.

Best way IMO would be to add SIGINT handler to all examples and modify `ExecuteTestTimeout.cmake` to send that instead. That way, proper cleanup could be done and "testing examples" would be cleaner and more useful as general tests. (As are current tests now).

A downside to this is that this would make examples bulkier and might obfuscate the main idea/lesson of the example in the surrounding "application boilerplate"

diablodale, 2021-12-08:
I don't think the examples should change. They have been good at surfacing errant code.
I discourage SIGINT handlers. That isn't real-world coding. SIGINT is never in Windows real-or-synthetic. And SIGINT doesn't help with app crashes and/or network goes down.

This isn't about making the test+examples run pretty. I've written this above but my communication isn't clear enough. My mistake. I'll try to explain in another way.....🤓

My goal here is to understand the ecosystem and how it can lead to a "device not found".
I don't yet think there is a "bug" in code. So far, this seems more timing/configuration issue.
⚠ However, I can often reproduce it in the examples and in `depthai_demo.py`. These 'device not found' are not rare. They are often, and occurs across C++, Python, release, debug, compiled by me, and python wheel compiled by you. If it happens that often, then this is something that needs my investigation.

Real-world use cases from my customers... but substituting Luxonis devices that have this "not found" issue:

* Multiple Luxonis devices are hanging 6 meters overhead during a live music performance... and the band's app crashes... and on restart of the app gets "device not found" for all three devices. A person can't physically go to the devices.
* Multiple Luxonis devices are part of a interactive museum installation. The app crashes. The museum has paying customers there for that installation. Sending a tech onsite is hours downtime. And turning off the power to the whole installation is excessive.

These are likely scenarios by my customers. Most of my customers use Microsoft Kinect devices. To use Luxonis devices, they need from me a solution that if their app crashes, and they restart their app, their app can immediately find the Luxonis devices and continue quickly.

Give me some more time to investigate. ⌚👍

diablodale, 2021-12-08:
Hey @alex-luxonis, in situations of a "fw crash" like #278... what is the behavior with respect to the device-side watchdog, and the device's ability through that watchdog to self-reset (back to a "MyriadX' device which can be detected and programmed)?

In #278 the device didn't send frames of data to the host. And...the device didn't self-reset. It was almost like the watchdogs on the host-side and device-side were running and pinging each other. Therefore the device didn't reset itself.

themarpe, 2021-12-10:
I think we are on the same page now - thought you meant from the perspective of testing examples. And yup, in that case SIGINT is just a more graceful shutdown, while any shutdown should still correctly work afterwards.

The issue of devices sometimes needing a powercycle, is probably a FW one and we are yet to completely resolve it. We've seen it here and there, but AFAIK aren't yet 100% on the nature of it.

By device didn't reset I assume you mean didn't come back into ROM BL state (and visible in Device Manager, etc...)? I think this is the artifact of above issue, but AFAIK isn't related to still watchdog being bumped on device side, but USB conn/controller getting stuck, as the reset itself almost 100% happens. 

@alex-luxonis on more details.

Also, Alex, I think this treads nicely into having the graceful shutdown on device side as well, as I've pointed out the other day? Probably should help with this cases (so when we still have power to the device but no more connection with the host)

alex-luxonis, 2021-12-10:
On the issue https://github.com/luxonis/depthai-core/issues/278, I believe the device did crash and the hardware watchdog should have soft-reset it back to boot ROM - "Movidius MyriadX", but in the Windows logs attached I do see lots of USB write errors, up to the point ctest was terminated (10 seconds timeout). Maybe that's a separate host library issue, that it didn't close the connection for a no longer available device, and kept retrying some communication?

https://github.com/luxonis/depthai-core/issues/278#issuecomment-977215028
> `#define ERROR_NO_SUCH_DEVICE 433L`

At least on Linux I didn't see that behavior, the program ended with something along:
`RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'preview' (X_LINK_ERROR)'`

About the watchdog not being able to soft-reset, yes, that should never happen. 
In very rare cases we saw a need to power-cycle:
- as the ROM USB device failed to enumerate, and IIRC on Linux it was possible to avoid the power-cycle by re-binding the driver / re-enumerating everything:
`echo -n "0000:00:14.0" | sudo tee /sys/bus/pci/drivers/xhci_hcd/unbind; sleep 1; echo -n "0000:00:14.0" | sudo tee /sys/bus/pci/drivers/xhci_hcd/bind`
- if sending the firmware blob was somehow interrupted, and the ROM ended up in a bad state, not being able to boot at the following attempts
`RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND`

diablodale, 2021-12-10:
Then this meta-topic involves USB and/or device-side firmware. I can't help directly with that code.
But I can find repro cases 😉  Another I found today, which I have seen before and now isolated in issue #300

diablodale, 2022-02-02:
Hi. The current `develop` branch is randomly failing some of the test+example suite on Windows, MSVC, share, debug.
Changes in host+device code are now more fragile in finding a device.

```
[ctest] The following tests FAILED:
[ctest] 	 18 - device_queue_event (Failed)
[ctest] 	 47 - script_nndata_example (Failed)
[ctest] 	 59 - system_information (Failed)
[ctest] 	 61 - rgb_encoding (Failed)
[ctest] 	 63 - rgb_mono_encoding (Failed)
[ctest] 	 68 - apriltag (Failed)
[ctest] Errors while running CTest
[ctest] CTest finished with return code 8
```

The fails all occur at the same place; it can't find a device.
```
[ctest] #2    Source "C:\njs\depthai-core\src\device\DeviceBase.cpp", line 258, in dai::DeviceBase::DeviceBase [00007FFBEE5A6770]
[ctest]         256:     // If no device found, throw
[ctest]         257:     if(!found) throw std::runtime_error("No available devices");
[ctest]       > 258:     init(version, false, "");
[ctest]         259: }
```

After the suite of 69 completes, I can manually run the tests that failed at the command line and they pass. I suspect race/timing issue.

themarpe, 2022-02-03:
@diablodale do you happen to have the "latest" commit you've tested it against which doesn't show these timing issues?

diablodale, 2022-02-03:
Since it is the same repo steps and same line of code failing and same race-like behavior...I suspect this is all the same issue of failure to reliably handle the shutdown/startup of the device while the app (the test suite) is trying to re-connect.

Yesterday released my first alpha for my app 🎉 so today I wanted to verify bug fixes from your team that are on develop. 
My app used a depthai-core private build based on v2.14.1 + my fixes that were merged to develop but not yet on main.

Since we have already identified a timing relation between app stop and/or fail...and the time when a device can again be connected... I suspect that subtle changes in my laptop, the develop branch code, or a butterfly in Japan caused just enough delay so to cause the reconnect timeout to randomly exceed its limit on 8% of the tests in a full suite run.

Thursday I'll change the connect timeout in depthai-core and see if that makes it go away. And if I shorten it that it makes it more common. If so, I think this is just choosing a timeout that is adequate for most cases. We touched on this idea elsewhere.
I'll also run the test suite with my app's private build to see if it repros there.

themarpe, 2022-02-03:
Nice! 
If you have it published somewhere to be able to take a look, feel free to share:)

> If so, I think this is just choosing a timeout that is adequate for most cases. We touched on this idea elsewhere.

I agree - we've discussed in another issue.

Thanks - hopefully I'll have WD protection ready to be merged also, just have to recheck one issue that seemed like that surfaced.

diablodale, 2022-02-03:
Casual test results of windows, shared lib, debug with `develop` 04d5149e7cfddc4dc871d55d1a3d9ee270b76835

`DEFAULT_SEARCH_TIME` | failed tests
---- | ----
2 sec | 18
3 sec | 6
5 sec | 0

PR #345 (2022-02-03): Added watchdog protection
Increases robustness when interfacing with USB devices. Protects against processes that exit before whole FW is sent to device at boot stage and removes the need to power cycle the device in that case.

Includes https://github.com/luxonis/XLink/pull/39 changes which address similar case but at MXID retrieval step.

Fixes: #291

CC: @diablodale for giving it a try on how it pairs with #291
EDIT: And on #300