luxonis/depthai-core #294: The model path exists in Chinese, depthai_demo.py cannot run error

hello-word-yang, 2021-12-06:
There is Chinese in my model path, and an error is reported when loading the file model as shown in the figure below.

[image]

We found that it was because the std::ifstream method failed to open a file with a Chinese path.

https://github.com/luxonis/depthai-core/blob/0cc6284767ae21e8747d26562cb058d181e0ca34/src/pipeline/AssetManager.cpp#L32

Can you help us solve this problem? thanks.

Luxonis-Brandon, 2021-12-07:
Asking team.  Not immediately sure.

themarpe, 2021-12-07:
Hi @hello-word-yang 
Seems like its a Windows specific issue. Checking on how to properly handle this for Windows platform...

themarpe, 2021-12-07:
@hello-word-yang I think what might immediately remedy the issue is setting UTF-8 support in Locale under Windows settings.

[image]

(Source: https://pete.akeo.ie/2019/11/powershell-script-to-convert-utf-8.html)

Otherwise it seems like we have to handle the path as UTF-16 and use wide strings in Windows case...
Will look into adding clean support for that as well

themarpe, 2021-12-07:
Note: this is a system wide change and may affect other software running on the computer. I advise to continue with precaution and rollback if undesired side effects show up

hello-word-yang, 2021-12-07:
hi, @themarpe 

Thanks for your reply, I will test the method you mentioned as soon as possible.

hello-word-yang, 2021-12-08:
hi, @themarpe 

After my test, after modifying the encoding format of the system to utf-8, depthai-demo.py can run normally when the model path contains Chinese path.

Thank you for the solution.

But I think it is better to solve this problem from the code level, because it is impossible for all users to modify the system.

themarpe, 2021-12-08:
> But I think it is better to solve this problem from the code level, because it is impossible for all users to modify the system.

I agree - I wish Windows will push more into UTF-8 direction, so this won't be a platform dependent thing anymore. 
Will see how I'll be with time the next couple of weeks. Thanks for testing it out:)

closed 2021-12-08 by themarpe