luxonis/depthai-core #300: DeviceBootloader::getFirstAvailableDevice() can return unusable result

diablodale, 2021-12-10:
Calls to `DeviceBootloader::getFirstAvailableDevice()` can return a "found" boolean yet the deviceinfo returned has a name of `"<error>"`

Timing of the repro steps is important to trigger the error state.

### Setup

* Microsoft Windows [Version 10.0.19044.1348]
* VisStudio v.16.11.7
* `depthai-core` v2.13.3 + local ownership fixes
* OAK-D

### Repro

1. cmake config for static library, x64, debug + build all tests and examples
2. cmake build
3. make a BAT file that runs `tests\image_manip_node_test.exe` in a loop
4. run that BAT file and let it loop at least 5 times
5. press Ctrl-C while the EXE is running during a loop. The exact timing of the ctrl-c seems to matter. You may have to retry this several times until you get the failure state.
6. Notice `image_manip_node_test.exe` stops due to the ctrl-c, and you are back at the command prompt. You might have to type yes to end the BAT file.
7. Run `build\examples\bootloader_version.exe`

### Result

```
Found device with name: <error>
```
and the EXE is still running...seems to be blocked or in a loop somewhere.
I can press ctrl-c to exit the app. And then run `bootloader_version.exe` over and over... it always gives the error and blocks.
Even as I am typing this bug, easily 15 minutes has passed. And still it reports an error if I run the app. This makes me suspect a problem in USB/OS or a device-side problem.
If I look in Windows Device Manager during this error state and the app is not running, I see the oak-d appear as "Movidius MyriadX".

If I unplug the USB cable computer<->oak-d and reconnect it. The app reports "No devices found". And then several seconds later, Windows itself reported: USB device not recognized.
[image]

### Expected

I get the valid name and bootloader version of the device.
Or...
I get found=false
And...
`DeviceBootloader` constructor doesn't infinitely block

### Notes

When I create the repo and put `bootloader_version` in the debugger, that error output goes to console, and the app doesn't exit.

After the apps's call to `getFirstAvailableDevice()` the vars are
[image]

That above is unexpected return result...

* It found an UNBOOTED device, yet that device has the std::string name of `"<error>"`. That string appears several times in the XLink library code.
* It reported back to the caller found=true in the tuple, yet that name is definitely not a valid name.

Separate from any problem in USB or device-side, this also seems to be problems in...

* getFirstAvailableDevice() api. It should have reported found=false.
* when this or any invalid `DeviceInfo` is passed to the `DeviceBootloader` constructor, it should throw an exception and not block

Tracing down into `getAllConnectedDevices()` and `XLinkFindAllSuitableDevices()` they return only 1 device and that it has that `<error>` name.

If I press *pause* in the debugger when the app is blocked, I see only 4 threads. Only one of those threads (the Main Thread) is doing something interesting...

```
ntdll.dll!00007ffa884ccdf4() (Unknown Source:0)
KernelBase.dll!00007ffa85ff1a5e() (Unknown Source:0)
KernelBase.dll!00007ffa85ffb295() (Unknown Source:0)
winusb.dll!00007ffa496f2d60() (Unknown Source:0)
bootloader_version.exe!usb_bulk_read(_usb_han * han, unsigned char ep, void * buffer, unsigned __int64 sz, unsigned int * read_bytes, int timeout_ms) Line 499 (c:\njs\XLink\src\pc\Win\src\win_usb.c:499)
bootloader_version.exe!gen_addr_mx_id(void * devInfo, _SP_DEVINFO_DATA * devInfoData, int pid, char * * refDevicePath) Line 696 (c:\njs\XLink\src\pc\Win\src\win_usb.c:696)
bootloader_version.exe!win_usb_find_device(unsigned int idx, char * addr, unsigned int addrsize, void * * device, int vid, int pid) Line 925 (c:\njs\XLink\src\pc\Win\src\win_usb.c:925)
bootloader_version.exe!usb_find_device(unsigned int idx, char * addr, unsigned int addrsize, void * * device, int vid, int pid) Line 647 (c:\njs\XLink\src\pc\protocols\usb_boot.c:647)
bootloader_version.exe!getUSBDeviceName(int index, XLinkDeviceState_t state, const deviceDesc_t in_deviceRequirements, deviceDesc_t * out_foundDevice) Line 345 (c:\njs\XLink\src\pc\PlatformDeviceSearch.c:345)
bootloader_version.exe!XLinkPlatformFindDeviceName(XLinkDeviceState_t state, const deviceDesc_t in_deviceRequirements, deviceDesc_t * out_foundDevice) Line 62 (c:\njs\XLink\src\pc\PlatformDeviceSearch.c:62)
bootloader_version.exe!XLinkFindFirstSuitableDevice(XLinkDeviceState_t state, const deviceDesc_t in_deviceRequirements, deviceDesc_t * out_foundDevice) Line 157 (c:\njs\XLink\src\shared\XLinkDevice.c:157)
bootloader_version.exe!dai::XLinkConnection::initDevice(const dai::DeviceInfo & deviceToInit, XLinkDeviceState_t expectedState) Line 255 (c:\njs\depthai-core\src\xlink\XLinkConnection.cpp:255)
bootloader_version.exe!dai::XLinkConnection::XLinkConnection(const dai::DeviceInfo & deviceDesc, std::vector<unsigned char,std::allocator<unsigned char>> mvcmdBinary, XLinkDeviceState_t expectedState) Line 148 (c:\njs\depthai-core\src\xlink\XLinkConnection.cpp:148)
bootloader_version.exe!std::_Construct_in_place<dai::XLinkConnection,dai::DeviceInfo &,std::vector<unsigned char,std::allocator<unsigned char>>,enum XLinkDeviceState_t>(dai::XLinkConnection & _Obj, dai::DeviceInfo & <_Args_0>, std::vector<unsigned char,std::allocator<unsigned char>> && <_Args_1>, XLinkDeviceState_t && <_Args_2>) Line 160 (c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\xutility:160)
bootloader_version.exe!std::_Ref_count_obj2<dai::XLinkConnection>::_Ref_count_obj2<dai::XLinkConnection><dai::DeviceInfo &,std::vector<unsigned char,std::allocator<unsigned char>>,enum XLinkDeviceState_t>(dai::DeviceInfo & <_Args_0>, std::vector<unsigned char,std::allocator<unsigned char>> && <_Args_1>, XLinkDeviceState_t && <_Args_2>) Line 2029 (c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\memory:2029)
bootloader_version.exe!std::make_shared<dai::XLinkConnection,dai::DeviceInfo &,std::vector<unsigned char,std::allocator<unsigned char>>,enum XLinkDeviceState_t>(dai::DeviceInfo & <_Args_0>, std::vector<unsigned char,std::allocator<unsigned char>> && <_Args_1>, XLinkDeviceState_t && <_Args_2>) Line 2748 (c:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include\memory:2748)
bootloader_version.exe!dai::DeviceBootloader::init(bool embeddedMvcmd, const std::string & pathToMvcmd, tl::optional<enum dai::bootloader::Type> type, bool allowBlFlash) Line 241 (c:\njs\depthai-core\src\device\DeviceBootloader.cpp:241)
bootloader_version.exe!dai::DeviceBootloader::DeviceBootloader(const dai::DeviceInfo & devInfo, bool allowFlashingBootloader) Line 216 (c:\njs\depthai-core\src\device\DeviceBootloader.cpp:216)
bootloader_version.exe!main(int argc, char * * argv) Line 10 (c:\njs\depthai-core\examples\bootloader\bootloader_version.cpp:10)
bootloader_version.exe!invoke_main() Line 79 (d:\a01\_work\6\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:79)
```

themarpe, 2021-12-10:
I think this is the state where ROM BL gets stuck as Alex mentioned. The `<error>` is mishandled here, will have to add a return result per `deviceDesc_t` found by XLink to properly notify about the issues. (Will add as a note to `device_search_improvements`)

As to why the `<error>` is set as name is because scanning for the device communicates with ROM BL to retrieve its MXID - I suspect when the loop is closed at the right timing, the ROM BL might receive a partial command and get stuck in an invalid state, which yields an error retrieving the MXID. (Note: there is a small caching between found USB addr and retrieved MXID, 500ms afaik - so its not retrieved multiple times if device scanning happens often) 

@alex-luxonis do we have any "atomicity" over USB writes? Could we modify our ROM BL interaction to always first send down a WD enable or something along this lines and afterwards a WD disable (again as an "atomic" write)?
Maybe 64B write for USB2? Although not sure if we'll get WD enable/disable into 64B, might require a bit of asm magic haha

diablodale, 2021-12-10:
This aligns with a refined repro. When I run `tests\image_manip_node_test.exe` there is a consistent sound experience.

1. I run the app.
2. Maybe a second passes.
3. Windows makes a "boop" sound.
4. Then a second passes.
5. Windows makes another "'boop" sound
6. The app exits

It seems I can get the error state when I press ctrl-c BEFORE the first "boop" sound (during step 2) from Windows. It varies within step 2 but if I get it towards the end of it just before step 3 "boop", then it creates the error scenario.

I put a string compare and spdlog::error() into `XLinkConnection::getAllConnectedDevices()` to look for `<error>`. It finds it in this error scenario and correctly cascades up a found=false. So the app doesn't block...but the device is still "dead".

The logging is showing some problem in `libusb_bulk_transfer`

```
C:\njs\depthai-core>build\examples\bootloader_version.exe 
E: [global] [         0] [] gen_addr_mx_id:697  libusb_bulk_transfer recv: Win32 Error 0xffffffff (Unable to retrieve error message)
[2021-12-10 20:40:06.545] [error] XLinkFindAllSuitableDevices() returned X_LINK_UNBOOTED device having name "<error>"
E: [global] [         0] [] gen_addr_mx_id:697  libusb_bulk_transfer recv: Win32 Error 0xffffffff (Unable to retrieve error message)
[2021-12-10 20:40:07.314] [error] XLinkFindAllSuitableDevices() returned X_LINK_UNBOOTED device having name "<error>"
E: [global] [         0] [] gen_addr_mx_id:697  libusb_bulk_transfer recv: Win32 Error 0xffffffff (Unable to retrieve error message)
[2021-12-10 20:40:08.082] [error] XLinkFindAllSuitableDevices() returned X_LINK_UNBOOTED device having name "<error>"
No devices found
```

themarpe, 2021-12-10:
> It seems I can get the error state when I press ctrl-c BEFORE the first "boop" sound (during step 2) from Windows. It varies within step 2 but if I get it towards the end of it just before step 3 "boop", then it creates the error scenario.

I think this is then either the MXID retrieval communication with ROM BL which is partial and confuses the ROM BL, or sending of the FW which is partial which confuses the ROM BL.

Eitherway, I think the above proposition is an option here, we'll discuss offline a bit more and see if anything else would be applicable as well. 

Thanks for pointing it out clearly - will get back to you soon.

themarpe, 2022-01-04:
@diablodale 
Addressed initially in `watchdog_protection` (bc4e6076cd8c26018a0155009af3bfd5a985f23d)
Upon an unclean exit the device might still present itself with an "error" mxid, but will recover in that state in around ~5s - without requiring a powercycle.

(TODO, some potential XLink improvements, tightening or exposing timings)

diablodale, 2022-01-06:
Got it. When this work is merged into develop, I will try and reproduce the OP.

closed 2022-02-23 by themarpe