luxonis/depthai-core #309: Examples

bobdavies2000, 2021-12-23:
It is probably my mistake but I have successfully followed the instructions to install the depthai-core SDK but when I cmake the examples, they are not found.  Specifically these instructions are the problem for me:

cmake -S. -Bbuild -D'DEPTHAI_BUILD_EXAMPLES=ON'  <<< works fine - no error messages
cmake --build build  <<< works fine - no errors
Then navigate to build/examples folder and run a preferred example

cd build/examples  <<< examples directory not there...

The environment is Windows 10 (latest build).  The depthai-core SDK is the latest as well.  I can see the examples directory with the source but there is no .sln file and the examples are not included in the depthai.sln.  I just got my Oak-D Lite cameras.

Bob Davies

themarpe, 2021-12-24:
@bobdavies2000 In that case it might be that the `DEPTHAI_BUILD_EXAMPLES` wasn't applied...
Can you specify it as such:
```
cmake -S . -B build -D DEPTHAI_BUILD_EXAMPLES=ON 
cmake --build build
```

Also try building a specific example:
```
cmake --build build --target rgb_preview
```

The above should work if examples are specified to be built

bobdavies2000, 2021-12-24:
themarpe - thanks for working late.  I ran your suggested command and it was enough for me to see the error.  Buried in the log was a log entry describing the location of the OpenCV that was found and it was a very old version in an SDK for another camera.  Once I defined the OPENCV_DIR environmental variable to point to my most recent version, the build worked.  Thanks for your help.

I am going to review all the examples now but the key question for me is whether the camera is now producing a point cloud.  Do you know if that is the case? 

Bob Davies

closed 2021-12-24 by bobdavies2000

themarpe, 2021-12-24:
@bobdavies2000 great - glad its working for you:)

As of right now, the device doesn't yet support outputting point cloud data directly. You may however do the conversion on the host with a library of choice.

bobdavies2000, 2021-12-24:
TheMarpe - thanks.  I am already doing that and I am just trying to save
the cycles on the host....

> Message ID: ***@***.***>
>

themarpe, 2021-12-24:
Sounds good - we'll notify you on this progress though if we decide to go for it.