luxonis/depthai-core #313: Cannot build using Visual Studio 2022.

yusuke-ota, 2021-12-29:
Hi.

I'm reporting an error that occurred when I tried to build this project in Visual Studio 2022.

## Problem

Currently, the "cpp-pm/hunter v0.23.258" used in this project supports Visual Studio 2019 and earlier.
ref: https://github.com/cpp-pm/hunter/blob/286702056cd26b99d1ee95a30327e6ad834bc2ad/cmake/modules/hunter_setup_msvc.cmake#L87-L92

So, I cannot build this project using Visual Studio 2022.

Error:
```
[hunter ** INTERNAL **] Unexpected MSVC_VERSION: '1930'
```

## Solution

Since "[cpp-pm/hunter v0.23.317](https://github.com/cpp-pm/hunter/releases/tag/v0.23.317)", "cpp-pm/hunter" supports building with Visual Studio 2022.
And I used v0.23.320 to avoid this error.

I hope you consider upgrading the dependent cpp-pm/hunter in the next update of this project.

Thanks.

themarpe, 2021-12-30:
@yusuke-ota 
Thanks for reporting - great timing, we'll update this before the release as well. We've made necessary changes a while ago, so should be clean update: https://github.com/cpp-pm/hunter/pull/498

themarpe, 2021-12-30:
Updated in 78bbc384135bd2d302411b26a94d17c8e5100e24

closed 2021-12-30 by themarpe