luxonis/depthai-core #316: feature request: add stb single-file libraries for C/C++

goloskokovic, 2021-12-30:
It should be possible to build examples without  OpenCV overhead using stb
https://github.com/nothings/stb

    image loader: stb_image.h
    image writer: stb_image_write.h
    image resizer: stb_image_resize.h
    font text rasterizer: stb_truetype.h
    typesafe containers: stb_ds.h
    2D rectangle: stb_rect_pack.h

Luxonis-Brandon, 2021-12-30:
Adding @themarpe for insight on this. (I'm not sufficiently technical to comment.)

themarpe, 2021-12-30:
Hi @goloskokovic 
I don't see a rendering library listed on their page. 

Eitherway, the main intention of the examples is to show a use case - which can be created from any set of libraries that one needs, OpenCV, SDL, OpenGL, ...

Core library itself can also be compiled without OpenCV at all to ease integration in various systems.

The reason we used OpenCV is that is readily available on development hosts and has a mostly equivalent use from Python as well. This creates further similarity between these two languages and makes examples more concise and easier to switch languages between them.

I'll close the issue for the time being, but feel free to keep debate on this going

closed 2021-12-30 by themarpe

goloskokovic, 2021-12-31:
There is a library based on stb that has rendering - SmallCV
https://github.com/zchrissirhcz/smallcv 

APIs: 
   
    Mat/imread()/imwrite()/imshow()

Replace #include <opencv2/opencv.hpp> with #include <smallcv/smallcv.hpp>

P.S. Happy New Year!