luxonis/depthai-core #318: nop serialization lib fails to correctly express large numbers, breaks builds and logic errors too

diablodale, 2021-12-30:
Main app builds are breaking due to the new NOP serialization library bugs as it was introduced into the `develop` branch.

```
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/encoding.h(784): error C4146: unary minus operator applied to unsigned type, result still unsigned
```

### Setup

* Microsoft Windows [Version 10.0.19044.1415]
* VS2019 16.11.8
* depthai-core develop 84f7c69ea3478d08615e0ed4d3fcb6c42b806823 + move/thread commit e90d8a71731ff5a2ee7172be3fe1a6047987d689

### Repro

1. config and build for x64, static library, debug
2. cmake install
3. point your main app to the install location
4. cmake config and build your main app

### Result

Repeated build errors ...

```
[build] C:\njs\depthai-core\.install\Debug\lib\cmake\depthai\dependencies\include\nop/base/encoding.h(784): error C4146: unary minus operator applied to unsigned type, result still unsigned
```

### Expected

No build errors in NOP code

### Notes

The error is due to a coding bug in NOP. The exact error and cause is discussed in MSVC documentation
https://docs.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warning-level-2-c4146?view=msvc-170#remarks

The compiler error is accurate and this probably leads to the comparison failing to match the dev's expectations.

The fix is at the link, excerpt...
> You can avoid C4146 by using INT_MIN or LLONG_MIN from <limits.h> or the C++ equivalent, <climits>. These values have signed types.

themarpe, 2021-12-30:
Thanks for reporting - we are debugging some libnop differences/issues at the moment and we'll take a look at this as well

themarpe, 2022-01-04:
Fixed in: 6fae450396133e6a850713ed4403efe0ff3a799b
By modifying libnop: https://github.com/luxonis/libnop/commit/83967ad391a30d8f871d57777a91ebd7f9370ad4

closed 2022-01-04 by themarpe